CommonWealthRobotics/BowlerStudio #3: Occasional HTTP 404

Octogonapus, 2016-12-30:
### Installed Version 
v0.20.1
### OS type and version number
Windows 10
### Expected behavior
BowlerStudio should open and display its welcome page like normal.
### Actual Behavior
Instead of displaying the welcome page, an HTTP error 404 page is displayed:

HTTP ERROR: 404
Problem accessing /BowlerStudio/Welcome-To-BowlerStudio/. Reason:
    Not Found

Here is a pastebin of the output: http://pastebin.com/uD7psjrP
### Steps to reproduce the behavior
Try deleting bowler-workspace and restarting.

Octogonapus, 2016-12-30:
I should add that restarting fixes the issue.

madhephaestus, 2017-03-13:
This was resolved by 9d4b9346873b594b64b49610291845f4a32bd91b

The issue was caused by loading the old asset repo initializeing.

closed 2017-03-13 by madhephaestus