CommonWealthRobotics/BowlerStudio #8: [FEATURE ADD] Auto-updater

madhephaestus, 2016-12-30:
_From @madhephaestus on May 30, 2016 23:8_

It would be nice for BowlerStudio to use its GitHub API hooks to find new versions and download them. THis would require a native script to manage the downloader. i think the best way to do it would be to have BowlerStudio have an updater mode. The native script would copy the normal jar to a second updater jar, then use that copy to run the update. This keeps the development in a single application, with all the API features, and none of the increased size of bundling a fat downloader application. The windows .bat file is the only OS script missing and would need to be added to the installer packaging before this could begin. 

_Copied from original issue: NeuronRobotics/BowlerStudio#52_

Octogonapus, 2019-12-04:
@madhephaestus Want to update this with the latest developments?

madhephaestus, 2019-12-05:
This feature exists for the Linux installer and is implemented in BASH. a similar script may be able to be made to work for MacOS, but Windows would need the script translated into .bat script

madhephaestus, 2020-03-19:
This issue is resolved and feature added as of 1.5.1

This project adds the functionality as a stand-alone bootstrapper application:

https://github.com/CommonWealthRobotics/LatestFromGithubLaunch

closed 2020-03-19 by madhephaestus