CommonWealthRobotics/BowlerStudio #54: Docs on hardware?

justinclift, 2018-04-28:
Just watched the initial [BowlerStudio intro video](https://www.youtube.com/watch?v=zx1Q_pZnQZE), it demos the software decently. :smile:

On the left hand menu in the demo, it shows a "Hardware" tab.  The demo though completely skips over that. :frowning:

Clicking the wiki link here in the repo... just goes back to the repo front (Code) page.  eg no docs in this repo, apart from the README.

Looking through the README... there doesn't seem to be any docs on what hardware is needed, useful, usable (etc).  As a practical example, I have a 3D printer, 3 axis CNC machine, hobby lathe, various motors and stepper drivers, webcams, Realsense camera - does infra + standard light spectrum - (and other stuff) at hand.

No seeing any info on what out of that would potentially be useful (apart from the mentions of 3D printing), how start with it, etc.

Err... any pointers to what the "Hardware" tab on the menu in the video contains? :smile:

madhephaestus, 2018-04-28:
Hey, Thanks for checking out my software! 
The tutorial of supported hardware as built in is here:
http://commonwealthrobotics.com/Bowler-Studio-Hardware/DyIOScripting/

- All webcams (supported by OpenCV) work with BowlerStudio. 
- All Marlin flavored GCODE devices
- Add new devices with the Arduino library 'BowlerCom'. See the DyIO example for using it.
- Game Controllers that have OS support (Xbox and generic usb game controllers tested working)
- Hokuyo LIDAR's
- Arduino Firmata devices (unstable because firmata is unstable)
- New UDP and HID connected devices are coming soon, but can be used by extending BowlerStudio with Maven dependencies for scripts like this one: https://gist.github.com/madhephaestus/5cfce02a38ce9b3a67f554121fcab5a8

justinclift, 2018-04-28:
Thanks @madhephaestus.  That looks like it'll point me in the right direction. :smile:

As a side thought, since the wiki link for this repo appears to be non-functional... maybe turn that off? 
(it's in the repo Settings area somewhere).  Altenatively, maybe create a wiki front page which links to http://commonwealthrobotics.com/tutorials? (probably more optimal)

closed 2018-04-28 by justinclift

madhephaestus, 2018-04-28:
Good suggestion, i made a landing page on the wiki