CommonWealthRobotics/BowlerStudio #59: Memory Leak error

alextac98, 2019-02-17:
### Installed Version: 0.29.2

### OS type and version number: Ubuntu 16.04

The program randomly shut down, I was in debug mode so I was able to get the location of the log file. I have attached it below
[hs_err_pid686.log](https://github.com/CommonWealthRobotics/BowlerStudio/files/2873251/hs_err_pid686.log)

Octogonapus, 2019-02-17:
Not sure why this happened. In the log I see lots of jog widget commands and a cad thread. Were you working with a lot of cad objects? How long was BowlerStudio running? If your system has more free memory, it might be worth giving BowlerStudio more.

alextac98, 2019-02-17:
It was a lot of CAD objects, I know there was a memory leak somewhere, thought this might help fix it?

Octogonapus, 2019-02-17:
Well, @madhephaestus can provide their input on this, I'm not familiar with any memory leaks in BowlerStudio.

madhephaestus, 2019-02-24:
can you send me the script that caused the error? Can you run JVisualVM on your system and look at BowlerStudio while the error occurs?

alextac98, 2019-04-28:
After talking to @Octogonapus, it seems like it was due not having enough memory when running Bowler Studio.

closed 2019-04-28 by alextac98