CommonWealthRobotics/BowlerStudio #65: Devices may not be fully disconnected

Octogonapus, 2019-12-10:
If a device throws an uncaught exception in its event callback, it will be disconnected "in code" but not in the UI (the `PhysicsManagerExample` prints its disconnection message to the terminal, but I still need to go to the `My Devices` tab and disconnect it there as well).

madhephaestus, 2019-12-27:
This bug is a concurrency issue and one that should be only reachable if you disconnect immediately after connecting, but before the platform.runlater() manages to run adding the device to the UI. 

This is a user level bug, not something that can be fixed in the IDE.

closed 2019-12-27 by madhephaestus

Octogonapus, 2019-12-27:
So you don't want to do anything for devices that throw immediately after connecting? I think this falls in the quality of life category, even though it is a user-created problem.

madhephaestus, 2019-12-28:
if it had thrown the exception, it wouldnt have been added, that example caught an exception inside the event callback and called disconnect(). This happened before the device had been added to the UI, since the device service thread ran faster than the UI update thread. Its a race condition, and the mistake was in catching that exception and silently disconnecting.