CommonWealthRobotics/BowlerStudio #68: UI freezes randomly and locks up the application

madhephaestus, 2019-12-20:
### Installed Version 
All known version

### OS type and version number
Linux and windows observed

### Expected behavior
The UI should never lock up to the point where the close button is unavailable

### Actual Behavior
VERY rarely the UI will freeze and there is no way to save state or make a clean exit. It needs to be killed from the OS. 

### Steps to reproduce the behavior

This is the hard part, i can not reliably recreate the bug although it seems to have something to do with the Affines on the links getting updated. When a creature fails IK over and over, it sometimes will lock up, sometimes not. 

When it locks up the syserr prints look like this:

`
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
`

Octogonapus, 2019-12-20:
To reproduce, it feels like spamming cad regeneration increases the probability of running into this bug.

I couldn't fix it in BowlerBuilder but I did find a workaround: remove everything from the scene graph, add them into a new one, and replace the current scene graph with the new one. BowlerBuilder identifies the "explosion" here by looking for a specific stacktrace: https://github.com/CommonWealthRobotics/BowlerBuilder/blob/a1ca2012412a66688c25405bf05f493b783d5ccd/BowlerBuilder/KotlinUI/src/main/kotlin/com/neuronrobotics/bowlerbuilder/BowlerBuilder.kt#L36. It posts an event to the main ui event bus which causes this method to run: https://github.com/CommonWealthRobotics/BowlerBuilder/blob/a1ca2012412a66688c25405bf05f493b783d5ccd/BowlerBuilder/KotlinUI/src/main/kotlin/com/neuronrobotics/bowlerbuilder/view/main/MainWindowView.kt#L203. It makes a new cad view, copies all the CSGs from the old cad view into the new one, and replaces the old one with the new one. The method `replacewith` is from TornadoFX and it does some fancy stuff, source is here: https://github.com/edvin/tornadofx/blob/7841a92eb491a8ec9e9f97a53d143e2995bc0d92/src/main/java/tornadofx/Nodes.kt#L892.

madhephaestus, 2019-12-21:
When running the code:

```
while(!Thread.interrupted()) {
	for (def limb:cat.getAllDHChains()) {
		try {
			limb.setDesiredTaskSpaceTransform(limb.getCurrentPoseTarget().translateX(1), 0)
		}catch(Exception ex) {}
	}
}
```

I can reliably produce:

```

Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
	at com.sun.scenario.animation.AbstractMasterTimer.timePulseImpl(AbstractMasterTimer.java:344)
	at com.sun.scenario.animation.AbstractMasterTimer$MainLoop.run(AbstractMasterTimer.java:267)
	at com.sun.javafx.tk.quantum.QuantumToolkit.pulse(QuantumToolkit.java:514)
	at com.sun.javafx.tk.quantum.QuantumToolkit.pulse(QuantumToolkit.java:498)
	at com.sun.javafx.tk.quantum.QuantumToolkit.pulseFromQueue(QuantumToolkit.java:491)
	at com.sun.javafx.tk.quantum.QuantumToolkit.lambda$runToolkit$11(QuantumToolkit.java:319)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)
```

madhephaestus, 2019-12-21:
I have tracked the source of the error to 

BowlerStudioController.highlightExceptionLocal(File fileEngineRunByName, Throwable ex)

It kicks off a thread that accesses the UI, but doesnt wait in case another thread of the same function source calls an exception highlight. One and only one can be called at a time. because the UI cant handle highliting more than one exception at once.

madhephaestus, 2019-12-21:
With more robust testing this is still not resolved yet.

madhephaestus, 2019-12-21:
The exception can be caught by adding:
```
    	Platform.runLater(() ->{
			Thread.currentThread().setUncaughtExceptionHandler(new UncaughtExceptionHandler() {
				@Override
				public void uncaughtException(Thread t, Throwable e) {
					e.printStackTrace();
					new RuntimeException("Caught the UI exception!").printStackTrace();
					//creatureLab3dDockNode.undock();
					//creatureLab3dDockNode.close();

				}
			});
			
		});
```

madhephaestus, 2019-12-21:
Upon closer inspection the error only happens when i interact with the mobile base from the jog widget while the script is talking to the mobile base as well. If i just run the script then the UI will never reach its error state, but if the home button on the jog widget is pressed while the script is moving the limbs, then the error happens.

madhephaestus, 2019-12-22:
This issue is definitely in the jog widget, the error is only reproducible when you click the Home button repeatedly while the script is running. The other jog widget buttons do not effect this issue. It is clearly some action triggered from the home button press in the jogWidget.

madhephaestus, 2019-12-22:
After updating the test script i determined that i can cause the error without  the jog widget. Error happens when the joints are all set in 2 threads and collide. The class where the function resides is synchronized on the class, so all joint updates should be thread synchronized, so it is unclear how this can be causing a conflict. 

```
new Thread({
		while(run){
			Thread.sleep(200);
			for (def limb:cat.getAllDHChains()) {
				homeLimb(limb)
			}
		}
				
	}).start()
while(!Thread.interrupted()) {
	//Thread.sleep(1);
	
	for (def limb:cat.getAllDHChains()) {
		try {
			limb.setDesiredTaskSpaceTransform(limb.getCurrentPoseTarget().translateX(1), 0)
		}catch(Exception ex) {
			//ex.printStackTrace()
		}
	}
}
```

madhephaestus, 2019-12-22:
By commenting out setSelectedCsg(List<CSG> selectedCsg), the issue but doesn't present any more.

madhephaestus, 2019-12-22:
The issue was with a colliding use of the FXTimer 					FxTimer.runLater(java.time.Duration.ofMillis(20)

The instance of the timer were started in close proximity and caused an error (not sure how)

Removal of hte FXTimer calls resolved this issue.

commit 663816c3 (2019-12-22): Close #68

The issue that caused this issue came from the use of FXTimer in the
group select of CSG's. Now a simple highlight of all CSG's is working.
closed 2019-12-22 by madhephaestus