CommonWealthRobotics/BowlerStudio #73: org.eclipse.jgit.lib.internal.BouncyCastleGpgSigner at line 162

Octogonapus, 2019-12-29:
Auto Reported Issue 
OS = Linux amd64 x64
org.eclipse.jgit.api.errors.JGitInternalException: missing credentials provider
	at org.eclipse.jgit.lib.internal.BouncyCastleGpgSigner.sign(BouncyCastleGpgSigner.java:162)
	at org.eclipse.jgit.api.CommitCommand.call(CommitCommand.java:275)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.commit(ScriptingEngine.java:486)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pushCodeToGit(ScriptingEngine.java:519)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pushCodeToGit(ScriptingEngine.java:436)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.save(ConfigurationDatabase.java:68)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$sort$3(BowlerStudioMenuWorkspace.java:173)
	at java.lang.Thread.run(Thread.java:748)
Caused by: org.bouncycastle.openpgp.PGPException: missing credentials provider
	at org.eclipse.jgit.lib.internal.BouncyCastleGpgKeyPassphrasePrompt.getPassphrase(BouncyCastleGpgKeyPassphrasePrompt.java:120)
	at org.eclipse.jgit.lib.internal.BouncyCastleGpgKeyLocator.findSecretKeyForKeyBoxPublicKey(BouncyCastleGpgKeyLocator.java:471)
	at org.eclipse.jgit.lib.internal.BouncyCastleGpgKeyLocator.findSecretKey(BouncyCastleGpgKeyLocator.java:368)
	at org.eclipse.jgit.lib.internal.BouncyCastleGpgSigner.locateSigningKey(BouncyCastleGpgSigner.java:124)
	at org.eclipse.jgit.lib.internal.BouncyCastleGpgSigner.sign(BouncyCastleGpgSigner.java:133)
	... 7 more

madhephaestus, 2019-12-30:
This looks like it tried to save without a valid login. Logout and log back in then close bowler studio to see if the issue is resolved.

Octogonapus, 2019-12-30:
That did not fix it. Here is a log: https://pastebin.com/yr5E8xMh

Here are the actions I took in order:
1. Open BS (and download the new version)
2. After BS was done loading, log out
3. Log in
4. Close BS
5. Open BS (looks like I got logged in properly?)
6. Close BS

madhephaestus, 2019-12-30:
This has something to do with where your store your gpg keys. Have you set up anything non-standard with these?

madhephaestus, 2019-12-30:
I believe you are having this issue: https://bugs.eclipse.org/bugs/show_bug.cgi?id=552091

Octogonapus, 2019-12-30:
I do have `gpgsign` set to `true`.

madhephaestus, 2019-12-30:
Is there anything special about the filenames or paths on this machine? Does turning that signing off resolve the issue?

Octogonapus, 2019-12-31:
> Is there anything special about the filenames or paths on this machine?

No, it's a very recent 18.04 install.

> Does turning that signing off resolve the issue?

Yes. What changed in the recent releases to break this?

madhephaestus, 2019-12-31:
How recent? If it is since i added 2fa, then that makes sense. Also, i updataded the JGIT version to enable SSH cloning/pushing. Ill bet thats the difference, jgit now supports ssh, but is interfering with this somehow...

madhephaestus, 2019-12-31:
https://github.com/CommonWealthRobotics/BowlerStudio/tree/0.36.0 is where the updated JGIT dep showed up.

madhephaestus, 2019-12-31:
Specifically this commit: 

https://github.com/CommonWealthRobotics/bowler-script-kernel/commit/f9ed927bffaab65c595c7bab13cc3d8cbe728bba

madhephaestus, 2019-12-31:
To resolve there may need to be additional logic added to the ssh callback:

https://github.com/CommonWealthRobotics/bowler-script-kernel/blob/f9ed927bffaab65c595c7bab13cc3d8cbe728bba/src/main/java/com/neuronrobotics/bowlerstudio/scripting/SshTransportConfigCallback.java

Or perhaps this is an issue inherent to the newer JGIT library. Its hard to fix when i cant reproduce the error.

Octogonapus, 2019-12-31:
You can't reproduce it even if you run `git config --global commit.gpgsign true`?

madhephaestus, 2020-01-01:
Can you set the global to false and the repos that need signing to true?

Octogonapus, 2020-01-01:
Technically, yes. But I want to sign all my commits.

Can you reproduce it?

Octogonapus, 2020-01-03:
I bisected the versions I had on disk by hand. These are the results:
1.0.7 bad
1.0.6 bad
1.0.5 good
1.0.4 good

So 1.0.6 introduced something that broke it.

madhephaestus, 2020-01-03:
are any other commits working for you? Is the issue just with configurations database?

The issue is in the kernel, and the only changes are to the logic of the configurations database, and some data sanitization on the incoming URL's. Do you see any reason this sanitization would be an issue?

```

	while(remoteURI.endsWith("/"))
		remoteURI=remoteURI.substring(0, remoteURI.length()-2);
    if(!remoteURI.endsWith(".git"))
    	remoteURI=remoteURI+".git";

```

Octogonapus, 2020-01-03:
`remoteURI.substring(0, remoteURI.length()-2);` is suspicious because you cut off an extra character. Can you cut me a build with those additions commented out and something that prints `remoteURI`?

madhephaestus, 2020-01-03:
try rev 1.0.8 and let me know if it worked?

Octogonapus, 2020-01-04:
BS auto-reported https://github.com/CommonWealthRobotics/BowlerStudio/issues/81 when I ran 1.0.8.
Log: https://pastebin.com/nYBLFCB6

madhephaestus, 2020-01-04:
> BS auto-reported #81 when I ran 1.0.8.
> Log: https://pastebin.com/nYBLFCB6

This was a different mistake, resolved now.

Octogonapus, 2020-01-04:
Yep. Same BouncyCastleGpgSigner issue, though.

madhephaestus, 2020-01-04:
Kernel version was 0.50.2 in version 1.0.5 and was 0.51.0 in version 1.0.6. I don't see anything in there to affect the signing.

Octogonapus, 2020-01-16:
Whatever the root cause may be, we can provide a workaround for now so I can still sign my commits by default. When BS clones a repo, it should run jgit's equivalent of `git config commit.gpgsign false` so the global-level commit signing is turned off.

Octogonapus, 2020-03-20:
@madhephaestus Bumping this one :)

Octogonapus, 2020-08-08:
Using this https://github.com/eclipse/jgit/blob/master/org.eclipse.jgit/src/org/eclipse/jgit/lib/Config.java

You may be able to set the boolean `commit.gpgsign` to `false` for each repo that BS clones