CommonWealthRobotics/BowlerStudio #86: org.eclipse.jgit.api.LsRemoteCommand at line 222

madhephaestusdemo, 2020-01-04:
Auto Reported Issue 
BowlerStudio Build 1.0.9
BowlerKernel 0.52.1
JavaCad Version: 0.22.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_232
JavaFX Version: 8.0.232

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.TransportException: https://github.com/madhephaestusdemo/FinalProjectSmallKat.git: Authentication is required but no CredentialsProvider has been registered
	at org.eclipse.jgit.api.LsRemoteCommand.execute(LsRemoteCommand.java:222)
	at org.eclipse.jgit.api.LsRemoteCommand.call(LsRemoteCommand.java:161)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.getAllBranches(ScriptingEngine.java:1490)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$5.lambda$0(BowlerStudioMenu.java:675)
	at java.lang.Thread.run(Thread.java:748)
Caused by: org.eclipse.jgit.errors.TransportException: https://github.com/madhephaestusdemo/FinalProjectSmallKat.git: Authentication is required but no CredentialsProvider has been registered
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:564)
	at org.eclipse.jgit.transport.TransportHttp.openFetch(TransportHttp.java:392)
	at org.eclipse.jgit.api.LsRemoteCommand.execute(LsRemoteCommand.java:200)
	... 4 more

```

madhephaestus, 2020-01-04:
Getting branches from a private repo requires credentials

keionbis, 2020-04-24:
Auto Reported Issue 
BowlerStudio Build 1.5.14
BowlerKernel 0.61.3
JavaCad Version: 0.24.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.TransportException: https://github.com/OperationSmallKat/smallkat_xl_bowler.git: https://github.com/OperationSmallKat/smallkat_xl_bowler.git/info/refs?service=git-upload-pack not found: Not Found
	at org.eclipse.jgit.api.LsRemoteCommand.execute(LsRemoteCommand.java:222)
	at org.eclipse.jgit.api.LsRemoteCommand.call(LsRemoteCommand.java:161)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.getAllBranches(ScriptingEngine.java:1571)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$5.lambda$handle$6(BowlerStudioMenu.java:711)
	at java.lang.Thread.run(Thread.java:748)
Caused by: org.eclipse.jgit.errors.NoRemoteRepositoryException: https://github.com/OperationSmallKat/smallkat_xl_bowler.git: https://github.com/OperationSmallKat/smallkat_xl_bowler.git/info/refs?service=git-upload-pack not found: Not Found
	at org.eclipse.jgit.transport.TransportHttp.createNotFoundException(TransportHttp.java:522)
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:554)
	at org.eclipse.jgit.transport.TransportHttp.openFetch(TransportHttp.java:392)
	at org.eclipse.jgit.api.LsRemoteCommand.execute(LsRemoteCommand.java:200)
	... 4 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:180)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:62)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$5.lambda$handle$6(BowlerStudioMenu.java:718)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:38)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.<clinit>(BowlerStudioMenu.java:128)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.initialize(BowlerStudioModularFrame.java:153)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.reflect.misc.Trampoline.invoke(MethodUtil.java:72)
	at sun.reflect.GeneratedMethodAccessor39.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.reflect.misc.MethodUtil.invoke(MethodUtil.java:276)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2566)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2441)
	at javafx.fxml.FXMLLoader.load(FXMLLoader.java:2409)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.lambda$start$7(BowlerStudio.java:626)
	at java.lang.Thread.run(Thread.java:748)

```

closed 2020-04-24 by madhephaestus

madhephaestus, 2022-09-12:
Auto Reported Issue 
BowlerStudio Build 1.57.11
BowlerKernel 1.52.1
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.TransportException: https://github.com/madhephaestus/BowlerStudioImageAssets.git: cannot open git-upload-pack
	at org.eclipse.jgit.api.LsRemoteCommand.execute(LsRemoteCommand.java:222)
	at org.eclipse.jgit.api.LsRemoteCommand.call(LsRemoteCommand.java:161)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.getAllBranches(ScriptingEngine.java:2068)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1467)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1641)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.filesInGit(ScriptingEngine.java:704)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAllAssets(AssetFactory.java:155)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.setGitSource(AssetFactory.java:150)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:374)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: org.eclipse.jgit.errors.TransportException: https://github.com/madhephaestus/BowlerStudioImageAssets.git: cannot open git-upload-pack
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:626)
	at org.eclipse.jgit.transport.TransportHttp.openFetch(TransportHttp.java:392)
	at org.eclipse.jgit.api.LsRemoteCommand.execute(LsRemoteCommand.java:200)
	... 19 more
Caused by: java.net.ConnectException: Connection refused (Connection refused) github.com
	at org.eclipse.jgit.util.HttpSupport.response(HttpSupport.java:237)
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:540)
	... 21 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1512)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1641)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.filesInGit(ScriptingEngine.java:704)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAllAssets(AssetFactory.java:155)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.setGitSource(AssetFactory.java:150)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:374)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:125)
	at com.neuronrobotics.bowlerstudio.PsudoSplash.setMessage(PsudoSplash.java:193)
	at com.neuronrobotics.bowlerstudio.SplashManager.renderSplashFrame(SplashManager.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.renderSplashFrame(BowlerStudio.java:485)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:256)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```