CommonWealthRobotics/BowlerStudio #100: org.eclipse.jgit.api.PushCommand at line 180

javatechs, 2020-01-07:
Auto Reported Issue 
BowlerStudio Build 1.1.0
BowlerKernel 0.54.0
JavaCad Version: 0.22.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_232
JavaFX Version: 8.0.232

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.TransportException: https://github.com/madhephaestus/BowlerStudioImageAssets.git: git-receive-pack not permitted on 'https://github.com/madhephaestus/BowlerStudioImageAssets.git/'
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:180)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranchLocal(ScriptingEngine.java:833)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:808)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:786)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1071)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: org.eclipse.jgit.errors.TransportException: https://github.com/madhephaestus/BowlerStudioImageAssets.git: git-receive-pack not permitted on 'https://github.com/madhephaestus/BowlerStudioImageAssets.git/'
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:578)
	at org.eclipse.jgit.transport.TransportHttp.openPush(TransportHttp.java:467)
	at org.eclipse.jgit.transport.PushProcess.execute(PushProcess.java:160)
	at org.eclipse.jgit.transport.Transport.push(Transport.java:1368)
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:170)
	... 16 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.lambda$except$0(IssueReportingExceptionHandler.java:54)
	at java.lang.Thread.run(Thread.java:748)

```

Octogonapus, 2020-04-23:
Auto Reported Issue 
BowlerStudio Build 1.5.13
BowlerKernel 0.61.3
JavaCad Version: 0.24.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_232
JavaFX Version: 8.0.232

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.TransportException: https://github.com/madhephaestus/BowlerStudioImageAssets.git: git-receive-pack not permitted on 'https://github.com/madhephaestus/BowlerStudioImageAssets.git/'
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:180)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranchLocal(ScriptingEngine.java:845)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:824)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:801)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1118)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:331)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: org.eclipse.jgit.errors.TransportException: https://github.com/madhephaestus/BowlerStudioImageAssets.git: git-receive-pack not permitted on 'https://github.com/madhephaestus/BowlerStudioImageAssets.git/'
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:578)
	at org.eclipse.jgit.transport.TransportHttp.openPush(TransportHttp.java:467)
	at org.eclipse.jgit.transport.PushProcess.execute(PushProcess.java:160)
	at org.eclipse.jgit.transport.Transport.push(Transport.java:1368)
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:170)
	... 16 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:180)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:62)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1138)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:331)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:38)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:111)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:273)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

madhephaestusdemo, 2021-01-02:
Auto Reported Issue 
BowlerStudio Build 1.17.4
BowlerKernel 0.64.0
JavaCad Version: 0.25.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
org.eclipse.jgit.api.errors.TransportException: https://github.com/madhephaestus/BowlerStudioImageAssets.git: git-receive-pack not permitted on 'https://github.com/madhephaestus/BowlerStudioImageAssets.git/'
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:180)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranchLocal(ScriptingEngine.java:847)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:826)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:803)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1120)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:345)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: org.eclipse.jgit.errors.TransportException: https://github.com/madhephaestus/BowlerStudioImageAssets.git: git-receive-pack not permitted on 'https://github.com/madhephaestus/BowlerStudioImageAssets.git/'
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:578)
	at org.eclipse.jgit.transport.TransportHttp.openPush(TransportHttp.java:467)
	at org.eclipse.jgit.transport.PushProcess.execute(PushProcess.java:160)
	at org.eclipse.jgit.transport.Transport.push(Transport.java:1368)
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:170)
	... 16 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1140)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:345)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:113)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:292)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

madhephaestusdemo, 2021-08-15:
Auto Reported Issue 
BowlerStudio Build 1.31.2
BowlerKernel 0.65.2
JavaCad Version: 0.25.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.TransportException: https://github.com/madhephaestus/BowlerStudioImageAssets.git: git-receive-pack not permitted on 'https://github.com/madhephaestus/BowlerStudioImageAssets.git/'
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:180)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranchLocal(ScriptingEngine.java:853)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:832)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:809)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1149)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1319)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.filesInGit(ScriptingEngine.java:420)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAllAssets(AssetFactory.java:148)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.setGitSource(AssetFactory.java:143)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:391)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: org.eclipse.jgit.errors.TransportException: https://github.com/madhephaestus/BowlerStudioImageAssets.git: git-receive-pack not permitted on 'https://github.com/madhephaestus/BowlerStudioImageAssets.git/'
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:578)
	at org.eclipse.jgit.transport.TransportHttp.openPush(TransportHttp.java:467)
	at org.eclipse.jgit.transport.PushProcess.execute(PushProcess.java:160)
	at org.eclipse.jgit.transport.Transport.push(Transport.java:1368)
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:170)
	... 20 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1167)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1319)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.filesInGit(ScriptingEngine.java:420)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAllAssets(AssetFactory.java:148)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.setGitSource(AssetFactory.java:143)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:391)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:115)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:293)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

Wegault, 2022-03-25:
Auto Reported Issue 
BowlerStudio Build 1.42.2
BowlerKernel 1.41.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.TransportException: https://gist.github.com/a991ca954460c1ba9860.git: git-receive-pack not permitted on 'https://gist.github.com/a991ca954460c1ba9860.git/'
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:180)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranchLocal(ScriptingEngine.java:1172)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1137)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.getBranch(ScriptingEngine.java:1033)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1396)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$1.extracted(BowlerStudioMenu.java:181)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$1.run(BowlerStudioMenu.java:171)
Caused by: org.eclipse.jgit.errors.TransportException: https://gist.github.com/a991ca954460c1ba9860.git: git-receive-pack not permitted on 'https://gist.github.com/a991ca954460c1ba9860.git/'
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:578)
	at org.eclipse.jgit.transport.TransportHttp.openPush(TransportHttp.java:467)
	at org.eclipse.jgit.transport.PushProcess.execute(PushProcess.java:160)
	at org.eclipse.jgit.transport.Transport.push(Transport.java:1368)
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:170)
	... 6 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$1.run(BowlerStudioMenu.java:173)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.<clinit>(BowlerStudioMenu.java:129)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.initialize(BowlerStudioModularFrame.java:153)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.reflect.misc.Trampoline.invoke(MethodUtil.java:72)
	at sun.reflect.GeneratedMethodAccessor29.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.reflect.misc.MethodUtil.invoke(MethodUtil.java:276)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2566)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2441)
	at javafx.fxml.FXMLLoader.load(FXMLLoader.java:2409)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.lambda$start$8(BowlerStudio.java:665)
	at java.lang.Thread.run(Thread.java:748)

```

closed 2022-04-14 by madhephaestus

FlannelAxis, 2023-03-11:
Auto Reported Issue 
BowlerStudio Build 1.76.0
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.TransportException: https://github.com/TechnocopiaPlant/TendyTheTankEngine.git: git-receive-pack not permitted on 'https://github.com/TechnocopiaPlant/TendyTheTankEngine.git/'
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:180)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pushCodeToGit(ScriptingEngine.java:884)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pushCodeToGit(ScriptingEngine.java:742)
	at com.neuronrobotics.bowlerstudio.AddFileToGistController.lambda$onAddFile$3(AddFileToGistController.java:187)
	at java.lang.Thread.run(Thread.java:748)
Caused by: org.eclipse.jgit.errors.TransportException: https://github.com/TechnocopiaPlant/TendyTheTankEngine.git: git-receive-pack not permitted on 'https://github.com/TechnocopiaPlant/TendyTheTankEngine.git/'
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:578)
	at org.eclipse.jgit.transport.TransportHttp.openPush(TransportHttp.java:467)
	at org.eclipse.jgit.transport.PushProcess.execute(PushProcess.java:160)
	at org.eclipse.jgit.transport.Transport.push(Transport.java:1368)
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:170)
	... 4 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.AddFileToGistController.lambda$onAddFile$3(AddFileToGistController.java:194)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.AddFileToGistController.lambda$onAddFile$3(AddFileToGistController.java:194)
	at java.lang.Thread.run(Thread.java:748)

```

commit 735fd511 (2020-01-07): Resolve #100

The issue order of operations for users was not throwing an exception
but rather producing an exception