CommonWealthRobotics/BowlerStudio #104: javafx.embed.swing.SwingNode at line 364

javatechs, 2020-01-07:
Auto Reported Issue 
BowlerStudio Build 1.1.0
BowlerKernel 0.54.0
JavaCad Version: 0.22.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_232
JavaFX Version: 8.0.232

OS = Linux amd64 x64
```
java.lang.NullPointerException
	at javafx.embed.swing.SwingNode.lambda$setImageBuffer$7(SwingNode.java:364)
	at javafx.embed.swing.SwingNode.lambda$setImageBuffer$8(SwingNode.java:371)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.lambda$except$0(IssueReportingExceptionHandler.java:54)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2020-01-07:
This is presumably some part of private class MySwingNode inside of LocalFileScriptTab but since the stack trace originates outside of my code, i am not sure where to begin. The best idea i have is to overload SwingNode.setImageBuffer(), catch the exception and move on.

madhephaestus, 2020-01-08:
closed with release 1.2.2

closed 2020-01-08 by madhephaestus