CommonWealthRobotics/BowlerStudio #130: Support editing vitamin property names

Octogonapus, 2020-01-23:
If I create a vitamin property and then want to change its name while still in the new vitamin dialog, I should be able to change it.

madhephaestus, 2020-01-27:
This is a good idea, but is complex to implement. Once a key is defined, it is added to the cad generator, so changing it is only possible for new sizes of a new vitamin. This is an edge case and no common, and very dangerous since it can render old code non-functional. It may be best to not do this.

Octogonapus, 2020-01-27:
> Once a key is defined, it is added to the cad generator

Don't modify anything external like that until I am done creating the vitamin.

madhephaestus, 2020-01-27:
sure, but having those fields editable should only happen for the brand new one, not for modifying existing types. You can add new fields, and it will add that field to all instances, but editing might conflict with the code. This is only valid for one of the many use cases and so its stateful and a bit dangerous, and also quite a bit of effort for this narrow edge case.

Octogonapus, 2020-01-27:
Any time I add a field, that field should not _actually_ be added until I confirm my changes. Do you disagree?