CommonWealthRobotics/BowlerStudio #138: Build 1.3.1 org.codehaus.groovy.control.ErrorCollector at line 309

madhephaestus, 2020-02-03:
Auto Reported Issue 
BowlerStudio Build 1.3.1
BowlerKernel 0.57.1
JavaCad Version: 0.22.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
org.codehaus.groovy.control.MultipleCompilationErrorsException: startup failed:
General error during conversion: Error grabbing Grapes -- [unresolved dependency: com.neuronrobotics#SimplePacketComsJava;0.12.0: not found]

java.lang.RuntimeException: Error grabbing Grapes -- [unresolved dependency: com.neuronrobotics#SimplePacketComsJava;0.12.0: not found]
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at org.codehaus.groovy.reflection.CachedConstructor.invoke(CachedConstructor.java:77)
	at org.codehaus.groovy.reflection.CachedConstructor.doConstructorInvoke(CachedConstructor.java:71)
	at org.codehaus.groovy.runtime.callsite.ConstructorSite$ConstructorSiteNoUnwrap.callConstructor(ConstructorSite.java:81)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCallConstructor(CallSiteArray.java:57)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callConstructor(AbstractCallSite.java:182)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callConstructor(AbstractCallSite.java:190)
	at groovy.grape.GrapeIvy.getDependencies(GrapeIvy.groovy:421)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.codehaus.groovy.runtime.callsite.PogoMetaMethodSite$PogoCachedMethodSite.invoke(PogoMetaMethodSite.java:166)
	at org.codehaus.groovy.runtime.callsite.PogoMetaMethodSite.callCurrent(PogoMetaMethodSite.java:56)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCallCurrent(CallSiteArray.java:49)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callCurrent(AbstractCallSite.java:133)
	at groovy.grape.GrapeIvy.resolve(GrapeIvy.groovy:566)
	at groovy.grape.GrapeIvy$resolve$3.callCurrent(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCallCurrent(CallSiteArray.java:49)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callCurrent(AbstractCallSite.java:133)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callCurrent(AbstractCallSite.java:153)
	at groovy.grape.GrapeIvy.resolve(GrapeIvy.groovy:535)
	at groovy.grape.GrapeIvy$resolve$2.callCurrent(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCallCurrent(CallSiteArray.java:49)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callCurrent(AbstractCallSite.java:133)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callCurrent(AbstractCallSite.java:149)
	at groovy.grape.GrapeIvy.grab(GrapeIvy.groovy:252)
	at groovy.grape.Grape.grab(Grape.java:163)
	at groovy.grape.GrabAnnotationTransformation.visit(GrabAnnotationTransformation.java:358)
	at org.codehaus.groovy.transform.ASTTransformationVisitor$3.call(ASTTransformationVisitor.java:319)
	at org.codehaus.groovy.control.CompilationUnit.applyToSourceUnits(CompilationUnit.java:923)
	at org.codehaus.groovy.control.CompilationUnit.doPhaseOperation(CompilationUnit.java:585)
	at org.codehaus.groovy.control.CompilationUnit.processPhaseOperations(CompilationUnit.java:561)
	at org.codehaus.groovy.control.CompilationUnit.compile(CompilationUnit.java:538)
	at groovy.lang.GroovyClassLoader.doParseClass(GroovyClassLoader.java:286)
	at groovy.lang.GroovyClassLoader.parseClass(GroovyClassLoader.java:259)
	at groovy.lang.GroovyShell.parseClass(GroovyShell.java:674)
	at groovy.lang.GroovyShell.parse(GroovyShell.java:686)
	at groovy.lang.GroovyShell.parse(GroovyShell.java:695)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:57)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:178)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.gitScriptRun(ScriptingEngine.java:640)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$gitScriptRun.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:124)
	at loadAll.run(loadAll.groovy:13)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:61)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:178)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.gitScriptRun(ScriptingEngine.java:640)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$gitScriptRun.call(Unknown Source)
[... 88 lines cut ...]
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:36)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.<clinit>(BowlerStudio.java:81)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:348)
	at com.sun.javafx.application.LauncherImpl.lambda$launchApplicationWithArgs$2(LauncherImpl.java:352)
	at com.sun.javafx.application.PlatformImpl.lambda$runAndWait$7(PlatformImpl.java:326)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$3(WinApplication.java:177)
	at java.lang.Thread.run(Thread.java:748)

```

keionbis, 2020-02-03:
Auto Reported Issue 
BowlerStudio Build 1.4.0
BowlerKernel 0.58.0
JavaCad Version: 0.22.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_232
JavaFX Version: 8.0.232

OS = Linux amd64 x64
```
org.codehaus.groovy.control.MultipleCompilationErrorsException: startup failed:
General error during conversion: Error grabbing Grapes -- [unresolved dependency: com.neuronrobotics#SimplePacketComsJava;0.12.0: not found]

java.lang.RuntimeException: Error grabbing Grapes -- [unresolved dependency: com.neuronrobotics#SimplePacketComsJava;0.12.0: not found]
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at org.codehaus.groovy.reflection.CachedConstructor.invoke(CachedConstructor.java:77)
	at org.codehaus.groovy.reflection.CachedConstructor.doConstructorInvoke(CachedConstructor.java:71)
	at org.codehaus.groovy.runtime.callsite.ConstructorSite$ConstructorSiteNoUnwrap.callConstructor(ConstructorSite.java:81)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCallConstructor(CallSiteArray.java:57)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callConstructor(AbstractCallSite.java:182)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callConstructor(AbstractCallSite.java:190)
	at groovy.grape.GrapeIvy.getDependencies(GrapeIvy.groovy:421)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.codehaus.groovy.runtime.callsite.PogoMetaMethodSite$PogoCachedMethodSite.invoke(PogoMetaMethodSite.java:166)
	at org.codehaus.groovy.runtime.callsite.PogoMetaMethodSite.callCurrent(PogoMetaMethodSite.java:56)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCallCurrent(CallSiteArray.java:49)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callCurrent(AbstractCallSite.java:133)
	at groovy.grape.GrapeIvy.resolve(GrapeIvy.groovy:566)
	at groovy.grape.GrapeIvy$resolve$3.callCurrent(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCallCurrent(CallSiteArray.java:49)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callCurrent(AbstractCallSite.java:133)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callCurrent(AbstractCallSite.java:153)
	at groovy.grape.GrapeIvy.resolve(GrapeIvy.groovy:535)
	at groovy.grape.GrapeIvy$resolve$2.callCurrent(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCallCurrent(CallSiteArray.java:49)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callCurrent(AbstractCallSite.java:133)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callCurrent(AbstractCallSite.java:149)
	at groovy.grape.GrapeIvy.grab(GrapeIvy.groovy:252)
	at groovy.grape.Grape.grab(Grape.java:163)
	at groovy.grape.GrabAnnotationTransformation.visit(GrabAnnotationTransformation.java:358)
	at org.codehaus.groovy.transform.ASTTransformationVisitor$3.call(ASTTransformationVisitor.java:319)
	at org.codehaus.groovy.control.CompilationUnit.applyToSourceUnits(CompilationUnit.java:923)
	at org.codehaus.groovy.control.CompilationUnit.doPhaseOperation(CompilationUnit.java:585)
	at org.codehaus.groovy.control.CompilationUnit.processPhaseOperations(CompilationUnit.java:561)
	at org.codehaus.groovy.control.CompilationUnit.compile(CompilationUnit.java:538)
	at groovy.lang.GroovyClassLoader.doParseClass(GroovyClassLoader.java:286)
	at groovy.lang.GroovyClassLoader.parseClass(GroovyClassLoader.java:259)
	at groovy.lang.GroovyShell.parseClass(GroovyShell.java:674)
	at groovy.lang.GroovyShell.parse(GroovyShell.java:686)
	at groovy.lang.GroovyShell.parse(GroovyShell.java:695)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:57)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:178)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.gitScriptRun(ScriptingEngine.java:642)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$gitScriptRun.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:124)
	at loadAll.run(loadAll.groovy:13)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:61)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:178)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.gitScriptRun(ScriptingEngine.java:642)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:459)
[... 70 lines cut ...]
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
[... cut ...]
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2020-02-03:
resolved in the script : https://github.com/OperationSmallKat/SmallKatLinkLoader/commit/d48e280a3fac33a2b711bcd8022788f737a0bdc8

madhephaestus, 2020-02-04:
the solution is to add

```
@GrabResolver(name='mvnRepository', root='https://repo1.maven.org/maven2/')
```

to the groovy script

JansenSmith, 2023-02-24:
Auto Reported Issue 
BowlerStudio Build source
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
org.codehaus.groovy.control.MultipleCompilationErrorsException: startup failed:
C:\Users\lenovo\bowler-workspace\gitcache\github.com\madhephaestus\CSGSlicer\slice.groovy: 9: unable to resolve class com.neuronrobotics.bowlerstudio.threed.Line3D
 @ line 9, column 1.
   import com.neuronrobotics.bowlerstudio.threed.Line3D;
   ^

1 error

	at org.codehaus.groovy.control.ErrorCollector.failIfErrors(ErrorCollector.java:309)
	at org.codehaus.groovy.control.CompilationUnit.applyToSourceUnits(CompilationUnit.java:938)
	at org.codehaus.groovy.control.CompilationUnit.doPhaseOperation(CompilationUnit.java:585)
	at org.codehaus.groovy.control.CompilationUnit.compile(CompilationUnit.java:534)
	at groovy.lang.GroovyClassLoader.doParseClass(GroovyClassLoader.java:286)
	at groovy.lang.GroovyClassLoader.parseClass(GroovyClassLoader.java:259)
	at groovy.lang.GroovyShell.parseClass(GroovyShell.java:674)
	at groovy.lang.GroovyShell.parse(GroovyShell.java:686)
	at groovy.lang.GroovyShell.parse(GroovyShell.java:695)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:57)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:81)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:423)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.gitScriptRun(ScriptingEngine.java:944)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$gitScriptRun.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:124)
	at hotfix.run(hotfix.groovy:8)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:61)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:81)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:423)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.gitScriptRun(ScriptingEngine.java:944)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:478)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:210)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:484)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.<clinit>(BowlerStudio.java:74)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:348)
	at com.sun.javafx.application.LauncherImpl.lambda$launchApplicationWithArgs$2(LauncherImpl.java:352)
	at com.sun.javafx.application.PlatformImpl.lambda$runAndWait$7(PlatformImpl.java:326)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$3(WinApplication.java:177)
	at java.lang.Thread.run(Thread.java:748)

```