CommonWealthRobotics/BowlerStudio #151: Build 1.4.0 java.util.Arrays at line 3664

jdalmeida, 2020-02-08:
Auto Reported Issue 
BowlerStudio Build 1.4.0
BowlerKernel 0.58.0
JavaCad Version: 0.22.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.OutOfMemoryError: Java heap space
	at java.util.Arrays.copyOfRange(Arrays.java:3664)
	at java.lang.String.<init>(String.java:207)
	at java.lang.StringBuilder.toString(StringBuilder.java:407)
	at eu.mihosoft.vrl.v3d.CSG.toStlString(CSG.java:1219)
	at com.neuronrobotics.bowlerstudio.creature.CadFileExporter.makeStl(CadFileExporter.java:119)
	at com.neuronrobotics.bowlerstudio.creature.CadFileExporter.generateManufacturingParts(CadFileExporter.java:87)
	at com.neuronrobotics.bowlerstudio.threed.BowlerStudio3dEngine$1.run(BowlerStudio3dEngine.java:344)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:156)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:54)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:73)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:36)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:65)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:64)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:130)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:340)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:143)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1234)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2020-02-08:
it looks like you ran out of ram, try running on a computer with more ram?

madhephaestus, 2020-03-21:
I found and fixed a memory leak, try this again with the latest version please?

closed 2020-03-21 by madhephaestus