CommonWealthRobotics/BowlerStudio #163: Build 1.4.4 org.eclipse.jgit.api.PullCommand at line 245

madhephaestus, 2020-02-19:
Auto Reported Issue 
BowlerStudio Build 1.4.4
BowlerKernel 0.59.1
JavaCad Version: 0.22.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_221
JavaFX Version: 8.0.221

OS = Windows 10 amd64 x64
```
org.eclipse.jgit.api.errors.NoHeadException: Cannot check out from unborn branch
	at org.eclipse.jgit.api.PullCommand.call(PullCommand.java:245)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:962)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1024)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$pull$0.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:116)
	at loadAll.run(loadAll.groovy:12)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:61)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:179)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.gitScriptRun(ScriptingEngine.java:643)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:459)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:156)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:54)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:996)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1024)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$pull$0.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:116)
	at loadAll.run(loadAll.groovy:12)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:61)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:179)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.gitScriptRun(ScriptingEngine.java:643)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:459)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:36)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:114)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:282)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)

```

madhephaestus, 2020-02-19:
Auto Reported Issue 
BowlerStudio Build 1.4.4
BowlerKernel 0.59.1
JavaCad Version: 0.22.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_221
JavaFX Version: 8.0.221

OS = Windows 10 amd64 x64
```
org.eclipse.jgit.api.errors.NoHeadException: Cannot check out from unborn branch
	at org.eclipse.jgit.api.PullCommand.call(PullCommand.java:245)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:962)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1024)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$loginEvent$0(BowlerStudioMenuWorkspace.java:41)
	at java.lang.Thread.run(Unknown Source)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:156)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:54)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:996)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1024)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$loginEvent$0(BowlerStudioMenuWorkspace.java:41)
	at java.lang.Thread.run(Unknown Source)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:36)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:114)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:282)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)

```

madhephaestus, 2020-02-19:
Auto Reported Issue 
BowlerStudio Build 1.4.4
BowlerKernel 0.59.1
JavaCad Version: 0.22.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_221
JavaFX Version: 8.0.221

OS = Windows 10 amd64 x64
```
org.eclipse.jgit.api.errors.NoHeadException: Cannot check out from unborn branch
	at org.eclipse.jgit.api.PullCommand.call(PullCommand.java:245)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:962)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1024)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$pull$0.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:116)
	at loadAll.run(loadAll.groovy:12)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:61)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:179)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.gitScriptRun(ScriptingEngine.java:643)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:459)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:156)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:54)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:996)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1024)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$pull$0.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:116)
	at loadAll.run(loadAll.groovy:12)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:61)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:179)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.gitScriptRun(ScriptingEngine.java:643)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:459)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:36)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:114)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:282)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)

```

madhephaestus, 2020-02-27:
Auto Reported Issue 
BowlerStudio Build 1.4.4
BowlerKernel 0.59.1
JavaCad Version: 0.22.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_221
JavaFX Version: 8.0.221

OS = Windows 10 amd64 x64
```
org.eclipse.jgit.api.errors.NoHeadException: Cannot check out from unborn branch
	at org.eclipse.jgit.api.PullCommand.call(PullCommand.java:245)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:962)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:974)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$3$3.run(BowlerStudioMenu.java:433)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:156)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:54)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:996)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:974)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$3$3.run(BowlerStudioMenu.java:433)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:36)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:114)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:282)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)

```

madhephaestus, 2020-03-04:
Auto Reported Issue 
BowlerStudio Build 1.4.4
BowlerKernel 0.59.1
JavaCad Version: 0.22.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_221
JavaFX Version: 8.0.221

OS = Windows 10 amd64 x64
```
org.eclipse.jgit.api.errors.NoHeadException: Cannot check out from unborn branch
	at org.eclipse.jgit.api.PullCommand.call(PullCommand.java:245)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:962)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:974)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1024)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$loginEvent$0(BowlerStudioMenuWorkspace.java:41)
	at java.lang.Thread.run(Unknown Source)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:156)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:54)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:996)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:974)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1024)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$loginEvent$0(BowlerStudioMenuWorkspace.java:41)
	at java.lang.Thread.run(Unknown Source)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:36)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:114)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:282)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)

```

madhephaestus, 2020-03-19:
Auto Reported Issue 
BowlerStudio Build 1.5.0
BowlerKernel 0.61.0
JavaCad Version: 0.23.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
org.eclipse.jgit.api.errors.NoHeadException: Cannot check out from unborn branch
	at org.eclipse.jgit.api.PullCommand.call(PullCommand.java:245)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:974)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:697)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:659)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.loadFile(ConfigurationDatabase.java:105)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getDatabase(ConfigurationDatabase.java:92)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.setGitSource(ConfigurationDatabase.java:164)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.setRepo(ConfigurationDatabase.java:153)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getGitSource(ConfigurationDatabase.java:125)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.loadFile(ConfigurationDatabase.java:105)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getDatabase(ConfigurationDatabase.java:92)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getParamMap(ConfigurationDatabase.java:45)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getObject(ConfigurationDatabase.java:37)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:304)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:156)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:54)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1008)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:697)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:659)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.loadFile(ConfigurationDatabase.java:105)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getDatabase(ConfigurationDatabase.java:92)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.setGitSource(ConfigurationDatabase.java:164)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.setRepo(ConfigurationDatabase.java:153)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getGitSource(ConfigurationDatabase.java:125)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.loadFile(ConfigurationDatabase.java:105)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getDatabase(ConfigurationDatabase.java:92)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getParamMap(ConfigurationDatabase.java:45)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getObject(ConfigurationDatabase.java:37)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:304)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:36)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:114)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:282)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[... cut ...]
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

deschopa, 2020-03-27:
Auto Reported Issue 
BowlerStudio Build 1.5.5
BowlerKernel 0.61.1
JavaCad Version: 0.23.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
org.eclipse.jgit.api.errors.NoHeadException: Cannot check out from unborn branch
	at org.eclipse.jgit.api.PullCommand.call(PullCommand.java:245)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:974)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:697)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:659)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.loadFile(ConfigurationDatabase.java:105)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getDatabase(ConfigurationDatabase.java:92)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.setGitSource(ConfigurationDatabase.java:164)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.setRepo(ConfigurationDatabase.java:153)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getGitSource(ConfigurationDatabase.java:125)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.loadFile(ConfigurationDatabase.java:105)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getDatabase(ConfigurationDatabase.java:92)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getParamMap(ConfigurationDatabase.java:45)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getObject(ConfigurationDatabase.java:37)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:304)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:155)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:58)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1008)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:697)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:659)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.loadFile(ConfigurationDatabase.java:105)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getDatabase(ConfigurationDatabase.java:92)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.setGitSource(ConfigurationDatabase.java:164)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.setRepo(ConfigurationDatabase.java:153)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getGitSource(ConfigurationDatabase.java:125)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.loadFile(ConfigurationDatabase.java:105)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getDatabase(ConfigurationDatabase.java:92)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getParamMap(ConfigurationDatabase.java:45)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getObject(ConfigurationDatabase.java:37)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:304)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:37)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:114)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:282)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[... cut ...]
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

madhephaestus, 2020-03-27:
Resolved in 1.5.6, close and re-open to get latest version

commit ba695d8e (2020-03-27): close #163
closed 2020-03-27 by madhephaestus