CommonWealthRobotics/BowlerStudio #175: Build 1.5.0 org.eclipse.jgit.lib.ObjectId at line 289

Octogonapus, 2020-03-15:
Auto Reported Issue 
BowlerStudio Build 1.5.0
BowlerKernel 0.61.0
JavaCad Version: 0.23.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_232
JavaFX Version: 8.0.232

OS = Linux amd64 x64
```
java.lang.NullPointerException
	at org.eclipse.jgit.lib.ObjectId.<init>(ObjectId.java:289)
	at org.eclipse.jgit.lib.ObjectIdOwnerMap$Entry.<init>(ObjectIdOwnerMap.java:368)
	at org.eclipse.jgit.revwalk.RevObject.<init>(RevObject.java:64)
	at org.eclipse.jgit.revwalk.RevCommit.<init>(RevCommit.java:153)
	at org.eclipse.jgit.revwalk.RevWalk.createCommit(RevWalk.java:1507)
	at org.eclipse.jgit.revwalk.RevWalk.lookupCommit(RevWalk.java:754)
	at org.eclipse.jgit.api.LogCommand.add(LogCommand.java:373)
	at org.eclipse.jgit.api.LogCommand.add(LogCommand.java:210)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$4.lambda$handle$6(BowlerStudioMenu.java:523)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:156)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:54)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:36)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:130)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:340)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:143)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1239)
	at java.lang.Thread.run(Thread.java:748)

```

Octogonapus, 2020-03-15:
Might be an issue with having commit signing turned on (I forgot to turn it off when I opened BS).

madhephaestusdemo, 2020-05-20:
Auto Reported Issue 
BowlerStudio Build 1.8.1
BowlerKernel 0.62.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.NullPointerException
	at org.eclipse.jgit.lib.ObjectId.<init>(ObjectId.java:289)
	at org.eclipse.jgit.lib.ObjectIdOwnerMap$Entry.<init>(ObjectIdOwnerMap.java:368)
	at org.eclipse.jgit.revwalk.RevObject.<init>(RevObject.java:64)
	at org.eclipse.jgit.revwalk.RevCommit.<init>(RevCommit.java:153)
	at org.eclipse.jgit.revwalk.RevWalk.createCommit(RevWalk.java:1507)
	at org.eclipse.jgit.revwalk.RevWalk.lookupCommit(RevWalk.java:754)
	at org.eclipse.jgit.api.LogCommand.add(LogCommand.java:373)
	at org.eclipse.jgit.api.LogCommand.add(LogCommand.java:210)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$4.lambda$handle$6(BowlerStudioMenu.java:525)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$4.lambda$handle$6(BowlerStudioMenu.java:608)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.<clinit>(BowlerStudioMenu.java:128)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.initialize(BowlerStudioModularFrame.java:153)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.reflect.misc.Trampoline.invoke(MethodUtil.java:71)
	at sun.reflect.GeneratedMethodAccessor39.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.reflect.misc.MethodUtil.invoke(MethodUtil.java:275)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2566)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2441)
	at javafx.fxml.FXMLLoader.load(FXMLLoader.java:2409)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.lambda$start$8(BowlerStudio.java:640)
	at java.lang.Thread.run(Thread.java:748)

```

commit 229c28d0 (2020-05-20): close #175 again

The ObjectId returned when repo.resolve() is called can come back with
null if the branch name is unknown.
closed 2020-05-20 by madhephaestus