CommonWealthRobotics/BowlerStudio #176: Build 1.5.0 java.util.ArrayList at line 657

Octogonapus, 2020-03-16:
Auto Reported Issue 
BowlerStudio Build 1.5.0
BowlerKernel 0.61.0
JavaCad Version: 0.23.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_232
JavaFX Version: 8.0.232

OS = Linux amd64 x64
```
java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:657)
	at java.util.ArrayList.get(ArrayList.java:433)
	at com.neuronrobotics.bowlerstudio.physics.MobileBasePhysicsManager$2.run(MobileBasePhysicsManager.java:130)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:156)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:54)
	at com.sun.glass.ui.Application.reportException(Application.java:453)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:36)
	at com.neuronrobotics.bowlerstudio.threed.BowlerStudio3dEngine.lambda$static$0(BowlerStudio3dEngine.java:220)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2020-06-20:
Auto Reported Issue 
BowlerStudio Build 1.9.4
BowlerKernel 0.62.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:657)
	at java.util.ArrayList.get(ArrayList.java:433)
	at com.neuronrobotics.sdk.addons.kinematics.DHParameterKinematics.onJointSpaceUpdate(DHParameterKinematics.java:646)
	at com.neuronrobotics.sdk.addons.kinematics.DHParameterKinematics.lambda$refreshPose$1(DHParameterKinematics.java:815)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.sun.glass.ui.Application.reportException(Application.java:453)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.threed.BowlerStudio3dEngine.lambda$static$0(BowlerStudio3dEngine.java:218)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2020-06-22:
Auto Reported Issue 
BowlerStudio Build 1.9.4
BowlerKernel 0.62.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:657)
	at java.util.ArrayList.get(ArrayList.java:433)
	at com.neuronrobotics.sdk.addons.kinematics.DHParameterKinematics.onJointSpaceUpdate(DHParameterKinematics.java:646)
	at com.neuronrobotics.sdk.addons.kinematics.DHParameterKinematics.lambda$1(DHParameterKinematics.java:815)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.sun.glass.ui.Application.reportException(Application.java:453)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.threed.BowlerStudio3dEngine.lambda$0(BowlerStudio3dEngine.java:218)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2020-07-29:
Auto Reported Issue 
BowlerStudio Build 1.9.6
BowlerKernel 0.62.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:657)
	at java.util.ArrayList.get(ArrayList.java:433)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.loadSingleLink(MobleBaseMenueFactory.java:637)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.loadSingleLimb(MobleBaseMenueFactory.java:867)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.loadLimbs(MobleBaseMenueFactory.java:602)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.load(MobleBaseMenueFactory.java:139)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.finishLoading(CreatureLab.java:162)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.lambda$1(CreatureLab.java:98)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.load(MobleBaseMenueFactory.java:419)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.finishLoading(CreatureLab.java:162)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.lambda$1(CreatureLab.java:98)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.load(MobleBaseMenueFactory.java:419)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.finishLoading(CreatureLab.java:162)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.lambda$1(CreatureLab.java:98)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2022-08-29:
Auto Reported Issue 
BowlerStudio Build 1.55.3
BowlerKernel 1.50.1
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:657)
	at java.util.ArrayList.get(ArrayList.java:433)
	at com.neuronrobotics.sdk.addons.kinematics.DHParameterKinematics.getLinkTip(DHParameterKinematics.java:809)
	at com.neuronrobotics.bowlerstudio.creature.LinkSliderWidget$2.onJointSpaceUpdate(LinkSliderWidget.java:263)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.firePoseUpdate(AbstractKinematicsNR.java:849)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.runRenderWrangler(AbstractKinematicsNR.java:1614)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.setGlobalToFiducialTransform(AbstractKinematicsNR.java:953)
	at com.neuronrobotics.sdk.addons.kinematics.DHParameterKinematics.setGlobalToFiducialTransform(DHParameterKinematics.java:642)
	at com.neuronrobotics.sdk.addons.kinematics.MobileBase.setGlobalToFiducialTransform(MobileBase.java:900)
	at com.neuronrobotics.sdk.addons.kinematics.MobileBase.getEmbedableXml(MobileBase.java:629)
	at com.neuronrobotics.sdk.addons.kinematics.DHLink.getXml(DHLink.java:143)
	at com.neuronrobotics.sdk.addons.kinematics.DHParameterKinematics.getEmbedableXml(DHParameterKinematics.java:469)
	at com.neuronrobotics.sdk.addons.kinematics.MobileBase.makeLimbTag(MobileBase.java:723)
	at com.neuronrobotics.sdk.addons.kinematics.MobileBase.getEmbedableXml(MobileBase.java:674)
	at com.neuronrobotics.sdk.addons.kinematics.MobileBase.getXml(MobileBase.java:613)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.lambda$3(MobleBaseMenueFactory.java:95)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab$1$1.run(CreatureLab.java:176)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:132)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:350)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:157)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:279)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1(BowlerStudioMenu.java:264)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.run(BowlerStudioMenu.java:249)

```

madhephaestus, 2022-08-29:
Auto Reported Issue 
BowlerStudio Build 1.55.3
BowlerKernel 1.50.1
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.IndexOutOfBoundsException: Index: 6, Size: 6
	at java.util.ArrayList.rangeCheck(ArrayList.java:657)
	at java.util.ArrayList.get(ArrayList.java:433)
	at com.neuronrobotics.bowlerstudio.creature.LinkSliderWidget$2.onJointSpaceUpdate(LinkSliderWidget.java:269)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.firePoseUpdate(AbstractKinematicsNR.java:849)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.runRenderWrangler(AbstractKinematicsNR.java:1614)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.setGlobalToFiducialTransform(AbstractKinematicsNR.java:953)
	at com.neuronrobotics.sdk.addons.kinematics.DHParameterKinematics.setGlobalToFiducialTransform(DHParameterKinematics.java:642)
	at com.neuronrobotics.sdk.addons.kinematics.MobileBase$1.onFiducialToGlobalUpdate(MobileBase.java:157)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.setGlobalToFiducialTransform(AbstractKinematicsNR.java:950)
	at com.neuronrobotics.sdk.addons.kinematics.MobileBase.setGlobalToFiducialTransform(MobileBase.java:898)
	at com.neuronrobotics.sdk.addons.kinematics.MobileBase.getEmbedableXml(MobileBase.java:629)
	at com.neuronrobotics.sdk.addons.kinematics.MobileBase.getXml(MobileBase.java:613)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.lambda$3(MobleBaseMenueFactory.java:95)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab$1$1.run(CreatureLab.java:176)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:132)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:350)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:157)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:279)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1(BowlerStudioMenu.java:264)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.run(BowlerStudioMenu.java:249)

```

madhephaestus, 2022-09-12:
Auto Reported Issue 
BowlerStudio Build 1.57.11
BowlerKernel 1.52.1
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.IndexOutOfBoundsException: Index: 6, Size: 6
	at java.util.ArrayList.rangeCheck(ArrayList.java:657)
	at java.util.ArrayList.get(ArrayList.java:433)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.getLinkConfiguration(AbstractKinematicsNR.java:395)
	at com.neuronrobotics.bowlerstudio.creature.LinkSliderWidget.<init>(LinkSliderWidget.java:80)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.lambda$loadSingleLink$27(MobleBaseMenueFactory.java:685)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab$1$1.run(CreatureLab.java:176)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:350)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:157)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:279)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1700(BowlerStudioMenu.java:66)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.run(BowlerStudioMenu.java:249)

```

madhephaestus, 2023-06-19:
Auto Reported Issue 
BowlerStudio Build source
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:657)
	at java.util.ArrayList.get(ArrayList.java:433)
	at com.neuronrobotics.bowlerstudio.vitamins.VitaminBomManager.saveLocal(VitaminBomManager.java:159)
	at com.neuronrobotics.bowlerstudio.vitamins.VitaminBomManager.lambda$1(VitaminBomManager.java:177)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:136)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:350)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:558)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:282)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1(BowlerStudioMenu.java:267)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$0(BowlerStudioMenu.java:250)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2023-06-19:
Auto Reported Issue 
BowlerStudio Build source
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:657)
	at java.util.ArrayList.get(ArrayList.java:433)
	at com.neuronrobotics.bowlerstudio.vitamins.VitaminBomManager.saveLocal(VitaminBomManager.java:150)
	at com.neuronrobotics.bowlerstudio.vitamins.VitaminBomManager.lambda$1(VitaminBomManager.java:178)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:136)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:350)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:558)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:282)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1(BowlerStudioMenu.java:267)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$0(BowlerStudioMenu.java:250)
	at java.lang.Thread.run(Thread.java:748)

```

dromero2025, 2024-04-05:
Auto Reported Issue 
BowlerStudio Build 2.14.7
BowlerKernel 1.59.0
JavaCad Version: 0.31.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:657)
	at java.util.ArrayList.get(ArrayList.java:433)
	at com.neuronrobotics.sdk.addons.kinematics.DHParameterKinematics.getDhLink(DHParameterKinematics.java:797)
	at com.neuronrobotics.sdk.addons.kinematics.DHParameterKinematics.getFollowerMobileBase(DHParameterKinematics.java:841)
	at com.neuronrobotics.bowlerstudio.creature.JogWidget.homeLimb(JogWidget.java:277)
	at com.neuronrobotics.bowlerstudio.creature.JogWidget.lambda$home$2(JogWidget.java:263)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:62)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:53)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:52)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:142)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:352)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:587)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:284)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1400(BowlerStudioMenu.java:71)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$run$0(BowlerStudioMenu.java:252)
	at java.lang.Thread.run(Thread.java:748)

```

JansenSmith, 2024-07-24:
Auto Reported Issue 
BowlerStudio Build 2.36.0
BowlerKernel 1.59.0
JavaCad Version: 1.1.1
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:657)
	at java.util.ArrayList.get(ArrayList.java:433)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$null$1(BowlerStudioMenuWorkspace.java:183)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:62)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:53)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:52)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:141)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:367)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:737)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:287)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1400(BowlerStudioMenu.java:73)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$run$0(BowlerStudioMenu.java:254)
	at java.lang.Thread.run(Thread.java:748)

```

commit 367f6c38 (2024-07-24): close #176
closed 2024-07-24 by madhephaestus