CommonWealthRobotics/BowlerStudio #180: Build 1.4.4 sun.swing.JLightweightFrame at line 276

keionbis, 2020-03-31:
Auto Reported Issue 
BowlerStudio Build 1.4.4
BowlerKernel 0.59.1
JavaCad Version: 0.22.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$628/1175704167.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:188)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at javafx.embed.swing.FXDnD$FXDragSourceContextPeer.startDrag(FXDnD.java:360)
	at sun.awt.dnd.SunDragSourceContextPeer.startDrag(SunDragSourceContextPeer.java:135)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:321)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:426)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:156)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:54)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:188)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at javafx.embed.swing.FXDnD$FXDragSourceContextPeer.startDrag(FXDnD.java:360)
	at sun.awt.dnd.SunDragSourceContextPeer.startDrag(SunDragSourceContextPeer.java:135)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:321)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:426)
	at java.awt.dnd.DragGestureEvent.startDrag(DragGestureEvent.java:263)
	at javax.swing.TransferHandler$DragHandler.dragGestureRecognized(TransferHandler.java:1608)
	at java.awt.dnd.DragGestureRecognizer.fireDragGestureRecognized(DragGestureRecognizer.java:361)
	at javax.swing.TransferHandler$SwingDragGestureRecognizer.gestured(TransferHandler.java:1669)
	at javax.swing.TransferHandler.exportAsDrag(TransferHandler.java:744)
	at javax.swing.plaf.basic.DragRecognitionSupport.mouseDraggedImpl(DragRecognitionSupport.java:174)
	at javax.swing.plaf.basic.DragRecognitionSupport.mouseDragged(DragRecognitionSupport.java:91)
	at javax.swing.plaf.basic.BasicTextUI$DragListener.mouseDragged(BasicTextUI.java:2161)
	at java.awt.AWTEventMulticaster.mouseDragged(AWTEventMulticaster.java:319)
	at java.awt.AWTEventMulticaster.mouseDragged(AWTEventMulticaster.java:319)
	at java.awt.AWTEventMulticaster.mouseDragged(AWTEventMulticaster.java:319)
	at java.awt.Component.processMouseMotionEvent(Component.java:6587)
	at javax.swing.JComponent.processMouseMotionEvent(JComponent.java:3342)
	at java.awt.Component.processEvent(Component.java:6308)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4552)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:36)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:130)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:340)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:143)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1236)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2020-04-01:
out of ram

xaveagle, 2020-07-09:
Auto Reported Issue 
BowlerStudio Build 1.9.4
BowlerKernel 0.62.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$648/1875398210.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:126)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:340)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:146)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1252)
	at java.lang.Thread.run(Thread.java:748)

```

acamilo, 2021-01-03:
Auto Reported Issue 
BowlerStudio Build 1.17.6
BowlerKernel 0.64.0
JavaCad Version: 0.25.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$650/1482120305.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:126)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:145)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1248)
	at java.lang.Thread.run(Thread.java:748)

```

keionbis, 2021-11-17:
Auto Reported Issue 
BowlerStudio Build 1.36.6
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$699/1947960464.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:188)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at javafx.embed.swing.FXDnD$FXDragSourceContextPeer.startDrag(FXDnD.java:360)
	at sun.awt.dnd.SunDragSourceContextPeer.startDrag(SunDragSourceContextPeer.java:135)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:321)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:426)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:188)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at javafx.embed.swing.FXDnD$FXDragSourceContextPeer.startDrag(FXDnD.java:360)
	at sun.awt.dnd.SunDragSourceContextPeer.startDrag(SunDragSourceContextPeer.java:135)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:321)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:426)
	at java.awt.dnd.DragGestureEvent.startDrag(DragGestureEvent.java:263)
[... 33 lines cut ...]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:146)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1265)
	at java.lang.Thread.run(Thread.java:748)

```

keionbis, 2021-11-28:
Auto Reported Issue 
BowlerStudio Build 1.36.6
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$686/324088959.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:188)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at javafx.embed.swing.FXDnD$FXDragSourceContextPeer.startDrag(FXDnD.java:360)
	at sun.awt.dnd.SunDragSourceContextPeer.startDrag(SunDragSourceContextPeer.java:135)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:321)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:426)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:188)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at javafx.embed.swing.FXDnD$FXDragSourceContextPeer.startDrag(FXDnD.java:360)
	at sun.awt.dnd.SunDragSourceContextPeer.startDrag(SunDragSourceContextPeer.java:135)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:321)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:426)
	at java.awt.dnd.DragGestureEvent.startDrag(DragGestureEvent.java:263)
[... 33 lines cut ...]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:146)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1265)
	at java.lang.Thread.run(Thread.java:748)

```

keionbis, 2021-11-28:
Auto Reported Issue 
BowlerStudio Build 1.36.6
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$680/1371413791.run(Unknown Source)
	at javafx.embed.swing.SwingFXUtils.runOnEDT(SwingFXUtils.java:288)
	at javafx.embed.swing.SwingNode.locateLwFrame(SwingNode.java:678)
	at javafx.embed.swing.SwingNode.lambda$resize$13(SwingNode.java:438)
	at javafx.embed.swing.SwingNode$$Lambda$679/828056225.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:188)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:188)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at javafx.embed.swing.FXDnD$FXDragSourceContextPeer.startDrag(FXDnD.java:360)
	at sun.awt.dnd.SunDragSourceContextPeer.startDrag(SunDragSourceContextPeer.java:135)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:321)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:426)
	at java.awt.dnd.DragGestureEvent.startDrag(DragGestureEvent.java:263)
	at javax.swing.TransferHandler$DragHandler.dragGestureRecognized(TransferHandler.java:1608)
[... 30 lines cut ...]
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:129)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:557)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$8$1$1.run(BowlerStudioMenu.java:884)

```

keionbis, 2021-12-06:
Auto Reported Issue 
BowlerStudio Build 1.36.6
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$703/77582682.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:188)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at javafx.embed.swing.FXDnD$FXDragSourceContextPeer.startDrag(FXDnD.java:360)
	at sun.awt.dnd.SunDragSourceContextPeer.startDrag(SunDragSourceContextPeer.java:135)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:321)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:426)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:188)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at javafx.embed.swing.FXDnD$FXDragSourceContextPeer.startDrag(FXDnD.java:360)
	at sun.awt.dnd.SunDragSourceContextPeer.startDrag(SunDragSourceContextPeer.java:135)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:321)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:426)
	at java.awt.dnd.DragGestureEvent.startDrag(DragGestureEvent.java:263)
[... 31 lines cut ...]
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$7.run(BowlerStudioMenu.java:811)

```

JansenSmith, 2022-04-08:
Auto Reported Issue 
BowlerStudio Build 1.45.0
BowlerKernel 1.44.1
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$744/1135771993.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:528)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$8$1$1.run(BowlerStudioMenu.java:914)

```

JansenSmith, 2023-01-12:
Auto Reported Issue 
BowlerStudio Build 1.64.3
BowlerKernel 1.52.1
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$652/1356087482.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:59)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:350)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:158)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:280)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1700(BowlerStudioMenu.java:67)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.run(BowlerStudioMenu.java:250)

```

JansenSmith, 2023-02-09:
Auto Reported Issue 
BowlerStudio Build 1.73.2
BowlerKernel 1.58.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$675/2107660942.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:59)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:350)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:158)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:281)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1(BowlerStudioMenu.java:266)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.run(BowlerStudioMenu.java:251)

```

JansenSmith, 2023-02-26:
Auto Reported Issue 
BowlerStudio Build 1.74.5
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$656/1186769064.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:210)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:59)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:350)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:158)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:281)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1700(BowlerStudioMenu.java:67)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.run(BowlerStudioMenu.java:251)

```

JansenSmith, 2023-02-26:
Auto Reported Issue 
BowlerStudio Build 1.74.5
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$666/2127293331.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:210)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:59)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:350)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:158)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:281)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1700(BowlerStudioMenu.java:67)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.run(BowlerStudioMenu.java:251)

```

JansenSmith, 2023-03-10:
Auto Reported Issue 
BowlerStudio Build 1.76.0
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$667/1677717461.run(Unknown Source)
	at javafx.embed.swing.SwingFXUtils.runOnEDT(SwingFXUtils.java:288)
	at javafx.embed.swing.SwingNode.locateLwFrame(SwingNode.java:678)
	at javafx.embed.swing.SwingNode.lambda$resize$13(SwingNode.java:438)
	at javafx.embed.swing.SwingNode$$Lambda$669/824445435.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:350)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:157)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:280)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1700(BowlerStudioMenu.java:66)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.run(BowlerStudioMenu.java:250)

```

JansenSmith, 2023-03-10:
interestingly, also had this notification:
[image]