CommonWealthRobotics/BowlerStudio #187: Build 1.5.9 org.kohsuke.github.Requester$PagingIterator at line 699

madhephaestus, 2020-04-02:
Auto Reported Issue 
BowlerStudio Build 1.5.9
BowlerKernel 0.61.3
JavaCad Version: 0.23.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.kohsuke.github.GHException: Failed to retrieve https://api.github.com/orgs/NeuronRobotics/repos?per_page=100
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:699)
	at org.kohsuke.github.Requester$PagingIterator.hasNext(Requester.java:661)
	at org.kohsuke.github.PagedIterator.fetch(PagedIterator.java:53)
	at org.kohsuke.github.PagedIterator.hasNext(PagedIterator.java:40)
	at org.kohsuke.github.GHPerson.getRepositories(GHPerson.java:65)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$9(BowlerStudioMenu.java:309)
	at java.lang.Thread.run(Thread.java:748)
Caused by: org.kohsuke.github.HttpException: Server returned HTTP response code: -1, message: 'null' for URL: https://api.github.com/orgs/NeuronRobotics/repos?per_page=100
	at org.kohsuke.github.Requester.parse(Requester.java:840)
	at org.kohsuke.github.Requester.parse(Requester.java:838)
	at org.kohsuke.github.Requester.parse(Requester.java:838)
	at org.kohsuke.github.Requester.parse(Requester.java:782)
	at org.kohsuke.github.Requester.access$300(Requester.java:75)
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:688)
	... 6 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1950)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1945)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.net.www.protocol.http.HttpURLConnection.getChainedException(HttpURLConnection.java:1944)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1514)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1498)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:352)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 11 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at java.net.SocketInputStream.socketRead0(Native Method)
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116)
	at java.net.SocketInputStream.read(SocketInputStream.java:171)
	at java.net.SocketInputStream.read(SocketInputStream.java:141)
	at sun.security.ssl.InputRecord.readFully(InputRecord.java:465)
	at sun.security.ssl.InputRecord.read(InputRecord.java:503)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:975)
	at sun.security.ssl.SSLSocketImpl.readDataRecord(SSLSocketImpl.java:933)
	at sun.security.ssl.AppInputStream.read(AppInputStream.java:105)
	at java.io.BufferedInputStream.fill(BufferedInputStream.java:246)
	at java.io.BufferedInputStream.read1(BufferedInputStream.java:286)
	at java.io.BufferedInputStream.read(BufferedInputStream.java:345)
	at sun.net.www.http.HttpClient.parseHTTPHeader(HttpClient.java:743)
	at sun.net.www.http.HttpClient.parseHTTP(HttpClient.java:678)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1593)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1498)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:352)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 9 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:158)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:61)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$9(BowlerStudioMenu.java:319)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:38)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$9(BowlerStudioMenu.java:319)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2020-04-02:
Auto Reported Issue 
BowlerStudio Build 1.5.9
BowlerKernel 0.61.3
JavaCad Version: 0.23.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.kohsuke.github.GHException: Failed to retrieve https://api.github.com/user/repos?type=all&per_page=30&page=12
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:699)
	at org.kohsuke.github.Requester$PagingIterator.hasNext(Requester.java:661)
	at org.kohsuke.github.PagedIterator.fetch(PagedIterator.java:53)
	at org.kohsuke.github.PagedIterator.hasNext(PagedIterator.java:40)
	at org.kohsuke.github.GHMyself.getAllRepositories(GHMyself.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$6(BowlerStudioMenu.java:277)
	at java.lang.Thread.run(Thread.java:748)
Caused by: org.kohsuke.github.HttpException: Server returned HTTP response code: -1, message: 'null' for URL: https://api.github.com/user/repos?type=all&per_page=30&page=12
	at org.kohsuke.github.Requester.parse(Requester.java:840)
	at org.kohsuke.github.Requester.parse(Requester.java:838)
	at org.kohsuke.github.Requester.parse(Requester.java:838)
	at org.kohsuke.github.Requester.parse(Requester.java:782)
	at org.kohsuke.github.Requester.access$300(Requester.java:75)
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:688)
	... 6 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1950)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1945)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.net.www.protocol.http.HttpURLConnection.getChainedException(HttpURLConnection.java:1944)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1514)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1498)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:352)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 11 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at java.net.SocketInputStream.socketRead0(Native Method)
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116)
	at java.net.SocketInputStream.read(SocketInputStream.java:171)
	at java.net.SocketInputStream.read(SocketInputStream.java:141)
	at sun.security.ssl.InputRecord.readFully(InputRecord.java:465)
	at sun.security.ssl.InputRecord.read(InputRecord.java:503)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:975)
	at sun.security.ssl.SSLSocketImpl.readDataRecord(SSLSocketImpl.java:933)
	at sun.security.ssl.AppInputStream.read(AppInputStream.java:105)
	at java.io.BufferedInputStream.fill(BufferedInputStream.java:246)
	at java.io.BufferedInputStream.read1(BufferedInputStream.java:286)
	at java.io.BufferedInputStream.read(BufferedInputStream.java:345)
	at sun.net.www.http.HttpClient.parseHTTPHeader(HttpClient.java:743)
	at sun.net.www.http.HttpClient.parseHTTP(HttpClient.java:678)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1593)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1498)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:352)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 9 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:158)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:61)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$6(BowlerStudioMenu.java:291)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:38)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$6(BowlerStudioMenu.java:291)
	at java.lang.Thread.run(Thread.java:748)

```

closed 2020-04-02 by madhephaestus

madhephaestus, 2022-03-31:
Auto Reported Issue 
BowlerStudio Build 1.43.2
BowlerKernel 1.43.7
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_292
JavaFX Version: 8.0.292

OS = Linux amd64 x64
```
org.kohsuke.github.GHException: Failed to retrieve https://api.github.com/user/repos?type=all&per_page=30&page=3
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:699)
	at org.kohsuke.github.Requester$PagingIterator.hasNext(Requester.java:661)
	at org.kohsuke.github.PagedIterator.fetch(PagedIterator.java:53)
	at org.kohsuke.github.PagedIterator.hasNext(PagedIterator.java:40)
	at org.kohsuke.github.GHMyself.getAllRepositories(GHMyself.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$6(BowlerStudioMenu.java:291)
	at java.lang.Thread.run(Thread.java:748)
Caused by: org.kohsuke.github.HttpException: Server returned HTTP response code: -1, message: 'null' for URL: https://api.github.com/user/repos?type=all&per_page=30&page=3
	at org.kohsuke.github.Requester.parse(Requester.java:840)
	at org.kohsuke.github.Requester.parse(Requester.java:838)
	at org.kohsuke.github.Requester.parse(Requester.java:838)
	at org.kohsuke.github.Requester.parse(Requester.java:782)
	at org.kohsuke.github.Requester.access$300(Requester.java:75)
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:688)
	... 6 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1950)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1945)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.net.www.protocol.http.HttpURLConnection.getChainedException(HttpURLConnection.java:1944)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1514)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1498)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:352)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 11 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at java.net.SocketInputStream.socketRead0(Native Method)
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116)
	at java.net.SocketInputStream.read(SocketInputStream.java:171)
	at java.net.SocketInputStream.read(SocketInputStream.java:141)
	at sun.security.ssl.SSLSocketInputRecord.read(SSLSocketInputRecord.java:476)
	at sun.security.ssl.SSLSocketInputRecord.readHeader(SSLSocketInputRecord.java:470)
	at sun.security.ssl.SSLSocketInputRecord.bytesInCompletePacket(SSLSocketInputRecord.java:70)
	at sun.security.ssl.SSLSocketImpl.readApplicationRecord(SSLSocketImpl.java:1355)
	at sun.security.ssl.SSLSocketImpl.access$300(SSLSocketImpl.java:73)
	at sun.security.ssl.SSLSocketImpl$AppInputStream.read(SSLSocketImpl.java:964)
	at java.io.BufferedInputStream.fill(BufferedInputStream.java:246)
	at java.io.BufferedInputStream.read1(BufferedInputStream.java:286)
	at java.io.BufferedInputStream.read(BufferedInputStream.java:345)
	at sun.net.www.http.HttpClient.parseHTTPHeader(HttpClient.java:743)
	at sun.net.www.http.HttpClient.parseHTTP(HttpClient.java:678)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1593)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1498)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:352)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 9 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$6(BowlerStudioMenu.java:305)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$6(BowlerStudioMenu.java:305)
	at java.lang.Thread.run(Thread.java:748)

```

acamilo, 2023-01-15:
Auto Reported Issue 
BowlerStudio Build 1.65.2
BowlerKernel 1.52.1
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
org.kohsuke.github.GHException: Failed to retrieve https://api.github.com/orgs/CommonWealthRobotics/repos?per_page=100
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:699)
	at org.kohsuke.github.Requester$PagingIterator.hasNext(Requester.java:661)
	at org.kohsuke.github.PagedIterator.fetch(PagedIterator.java:53)
	at org.kohsuke.github.PagedIterator.hasNext(PagedIterator.java:40)
	at org.kohsuke.github.GHPerson.getRepositories(GHPerson.java:65)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$loadOrganizations$11(BowlerStudioMenu.java:370)
	at java.lang.Thread.run(Thread.java:748)
Caused by: org.kohsuke.github.HttpException: Server returned HTTP response code: -1, message: 'null' for URL: https://api.github.com/orgs/CommonWealthRobotics/repos?per_page=100
	at org.kohsuke.github.Requester.parse(Requester.java:840)
	at org.kohsuke.github.Requester.parse(Requester.java:838)
	at org.kohsuke.github.Requester.parse(Requester.java:838)
	at org.kohsuke.github.Requester.parse(Requester.java:782)
	at org.kohsuke.github.Requester.access$300(Requester.java:75)
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:688)
	... 6 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1944)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1939)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.net.www.protocol.http.HttpURLConnection.getChainedException(HttpURLConnection.java:1938)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1508)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:347)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 11 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at java.net.SocketInputStream.socketRead0(Native Method)
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116)
	at java.net.SocketInputStream.read(SocketInputStream.java:171)
	at java.net.SocketInputStream.read(SocketInputStream.java:141)
	at sun.security.ssl.InputRecord.readFully(InputRecord.java:465)
	at sun.security.ssl.InputRecord.read(InputRecord.java:503)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:975)
	at sun.security.ssl.SSLSocketImpl.readDataRecord(SSLSocketImpl.java:933)
	at sun.security.ssl.AppInputStream.read(AppInputStream.java:105)
	at java.io.BufferedInputStream.fill(BufferedInputStream.java:246)
	at java.io.BufferedInputStream.read1(BufferedInputStream.java:286)
	at java.io.BufferedInputStream.read(BufferedInputStream.java:345)
	at sun.net.www.http.HttpClient.parseHTTPHeader(HttpClient.java:735)
	at sun.net.www.http.HttpClient.parseHTTP(HttpClient.java:678)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1587)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:347)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 9 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$loadOrganizations$11(BowlerStudioMenu.java:381)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$loadOrganizations$11(BowlerStudioMenu.java:381)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2023-01-17:
Auto Reported Issue 
BowlerStudio Build 1.66.2
BowlerKernel 1.52.1
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.kohsuke.github.GHException: Failed to retrieve https://api.github.com/user/1254726/gists?page=6
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:699)
	at org.kohsuke.github.Requester$PagingIterator.hasNext(Requester.java:661)
	at org.kohsuke.github.PagedIterator.fetch(PagedIterator.java:53)
	at org.kohsuke.github.PagedIterator.hasNext(PagedIterator.java:40)
	at org.kohsuke.github.PagedIterable.asList(PagedIterable.java:56)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$LoadGistMenu$13(BowlerStudioMenu.java:397)
	at java.lang.Thread.run(Thread.java:748)
Caused by: org.kohsuke.github.HttpException: {
  "message": "Server Error"
}

	at org.kohsuke.github.Requester.handleApiError(Requester.java:903)
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:693)
	... 6 more
Caused by: org.kohsuke.github.HttpException: Server returned HTTP response code: 502, message: 'Bad Gateway' for URL: https://api.github.com/user/1254726/gists?page=6
	at org.kohsuke.github.Requester.parse(Requester.java:840)
	at org.kohsuke.github.Requester.parse(Requester.java:782)
	at org.kohsuke.github.Requester.access$300(Requester.java:75)
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:688)
	... 6 more
Caused by: java.io.IOException: Server returned HTTP response code: 502 for URL: https://api.github.com/user/1254726/gists?page=6
	at sun.reflect.GeneratedConstructorAccessor46.newInstance(Unknown Source)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1950)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1945)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.net.www.protocol.http.HttpURLConnection.getChainedException(HttpURLConnection.java:1944)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1514)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1498)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getInputStream(HttpsURLConnectionImpl.java:268)
	at org.kohsuke.github.Requester.parse(Requester.java:818)
	... 9 more
Caused by: java.io.IOException: Server returned HTTP response code: 502 for URL: https://api.github.com/user/1254726/gists?page=6
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1900)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1498)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:352)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 9 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:59)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:350)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:158)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:280)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1700(BowlerStudioMenu.java:67)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.run(BowlerStudioMenu.java:250)

```

JansenSmith, 2023-02-16:
Auto Reported Issue 
BowlerStudio Build 1.73.2
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
org.kohsuke.github.GHException: Failed to retrieve https://api.github.com/user/repos?type=all&per_page=30&page=4
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:699)
	at org.kohsuke.github.Requester$PagingIterator.hasNext(Requester.java:661)
	at org.kohsuke.github.PagedIterator.fetch(PagedIterator.java:53)
	at org.kohsuke.github.PagedIterator.hasNext(PagedIterator.java:40)
	at org.kohsuke.github.GHMyself.getAllRepositories(GHMyself.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$6(BowlerStudioMenu.java:339)
	at java.lang.Thread.run(Thread.java:748)
Caused by: org.kohsuke.github.HttpException: Server returned HTTP response code: -1, message: 'null' for URL: https://api.github.com/user/repos?type=all&per_page=30&page=4
	at org.kohsuke.github.Requester.parse(Requester.java:840)
	at org.kohsuke.github.Requester.parse(Requester.java:838)
	at org.kohsuke.github.Requester.parse(Requester.java:838)
	at org.kohsuke.github.Requester.parse(Requester.java:782)
	at org.kohsuke.github.Requester.access$300(Requester.java:75)
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:688)
	... 6 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1944)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1939)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.net.www.protocol.http.HttpURLConnection.getChainedException(HttpURLConnection.java:1938)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1508)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:347)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 11 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at java.net.SocketInputStream.socketRead0(Native Method)
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116)
	at java.net.SocketInputStream.read(SocketInputStream.java:171)
	at java.net.SocketInputStream.read(SocketInputStream.java:141)
	at sun.security.ssl.InputRecord.readFully(InputRecord.java:465)
	at sun.security.ssl.InputRecord.read(InputRecord.java:503)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:975)
	at sun.security.ssl.SSLSocketImpl.readDataRecord(SSLSocketImpl.java:933)
	at sun.security.ssl.AppInputStream.read(AppInputStream.java:105)
	at java.io.BufferedInputStream.fill(BufferedInputStream.java:246)
	at java.io.BufferedInputStream.read1(BufferedInputStream.java:286)
	at java.io.BufferedInputStream.read(BufferedInputStream.java:345)
	at sun.net.www.http.HttpClient.parseHTTPHeader(HttpClient.java:735)
	at sun.net.www.http.HttpClient.parseHTTP(HttpClient.java:678)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1587)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:347)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 9 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$6(BowlerStudioMenu.java:353)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$6(BowlerStudioMenu.java:353)
	at java.lang.Thread.run(Thread.java:748)

```

JansenSmith, 2023-02-23:
Auto Reported Issue 
BowlerStudio Build 1.73.2
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
org.kohsuke.github.GHException: Failed to retrieve https://api.github.com/orgs/TechnocopiaPlant/repos?per_page=100
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:699)
	at org.kohsuke.github.Requester$PagingIterator.hasNext(Requester.java:661)
	at org.kohsuke.github.PagedIterator.fetch(PagedIterator.java:53)
	at org.kohsuke.github.PagedIterator.hasNext(PagedIterator.java:40)
	at org.kohsuke.github.GHPerson.getRepositories(GHPerson.java:65)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$9(BowlerStudioMenu.java:371)
	at java.lang.Thread.run(Thread.java:748)
Caused by: org.kohsuke.github.HttpException: Server returned HTTP response code: -1, message: 'null' for URL: https://api.github.com/orgs/TechnocopiaPlant/repos?per_page=100
	at org.kohsuke.github.Requester.parse(Requester.java:840)
	at org.kohsuke.github.Requester.parse(Requester.java:838)
	at org.kohsuke.github.Requester.parse(Requester.java:838)
	at org.kohsuke.github.Requester.parse(Requester.java:782)
	at org.kohsuke.github.Requester.access$300(Requester.java:75)
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:688)
	... 6 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1944)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1939)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.net.www.protocol.http.HttpURLConnection.getChainedException(HttpURLConnection.java:1938)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1508)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:347)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 11 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at java.net.SocketInputStream.socketRead0(Native Method)
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116)
	at java.net.SocketInputStream.read(SocketInputStream.java:171)
	at java.net.SocketInputStream.read(SocketInputStream.java:141)
	at sun.security.ssl.InputRecord.readFully(InputRecord.java:465)
	at sun.security.ssl.InputRecord.read(InputRecord.java:503)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:975)
	at sun.security.ssl.SSLSocketImpl.readDataRecord(SSLSocketImpl.java:933)
	at sun.security.ssl.AppInputStream.read(AppInputStream.java:105)
	at java.io.BufferedInputStream.fill(BufferedInputStream.java:246)
	at java.io.BufferedInputStream.read1(BufferedInputStream.java:286)
	at java.io.BufferedInputStream.read(BufferedInputStream.java:345)
	at sun.net.www.http.HttpClient.parseHTTPHeader(HttpClient.java:735)
	at sun.net.www.http.HttpClient.parseHTTP(HttpClient.java:678)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1587)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:347)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 9 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$9(BowlerStudioMenu.java:382)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$9(BowlerStudioMenu.java:382)
	at java.lang.Thread.run(Thread.java:748)

```

JansenSmith, 2023-02-27:
Auto Reported Issue 
BowlerStudio Build 1.74.5
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
org.kohsuke.github.GHException: Failed to retrieve https://api.github.com/orgs/CommonWealthRobotics/repos?per_page=100
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:699)
	at org.kohsuke.github.Requester$PagingIterator.hasNext(Requester.java:661)
	at org.kohsuke.github.PagedIterator.fetch(PagedIterator.java:53)
	at org.kohsuke.github.PagedIterator.hasNext(PagedIterator.java:40)
	at org.kohsuke.github.GHPerson.getRepositories(GHPerson.java:65)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$loadOrganizations$11(BowlerStudioMenu.java:371)
	at java.lang.Thread.run(Thread.java:748)
Caused by: org.kohsuke.github.HttpException: Server returned HTTP response code: -1, message: 'null' for URL: https://api.github.com/orgs/CommonWealthRobotics/repos?per_page=100
	at org.kohsuke.github.Requester.parse(Requester.java:840)
	at org.kohsuke.github.Requester.parse(Requester.java:838)
	at org.kohsuke.github.Requester.parse(Requester.java:838)
	at org.kohsuke.github.Requester.parse(Requester.java:782)
	at org.kohsuke.github.Requester.access$300(Requester.java:75)
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:688)
	... 6 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1944)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1939)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.net.www.protocol.http.HttpURLConnection.getChainedException(HttpURLConnection.java:1938)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1508)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:347)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 11 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at java.net.SocketInputStream.socketRead0(Native Method)
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116)
	at java.net.SocketInputStream.read(SocketInputStream.java:171)
	at java.net.SocketInputStream.read(SocketInputStream.java:141)
	at sun.security.ssl.InputRecord.readFully(InputRecord.java:465)
	at sun.security.ssl.InputRecord.read(InputRecord.java:503)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:975)
	at sun.security.ssl.SSLSocketImpl.readDataRecord(SSLSocketImpl.java:933)
	at sun.security.ssl.AppInputStream.read(AppInputStream.java:105)
	at java.io.BufferedInputStream.fill(BufferedInputStream.java:246)
	at java.io.BufferedInputStream.read1(BufferedInputStream.java:286)
	at java.io.BufferedInputStream.read(BufferedInputStream.java:345)
	at sun.net.www.http.HttpClient.parseHTTPHeader(HttpClient.java:735)
	at sun.net.www.http.HttpClient.parseHTTP(HttpClient.java:678)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1587)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:347)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 9 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:210)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$loadOrganizations$11(BowlerStudioMenu.java:382)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$loadOrganizations$11(BowlerStudioMenu.java:382)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2024-04-24:
Auto Reported Issue 
BowlerStudio Build 2.17.3
BowlerKernel 1.59.0
JavaCad Version: 0.31.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.kohsuke.github.GHException: Failed to retrieve https://api.github.com/user/1254726/gists?page=8
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:699)
	at org.kohsuke.github.Requester$PagingIterator.hasNext(Requester.java:661)
	at org.kohsuke.github.PagedIterator.fetch(PagedIterator.java:53)
	at org.kohsuke.github.PagedIterator.hasNext(PagedIterator.java:40)
	at org.kohsuke.github.PagedIterable.asList(PagedIterable.java:56)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.lambda$LoadGistMenu$13(BowlerStudioMenu.java:411)
	at java.lang.Thread.run(Thread.java:748)
Caused by: org.kohsuke.github.HttpException: Server returned HTTP response code: 500, message: 'Internal Server Error' for URL: https://api.github.com/user/1254726/gists?page=8
	at org.kohsuke.github.Requester.parse(Requester.java:840)
	at org.kohsuke.github.Requester.parse(Requester.java:782)
	at org.kohsuke.github.Requester.access$300(Requester.java:75)
	at org.kohsuke.github.Requester$PagingIterator.fetch(Requester.java:688)
	... 6 more
Caused by: java.io.IOException: Server returned HTTP response code: 500 for URL: https://api.github.com/user/1254726/gists?page=8
	at sun.reflect.GeneratedConstructorAccessor65.newInstance(Unknown Source)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1950)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1945)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.net.www.protocol.http.HttpURLConnection.getChainedException(HttpURLConnection.java:1944)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1514)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1498)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getInputStream(HttpsURLConnectionImpl.java:268)
	at org.kohsuke.github.Requester.parse(Requester.java:818)
	... 9 more
Caused by: java.io.IOException: Server returned HTTP response code: 500 for URL: https://api.github.com/user/1254726/gists?page=8
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1900)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1498)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:352)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 9 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:62)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:53)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:52)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:141)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:352)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:635)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:285)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1400(BowlerStudioMenu.java:72)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$run$0(BowlerStudioMenu.java:253)
	at java.lang.Thread.run(Thread.java:748)

```