CommonWealthRobotics/BowlerStudio #189: Cannot edit robot XML definition

Octogonapus, 2020-04-23:
I am trying to edit the scale value for a servo. When I hit Publish to push my changes, everything appears to work fine. BowlerStudio even pulls and logs `PUSH OK`:
```
Pulling https://github.com/Octogonapus/greycat.git   ... Success!
PUSH OK! file: /home/salmon/bowler-workspace/gitcache/github.com/Octogonapus/greycat/MediumKat_black.xml on branch master
```

However, nothing is pushed. There are no new commits on the repo. The robot's XML file is never changed. Disconnecting and reloading the robot shows that no XML changes were saved.

BowlerStudio Build 1.5.12
BowlerKernel 0.61.3
JavaCad Version: 0.24.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_232
JavaFX Version: 8.0.232

OS = Linux amd64 x64

madhephaestus, 2020-04-23:
did you edit the UI? in which case you need to use the UI publish. If you edited the text file, you need to publish from the text editor's publish. Which publish did you use, and where did you make the changed to the values?

Octogonapus, 2020-04-23:
I changed the value in the UI. I used the publish button in the UI (right above the "make copy of creature" button).

Octogonapus, 2020-04-23:
Closing this until I can reproduce it again.

closed 2020-04-23 by Octogonapus