CommonWealthRobotics/BowlerStudio #194: Build 1.5.14 org.eclipse.jgit.api.FetchCommand at line 252

keionbis, 2020-04-24:
Auto Reported Issue 
BowlerStudio Build 1.5.14
BowlerKernel 0.61.3
JavaCad Version: 0.24.0
Java-Bowler Version: 3.28.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.InvalidRemoteException: Invalid remote: origin
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:252)
	at org.eclipse.jgit.api.PullCommand.call(PullCommand.java:296)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:968)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1040)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$loginEvent$0(BowlerStudioMenuWorkspace.java:41)
	at java.lang.Thread.run(Thread.java:748)
Caused by: org.eclipse.jgit.errors.NoRemoteRepositoryException: https://github.com/OperationSmallKat/smallkat_xl_bowler.git: https://github.com/OperationSmallKat/smallkat_xl_bowler.git/info/refs?service=git-upload-pack not found: Not Found
	at org.eclipse.jgit.transport.TransportHttp.createNotFoundException(TransportHttp.java:522)
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:554)
	at org.eclipse.jgit.transport.TransportHttp.openFetch(TransportHttp.java:392)
	at org.eclipse.jgit.transport.FetchProcess.executeImp(FetchProcess.java:138)
	at org.eclipse.jgit.transport.FetchProcess.execute(FetchProcess.java:124)
	at org.eclipse.jgit.transport.Transport.fetch(Transport.java:1293)
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:244)
	... 5 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:180)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:62)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:994)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1040)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$loginEvent$0(BowlerStudioMenuWorkspace.java:41)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:38)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:111)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:274)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

madhephaestus, 2020-04-24:
You are pulling some junk URL

madhephaestus, 2021-03-30:
Auto Reported Issue 
BowlerStudio Build 1.27.2
BowlerKernel 0.65.0
JavaCad Version: 0.25.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.InvalidRemoteException: Invalid remote: origin
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:252)
	at org.eclipse.jgit.api.CloneCommand.fetch(CloneCommand.java:305)
	at org.eclipse.jgit.api.CloneCommand.call(CloneCommand.java:204)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1206)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.getBranch(ScriptingEngine.java:732)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1044)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$pull$0.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:116)
	at PushOutUpdate.run(PushOutUpdate.groovy:51)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:61)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:180)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineFileScriptRun(ScriptingEngine.java:636)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget$2.run(ScriptingFileWidget.java:278)
Caused by: org.eclipse.jgit.errors.NoRemoteRepositoryException: https://github.com/RBE200x-lab/2002_LabTeam_01.git: https://github.com/RBE200x-lab/2002_LabTeam_01.git/info/refs?service=git-upload-pack not found: Not Found
	at org.eclipse.jgit.transport.TransportHttp.createNotFoundException(TransportHttp.java:522)
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:554)
	at org.eclipse.jgit.transport.TransportHttp.openFetch(TransportHttp.java:392)
	at org.eclipse.jgit.transport.FetchProcess.executeImp(FetchProcess.java:138)
	at org.eclipse.jgit.transport.FetchProcess.execute(FetchProcess.java:124)
	at org.eclipse.jgit.transport.Transport.fetch(Transport.java:1293)
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:244)
	... 15 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1239)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.getBranch(ScriptingEngine.java:732)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1044)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$pull$0.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:116)
	at PushOutUpdate.run(PushOutUpdate.groovy:51)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:61)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:180)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineFileScriptRun(ScriptingEngine.java:636)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget$2.run(ScriptingFileWidget.java:278)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:114)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:293)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

acamilo, 2021-07-20:
Auto Reported Issue 
BowlerStudio Build 1.29.0
BowlerKernel 0.65.1
JavaCad Version: 0.25.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
org.eclipse.jgit.api.errors.InvalidRemoteException: Invalid remote: origin
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:252)
	at org.eclipse.jgit.api.CloneCommand.fetch(CloneCommand.java:305)
	at org.eclipse.jgit.api.CloneCommand.call(CloneCommand.java:204)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1255)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.getFullBranch(ScriptingEngine.java:749)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:658)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.codeFromGit(ScriptingEngine.java:609)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.gitScriptRun(ScriptingEngine.java:652)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$gitScriptRun.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:124)
	at launch.run(launch.groovy:12)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:61)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:183)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineFileScriptRun(ScriptingEngine.java:639)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget$2.run(ScriptingFileWidget.java:278)
Caused by: org.eclipse.jgit.errors.NoRemoteRepositoryException: https://github.com/madhephaestus/HephaestusArm2.git: https://github.com/madhephaestus/HephaestusArm2.git/info/refs?service=git-upload-pack not found: Not Found
	at org.eclipse.jgit.transport.TransportHttp.createNotFoundException(TransportHttp.java:522)
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:554)
	at org.eclipse.jgit.transport.TransportHttp.openFetch(TransportHttp.java:392)
	at org.eclipse.jgit.transport.FetchProcess.executeImp(FetchProcess.java:138)
	at org.eclipse.jgit.transport.FetchProcess.execute(FetchProcess.java:124)
	at org.eclipse.jgit.transport.Transport.fetch(Transport.java:1293)
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:244)
	... 17 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1288)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.getFullBranch(ScriptingEngine.java:749)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:658)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.codeFromGit(ScriptingEngine.java:609)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.gitScriptRun(ScriptingEngine.java:652)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$gitScriptRun.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:124)
	at launch.run(launch.groovy:12)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:61)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:183)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineFileScriptRun(ScriptingEngine.java:639)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget$2.run(ScriptingFileWidget.java:278)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:117)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:293)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```