CommonWealthRobotics/BowlerStudio #195: Calibration procedure/widget update

madhephaestus, 2020-04-29:
### Installed Version 
All
### OS type and version number
All
### Expected behavior
Calibration should be easy
### Actual Behavior
calibration is hard
### Steps to reproduce the behavior
calibrate a robots limb

First: in the setup for the DH parameters we should have a section for configuring the limb motion. This is where I would set the limb boundaries. How much off of the 0 degree home value you want the leg to be able to move. This would be indicated by the pi shapes as blue and green like they already are. Ideally at this point it would be nice to run the simulation without the hardware part just to see how the model moves with the given restraints. Once you are happy with the home values then you can calibrate the servo. This would be in the hardware section. This interface would ask for the servo range in degrees (will have to measure to make sure its right) and then overlay that range onto the model again preferably in orange. Idk why orange I just think it should be orange. The zero value of this should be in line with the link and by using a trim button you would shift that servo till it was in line and at the same time would shift the servo range to compensate. At the point you would just need to make sure your given boundaries fall within the servos range. And you should be able to have all the data in the backend from those numbers to convert the device center points and upper and lower boundaries while keeping it a little simpler to understand. See the attached Paint walk through. I think if you add an advanced tab we could put the more complex method in as well just to have both. Hopefully this makes sense and also actually works. Its late and I’m tired so I might have missed something but I think this should simply the process quite a bit.

 

 

Additional Ideas:

 

-I think we need a second control panel for robot roll and yaw. We don’t really have that right now and in my mind for a class, after you do your IK for a leg you should do the IK for the body and be able to control it moving up down side to side and roll in all while the feet are in the same place. All the stuff I was doing in Matlab cause I couldn’t in bowler.

-Fix the IK

-Isolate just the leg when calibrating in the window viewer and fix the camera so that it doesn’t always focus on one limb and cause the window screen to have a seizure.

@xaveagle 

[image]

madhephaestus, 2020-04-29:
so first things first, bounds are not part of D-H parameters, so that all belongs somewhere else.

Octogonapus, 2020-04-29:
You need to be able to set link and servo bounds, yes. But then you also need the tool to validate that your link bounds are feasible, since they are contrained by the servo bounds.

Representing these bounds graphically (maybe even in the 3D window) is a neat party trick but not really necessary.

madhephaestus, 2020-04-30:
@xaveagle please check out 1.5.17 for this feature

closed 2020-05-01 by madhephaestus