CommonWealthRobotics/BowlerStudio #199: Build 1.9.1 com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine at line 377

jhlee98y, 2020-05-26:
Auto Reported Issue 
BowlerStudio Build 1.9.1
BowlerKernel 0.62.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.RuntimeException
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:377)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteCache(ScriptingEngine.java:344)
	at com.neuronrobotics.bowlerstudio.BowlerStudio$3.run(BowlerStudio.java:751)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.BowlerStudio$3.run(BowlerStudio.java:742)

```

xaveagle, 2020-07-13:
Auto Reported Issue 
BowlerStudio Build 1.9.5
BowlerKernel 0.62.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.RuntimeException
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:377)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteCache(ScriptingEngine.java:344)
	at com.neuronrobotics.bowlerstudio.BowlerStudio$3.run(BowlerStudio.java:751)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.BowlerStudio$3.run(BowlerStudio.java:742)

```

jez2009, 2020-12-11:
Auto Reported Issue 
BowlerStudio Build 1.13.0
BowlerKernel 0.63.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 7 amd64 x64
```
java.lang.RuntimeException
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:377)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:354)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteCache(ScriptingEngine.java:344)
	at com.neuronrobotics.bowlerstudio.BowlerStudio$3.run(BowlerStudio.java:750)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.BowlerStudio$3.run(BowlerStudio.java:741)

```