CommonWealthRobotics/BowlerStudio #203: Build 1.9.4 org.eclipse.jgit.api.CheckoutCommand at line 285

wsmarshall, 2020-06-30:
Auto Reported Issue 
BowlerStudio Build 1.9.4
BowlerKernel 0.62.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_181
JavaFX Version: 8.0.181

OS = Windows 10 amd64 x64
```
org.eclipse.jgit.api.errors.CheckoutConflictException: Checkout conflict with files: 
testChessPiece.groovy
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:285)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1100)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1071)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.switchToThisNewBranch(BowlerStudioMenu.java:764)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.run(BowlerStudioMenu.java:752)
Caused by: org.eclipse.jgit.errors.CheckoutConflictException: Checkout conflict with files: 
testChessPiece.groovy
	at org.eclipse.jgit.dircache.DirCacheCheckout.doCheckout(DirCacheCheckout.java:528)
	at org.eclipse.jgit.dircache.DirCacheCheckout.checkout(DirCacheCheckout.java:493)
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:281)
	... 4 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1147)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1071)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.switchToThisNewBranch(BowlerStudioMenu.java:764)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.run(BowlerStudioMenu.java:752)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:111)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:293)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)

```

madhephaestus, 2020-10-12:
Auto Reported Issue 
BowlerStudio Build 1.12.0
BowlerKernel 0.62.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.CheckoutConflictException: Checkout conflict with files: 
cadGenHephaestusArm.groovy
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:285)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1100)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1071)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.switchToThisNewBranch(BowlerStudioMenu.java:763)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.run(BowlerStudioMenu.java:751)
Caused by: org.eclipse.jgit.errors.CheckoutConflictException: Checkout conflict with files: 
cadGenHephaestusArm.groovy
	at org.eclipse.jgit.dircache.DirCacheCheckout.doCheckout(DirCacheCheckout.java:528)
	at org.eclipse.jgit.dircache.DirCacheCheckout.checkout(DirCacheCheckout.java:493)
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:281)
	... 4 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1147)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1071)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.switchToThisNewBranch(BowlerStudioMenu.java:763)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.run(BowlerStudioMenu.java:751)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:111)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:292)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

madhephaestus, 2021-05-30:
Auto Reported Issue 
BowlerStudio Build 1.28.1
BowlerKernel 0.65.0
JavaCad Version: 0.25.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.CheckoutConflictException: Checkout conflict with files: 
.gitignore
bin/.gitignore
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:285)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1112)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1075)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.switchToThisNewBranch(BowlerStudioMenu.java:764)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.run(BowlerStudioMenu.java:752)
Caused by: org.eclipse.jgit.errors.CheckoutConflictException: Checkout conflict with files: 
.gitignore
bin/.gitignore
	at org.eclipse.jgit.dircache.DirCacheCheckout.doCheckout(DirCacheCheckout.java:528)
	at org.eclipse.jgit.dircache.DirCacheCheckout.checkout(DirCacheCheckout.java:493)
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:281)
	... 4 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1151)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1075)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.switchToThisNewBranch(BowlerStudioMenu.java:764)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.run(BowlerStudioMenu.java:752)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:114)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:293)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

madhephaestus, 2021-06-07:
Auto Reported Issue 
BowlerStudio Build 1.28.1
BowlerKernel 0.65.0
JavaCad Version: 0.25.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.CheckoutConflictException: Checkout conflict with files: 
cadGenHand.groovy
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:285)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1104)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1075)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.switchToThisNewBranch(BowlerStudioMenu.java:764)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.run(BowlerStudioMenu.java:752)
Caused by: org.eclipse.jgit.errors.CheckoutConflictException: Checkout conflict with files: 
cadGenHand.groovy
	at org.eclipse.jgit.dircache.DirCacheCheckout.doCheckout(DirCacheCheckout.java:528)
	at org.eclipse.jgit.dircache.DirCacheCheckout.checkout(DirCacheCheckout.java:493)
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:281)
	... 4 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1151)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1075)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.switchToThisNewBranch(BowlerStudioMenu.java:764)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.run(BowlerStudioMenu.java:752)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:114)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:293)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

keionbis, 2021-12-10:
Auto Reported Issue 
BowlerStudio Build 1.36.6
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.CheckoutConflictException: Checkout conflict with files: 
Fork.png
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:285)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1260)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1428)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.filesInGit(ScriptingEngine.java:510)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAllAssets(AssetFactory.java:148)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.setGitSource(AssetFactory.java:143)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:402)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: org.eclipse.jgit.errors.CheckoutConflictException: Checkout conflict with files: 
Fork.png
	at org.eclipse.jgit.dircache.DirCacheCheckout.doCheckout(DirCacheCheckout.java:528)
	at org.eclipse.jgit.dircache.DirCacheCheckout.checkout(DirCacheCheckout.java:493)
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:281)
	... 17 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1305)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1428)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.filesInGit(ScriptingEngine.java:510)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAllAssets(AssetFactory.java:148)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.setGitSource(AssetFactory.java:143)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:402)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:118)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:303)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

madhephaestus, 2022-08-20:
Auto Reported Issue 
BowlerStudio Build 1.55.1
BowlerKernel 1.49.1
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.CheckoutConflictException: Checkout conflict with files: 
jinputGrogu.groovy
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:285)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.shallowCheckout(ScriptingEngine.java:1533)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1472)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1641)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:956)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$fileFromGit$1.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:124)
	at jinputGrogu.meow(jinputGrogu.groovy:135)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.codehaus.groovy.reflection.CachedMethod.invoke(CachedMethod.java:90)
	at groovy.lang.MetaMethod.doMethodInvoke(MetaMethod.java:324)
	at groovy.lang.MetaClassImpl.invokeMethod(MetaClassImpl.java:1207)
	at org.codehaus.groovy.runtime.ScriptBytecodeAdapter.invokeMethodOnCurrentN(ScriptBytecodeAdapter.java:80)
	at jinputGrogu.this$dist$invoke$3(jinputGrogu.groovy)
	at jinputGrogu$2.methodMissing(jinputGrogu.groovy)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.codehaus.groovy.reflection.CachedMethod.invoke(CachedMethod.java:90)
	at groovy.lang.MetaClassImpl.invokeMissingMethod(MetaClassImpl.java:933)
	at groovy.lang.MetaClassImpl.invokePropertyOrMissing(MetaClassImpl.java:1256)
	at groovy.lang.MetaClassImpl.invokeMethod(MetaClassImpl.java:1209)
	at groovy.lang.MetaClassImpl.invokeMethod(MetaClassImpl.java:1016)
	at org.codehaus.groovy.runtime.callsite.PogoMetaClassSite.callCurrent(PogoMetaClassSite.java:66)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCallCurrent(CallSiteArray.java:49)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callCurrent(AbstractCallSite.java:133)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callCurrent(AbstractCallSite.java:137)
	at jinputGrogu$2.run(jinputGrogu.groovy:245)
Caused by: org.eclipse.jgit.errors.CheckoutConflictException: Checkout conflict with files: 
jinputGrogu.groovy
	at org.eclipse.jgit.dircache.DirCacheCheckout.doCheckout(DirCacheCheckout.java:528)
	at org.eclipse.jgit.dircache.DirCacheCheckout.checkout(DirCacheCheckout.java:493)
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:281)
	... 33 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1512)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1641)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:956)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$fileFromGit$1.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:124)
	at jinputGrogu.meow(jinputGrogu.groovy:135)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[... 23 lines cut ...]
```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:125)
	at com.neuronrobotics.bowlerstudio.PsudoSplash.setMessage(PsudoSplash.java:193)
	at com.neuronrobotics.bowlerstudio.SplashManager.renderSplashFrame(SplashManager.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.renderSplashFrame(BowlerStudio.java:472)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

commit 005991b7 (2021-06-09): resolve #203