CommonWealthRobotics/BowlerStudio #204: Build 1.9.4 org.eclipse.jgit.lib.Repository at line 542

wsmarshall, 2020-07-01:
Auto Reported Issue 
BowlerStudio Build 1.9.4
BowlerKernel 0.62.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_181
JavaFX Version: 8.0.181

OS = Windows 10 amd64 x64
```
java.lang.NullPointerException
	at org.eclipse.jgit.lib.Repository.resolve(Repository.java:542)
	at org.eclipse.jgit.lib.Repository.resolve(Repository.java:502)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$4.lambda$handle$6(BowlerStudioMenu.java:528)
	at java.lang.Thread.run(Unknown Source)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$4.lambda$handle$6(BowlerStudioMenu.java:616)
	at java.lang.Thread.run(Unknown Source)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.<clinit>(BowlerStudioMenu.java:129)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.initialize(BowlerStudioModularFrame.java:153)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at sun.reflect.misc.Trampoline.invoke(Unknown Source)
	at sun.reflect.GeneratedMethodAccessor39.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at sun.reflect.misc.MethodUtil.invoke(Unknown Source)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2566)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2441)
	at javafx.fxml.FXMLLoader.load(FXMLLoader.java:2409)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.lambda$start$8(BowlerStudio.java:640)
	at java.lang.Thread.run(Unknown Source)

```

wsmarshall, 2020-07-01:
Auto Reported Issue 
BowlerStudio Build 1.9.4
BowlerKernel 0.62.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_181
JavaFX Version: 8.0.181

OS = Windows 10 amd64 x64
```
java.lang.NullPointerException
	at org.eclipse.jgit.lib.Repository.resolve(Repository.java:542)
	at org.eclipse.jgit.lib.Repository.resolve(Repository.java:502)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$4.lambda$handle$6(BowlerStudioMenu.java:528)
	at java.lang.Thread.run(Unknown Source)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$4.lambda$handle$6(BowlerStudioMenu.java:616)
	at java.lang.Thread.run(Unknown Source)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.<clinit>(BowlerStudioMenu.java:129)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.initialize(BowlerStudioModularFrame.java:153)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at sun.reflect.misc.Trampoline.invoke(Unknown Source)
	at sun.reflect.GeneratedMethodAccessor39.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at sun.reflect.misc.MethodUtil.invoke(Unknown Source)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2566)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2441)
	at javafx.fxml.FXMLLoader.load(FXMLLoader.java:2409)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.lambda$start$8(BowlerStudio.java:640)
	at java.lang.Thread.run(Unknown Source)

```

wsmarshall, 2020-07-02:
so this all was from trying to create and add a branch to the repository. 
First, I tried from the menu: GitHub >> My Repositories >> MyUsername >> ~/JavaCAD >> add repo to workspace
Thereafter, it would add, but then using the menu to Workspace >> ~/JavaCAD >> branches would cause a crash. And then after the crash, the repo would intermittently not show up in the workspace anymore, starting the whole thing from the beginning. 

I also tried creating the new branch on github itself (through web browser), then accessing through bowler studio, but that didn't work either and caused crashes.

JansenSmith, 2022-03-08:
Auto Reported Issue 
BowlerStudio Build 1.41.4
BowlerKernel 1.40.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.NullPointerException
	at org.eclipse.jgit.lib.Repository.resolve(Repository.java:542)
	at org.eclipse.jgit.lib.Repository.resolve(Repository.java:502)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$4.lambda$handle$6(BowlerStudioMenu.java:551)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$4.lambda$handle$6(BowlerStudioMenu.java:639)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.<clinit>(BowlerStudioMenu.java:129)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.initialize(BowlerStudioModularFrame.java:153)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.reflect.misc.Trampoline.invoke(MethodUtil.java:72)
	at sun.reflect.GeneratedMethodAccessor42.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.reflect.misc.MethodUtil.invoke(MethodUtil.java:276)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2566)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2441)
	at javafx.fxml.FXMLLoader.load(FXMLLoader.java:2409)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.lambda$start$8(BowlerStudio.java:665)
	at java.lang.Thread.run(Thread.java:748)

```

closed 2022-03-09 by madhephaestus

JansenSmith, 2022-03-28:
Auto Reported Issue 
BowlerStudio Build 1.42.3
BowlerKernel 1.41.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.NullPointerException
	at org.eclipse.jgit.lib.Repository.resolve(Repository.java:542)
	at org.eclipse.jgit.lib.Repository.resolve(Repository.java:502)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$4.lambda$handle$6(BowlerStudioMenu.java:551)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$4.lambda$handle$6(BowlerStudioMenu.java:639)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.<clinit>(BowlerStudioMenu.java:129)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.initialize(BowlerStudioModularFrame.java:153)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.reflect.misc.Trampoline.invoke(MethodUtil.java:72)
	at sun.reflect.GeneratedMethodAccessor42.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.reflect.misc.MethodUtil.invoke(MethodUtil.java:276)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2566)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2441)
	at javafx.fxml.FXMLLoader.load(FXMLLoader.java:2409)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.lambda$start$8(BowlerStudio.java:665)
	at java.lang.Thread.run(Thread.java:748)

```