CommonWealthRobotics/BowlerStudio #210: Cannot clone parallel robot

Octogonapus, 2020-07-29:
### Installed Version 
v1.9.6

### OS type and version number
Fedora 32

### Expected behavior
The parallel robot should load into the creature lab like normal.

### Actual Behavior
An exception is printed to console and the robot does not load. I am left on the tutorial tab.

### Steps to reproduce the behavior
I deleted `bowler-workspace` and ran:
1. Add Device > Creatures > Paralell robot

This exception occurs:
```
Starting JavaFX initializer...class com.neuronrobotics.sdk.addons.kinematics.JavaFXInitializer
Finished JavaFX initializing! com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.<clinit>(AbstractKinematicsNR.java:118)
Device connected: com.neuronrobotics.sdk.pid.VirtualGenericPIDDevice@4a163b45
Device connected: com.neuronrobotics.sdk.pid.VirtualGenericPIDDevice@5e9f8ae8
java.lang.NullPointerException
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR._setDesiredJointSpaceVector(AbstractKinematicsNR.java:685)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.setDesiredTaskSpaceTransform(AbstractKinematicsNR.java:597)
	at com.neuronrobotics.sdk.addons.kinematics.MobileBase.loadConfigs(MobileBase.java:207)
	at com.neuronrobotics.sdk.addons.kinematics.MobileBase.<init>(MobileBase.java:89)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.fromGit(MobileBaseLoader.java:120)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$1.run(BowlerStudioMenu.java:167)
Loading Paralell group ParallelArmGroup limb ParalellArm1
Loading Paralell group ParallelArmGroup limb ParallelArm2
Mass of device ParallelArm is 0.01
Cloning files from: https://gist.github.com/bcb4760a449190206170.git
                to: /home/salmon/bowler-workspace/gitcache/gist.github.com/bcb4760a449190206170
org.eclipse.jgit.api.errors.InvalidRemoteException: Invalid remote: origin
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:252)
	at org.eclipse.jgit.api.CloneCommand.fetch(CloneCommand.java:305)
	at org.eclipse.jgit.api.CloneCommand.call(CloneCommand.java:204)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1202)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.getFullBranch(ScriptingEngine.java:742)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:651)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.setGitWalkingEngine(MobileBaseLoader.java:75)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.setDefaultWalkingEngine(MobileBaseLoader.java:63)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.<init>(MobileBaseLoader.java:24)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.initializeScripts(MobileBaseLoader.java:103)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.fromGit(MobileBaseLoader.java:123)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$1.run(BowlerStudioMenu.java:167)
Caused by: org.eclipse.jgit.errors.NoRemoteRepositoryException: https://gist.github.com/bcb4760a449190206170.git: https://gist.github.com/bcb4760a449190206170.git/info/refs?service=git-upload-pack not found: Not Found
	at org.eclipse.jgit.transport.TransportHttp.createNotFoundException(TransportHttp.java:522)
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:554)
	at org.eclipse.jgit.transport.TransportHttp.openFetch(TransportHttp.java:392)
	at org.eclipse.jgit.transport.FetchProcess.executeImp(FetchProcess.java:138)
	at org.eclipse.jgit.transport.FetchProcess.execute(FetchProcess.java:124)
	at org.eclipse.jgit.transport.Transport.fetch(Transport.java:1293)
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:244)
	... 11 more
org.eclipse.jgit.api.errors.InvalidRemoteException: Invalid remote: origin
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:252)
	at org.eclipse.jgit.api.CloneCommand.fetch(CloneCommand.java:305)
	at org.eclipse.jgit.api.CloneCommand.call(CloneCommand.java:204)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1202)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.getFullBranch(ScriptingEngine.java:742)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:651)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.setGitWalkingEngine(MobileBaseLoader.java:75)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.setDefaultWalkingEngine(MobileBaseLoader.java:63)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.<init>(MobileBaseLoader.java:24)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.initializeScripts(MobileBaseLoader.java:103)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.fromGit(MobileBaseLoader.java:123)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$1.run(BowlerStudioMenu.java:167)
Caused by: org.eclipse.jgit.errors.NoRemoteRepositoryException: https://gist.github.com/bcb4760a449190206170.git: https://gist.github.com/bcb4760a449190206170.git/info/refs?service=git-upload-pack not found: Not Found
	at org.eclipse.jgit.transport.TransportHttp.createNotFoundException(TransportHttp.java:522)
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:554)
	at org.eclipse.jgit.transport.TransportHttp.openFetch(TransportHttp.java:392)
	at org.eclipse.jgit.transport.FetchProcess.executeImp(FetchProcess.java:138)
	at org.eclipse.jgit.transport.FetchProcess.execute(FetchProcess.java:124)
	at org.eclipse.jgit.transport.Transport.fetch(Transport.java:1293)
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:244)
	... 11 more
java.lang.RuntimeException: Folder doesnt exist /home/salmon/bowler-workspace/gitcache/gist.github.com/bcb4760a449190206170
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.deleteFolder(ScriptingEngine.java:349)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1228)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.getFullBranch(ScriptingEngine.java:742)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:651)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.setGitWalkingEngine(MobileBaseLoader.java:75)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.setDefaultWalkingEngine(MobileBaseLoader.java:63)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.<init>(MobileBaseLoader.java:24)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.initializeScripts(MobileBaseLoader.java:103)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseLoader.fromGit(MobileBaseLoader.java:123)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$1.run(BowlerStudioMenu.java:167)
```

madhephaestus, 2020-08-04:
the bug was in the paralell robots definition, pushed a fix there.

madhephaestus, 2020-08-04:
gist.github.com/bcb4760a449190206170
org.eclipse.jgit.api.errors.InvalidRemoteException: Invalid remote: origin

that was the old carl the robot, before i moved it to its own repo. the paralell arm code was very old and untouched for a long time.

closed 2020-08-04 by Octogonapus