CommonWealthRobotics/BowlerStudio #212: Build 1.10.3 com.neuronrobotics.sdk.addons.kinematics.parallel.ParallelGroup at line 43

arop0191, 2020-08-07:
Auto Reported Issue 
BowlerStudio Build 1.10.3
BowlerKernel 0.62.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.RuntimeException: FK lim must be possible, one limb must not have a reference to another
	at com.neuronrobotics.sdk.addons.kinematics.parallel.ParallelGroup.getFKLimb(ParallelGroup.java:43)
	at com.neuronrobotics.sdk.addons.kinematics.parallel.ParallelGroup.forwardKinematics(ParallelGroup.java:230)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.getCurrentTaskSpaceTransform(AbstractKinematicsNR.java:529)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.firePoseUpdate(AbstractKinematicsNR.java:779)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.getCurrentJointSpaceVector(AbstractKinematicsNR.java:560)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.getCurrentTaskSpaceTransform(AbstractKinematicsNR.java:529)
	at com.neuronrobotics.bowlerstudio.creature.JogWidget.<init>(JogWidget.java:215)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.loadSingleLimb(MobleBaseMenueFactory.java:819)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.loadLimbs(MobleBaseMenueFactory.java:606)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.load(MobleBaseMenueFactory.java:142)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.finishLoading(CreatureLab.java:162)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.lambda$initializeUI$1(CreatureLab.java:98)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$3(WinApplication.java:177)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.load(MobleBaseMenueFactory.java:423)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.finishLoading(CreatureLab.java:162)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.lambda$initializeUI$1(CreatureLab.java:98)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$3(WinApplication.java:177)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.load(MobleBaseMenueFactory.java:423)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.finishLoading(CreatureLab.java:162)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.lambda$initializeUI$1(CreatureLab.java:98)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$3(WinApplication.java:177)
	at java.lang.Thread.run(Thread.java:748)

```

arop0191, 2020-08-07:
Please im really new to this any help would be Awesome

arop0191, 2020-08-07:
Auto Reported Issue 
BowlerStudio Build 1.10.3
BowlerKernel 0.62.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.RuntimeException: FK lim must be possible, one limb must not have a reference to another
	at com.neuronrobotics.sdk.addons.kinematics.parallel.ParallelGroup.getFKLimb(ParallelGroup.java:43)
	at com.neuronrobotics.sdk.addons.kinematics.parallel.ParallelGroup.forwardKinematics(ParallelGroup.java:230)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.getCurrentTaskSpaceTransform(AbstractKinematicsNR.java:529)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.firePoseUpdate(AbstractKinematicsNR.java:779)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.getCurrentJointSpaceVector(AbstractKinematicsNR.java:560)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.getCurrentTaskSpaceTransform(AbstractKinematicsNR.java:529)
	at com.neuronrobotics.bowlerstudio.creature.JogWidget.<init>(JogWidget.java:215)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.loadSingleLimb(MobleBaseMenueFactory.java:819)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.loadLimbs(MobleBaseMenueFactory.java:606)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.load(MobleBaseMenueFactory.java:142)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.finishLoading(CreatureLab.java:162)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.lambda$initializeUI$1(CreatureLab.java:98)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$3(WinApplication.java:177)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.load(MobleBaseMenueFactory.java:423)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.finishLoading(CreatureLab.java:162)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.lambda$initializeUI$1(CreatureLab.java:98)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$3(WinApplication.java:177)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.load(MobleBaseMenueFactory.java:423)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.finishLoading(CreatureLab.java:162)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.lambda$initializeUI$1(CreatureLab.java:98)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$3(WinApplication.java:177)
	at java.lang.Thread.run(Thread.java:748)

```

arop0191, 2020-08-07:
Bet Kevin please on this one too

madhephaestus, 2020-08-07:
Hey there, can you give me a bit of info about what you were doing when this went derpy on you?

madhephaestus, 2020-08-07:
Auto Reported Issue 
BowlerStudio Build 1.10.4
BowlerKernel 0.62.0
JavaCad Version: 0.24.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: FK lim must be possible, one limb must not have a reference to another
	at com.neuronrobotics.sdk.addons.kinematics.parallel.ParallelGroup.getFKLimb(ParallelGroup.java:43)
	at com.neuronrobotics.sdk.addons.kinematics.parallel.ParallelGroup.forwardKinematics(ParallelGroup.java:230)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.getCurrentTaskSpaceTransform(AbstractKinematicsNR.java:529)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.firePoseUpdate(AbstractKinematicsNR.java:779)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.getCurrentJointSpaceVector(AbstractKinematicsNR.java:560)
	at com.neuronrobotics.sdk.addons.kinematics.AbstractKinematicsNR.getCurrentTaskSpaceTransform(AbstractKinematicsNR.java:529)
	at com.neuronrobotics.bowlerstudio.creature.JogWidget.<init>(JogWidget.java:215)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.loadSingleLimb(MobleBaseMenueFactory.java:819)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.loadLimbs(MobleBaseMenueFactory.java:606)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.load(MobleBaseMenueFactory.java:142)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.finishLoading(CreatureLab.java:162)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.lambda$1(CreatureLab.java:98)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.load(MobleBaseMenueFactory.java:423)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.finishLoading(CreatureLab.java:162)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.lambda$1(CreatureLab.java:98)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.load(MobleBaseMenueFactory.java:423)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.finishLoading(CreatureLab.java:162)
	at com.neuronrobotics.bowlerstudio.creature.CreatureLab.lambda$1(CreatureLab.java:98)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2020-08-08:
ok, this is a bug in a new feature i just added. Ill fix it and have a release out this afternoon. Sorry for the problems :)

madhephaestus, 2020-08-08:
this should be fixed in the new version, close bowler studio and open it back up.

closed 2020-08-08 by madhephaestus

arop0191, 2020-08-08:
So do you think i can make the openkat with nano and use esp32cam to send
commands on the nano wirelessly and I will have a raspberry pi that run can
the kinematics server. And im using a pca9685 and gyroscope. You think u
can me wiring and the basics guide path.

madhephaestus, 2020-08-08:
so in the smallkat architecture you have an esp32 on the cat, but then no other controller, no pi. The controller is BowlerStudio. This is similar to the Boston Dynamics Spot, the controller is half of the robots brain. Coms for the robot go over Wifi. 

As for reengineering the stack to use the hardware you have, i would suggest that you do the much easier job of switching to the cheap SmallKat style electronics. If you have experience writing C++, modifying the default firmware is easy. if you have no experience, then the best thing for you to do is to follow the instructions for the electronics. 

As for the Nyblbe (The only thing i can find when i google openkat), it's legs are missing a key degree of freedom to walk without "skid steer". This stack would need heavy modification from a professional engineer to make it work on that hardware. 

I recommend following the instructions here: https://hackaday.io/project/164727/instructions and to ask followup questions in the comments here: https://hackaday.io/project/164727-smallkat-an-adorable-dynamics-oriented-robot-cat

commit a73ae9c2 (2020-08-08): close #212

Fix the loading of the non parallel bases