CommonWealthRobotics/BowlerStudio #213: Cannot make a copy of a robot

Octogonapus, 2020-08-09:
I opened https://github.com/madhephaestus/6dofServoArm.git, clicked Make Copy, and got this exception
```
org.kohsuke.github.GHFileNotFoundException: {"message":"Not Found","documentation_url":"https://docs.github.com/rest/reference/gists#get-a-gist"}
	at org.kohsuke.github.Requester.handleApiError(Requester.java:900)
	at org.kohsuke.github.Requester._to(Requester.java:415)
	at org.kohsuke.github.Requester.to(Requester.java:353)
	at org.kohsuke.github.GitHub.getGist(GitHub.java:835)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fork(ScriptingEngine.java:1411)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingWebWidget$2.run(ScriptingWebWidget.java:112)
Caused by: java.io.FileNotFoundException: https://api.github.com/gists/null
	at sun.reflect.GeneratedConstructorAccessor54.newInstance(Unknown Source)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1950)
	at sun.net.www.protocol.http.HttpURLConnection$10.run(HttpURLConnection.java:1945)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.net.www.protocol.http.HttpURLConnection.getChainedException(HttpURLConnection.java:1944)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1514)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1498)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getInputStream(HttpsURLConnectionImpl.java:268)
	at org.kohsuke.github.Requester.parse(Requester.java:818)
	at org.kohsuke.github.Requester.parse(Requester.java:782)
	at org.kohsuke.github.Requester._to(Requester.java:393)
	... 4 more
Caused by: java.io.FileNotFoundException: https://api.github.com/gists/null
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1896)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1498)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:352)
	at org.kohsuke.github.Requester.parse(Requester.java:790)
	... 6 more
```

madhephaestus, 2020-08-10:
Can not reproduce. 

Check your cache and the state of the Git signing?

Octogonapus, 2020-08-11:
Still happening.

madhephaestus, 2020-08-12:
OOOH, you are using the make-a-copy button on the XML. Ill fix that bug, its expecting gists there.

To fork a robot, open the robot and use the Make a copy of creature button in the creature lab ui.