CommonWealthRobotics/BowlerStudio #225: Build 1.16.0 com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget$2 at line 294

javatechs, 2021-01-01:
Auto Reported Issue 
BowlerStudio Build 1.16.0
BowlerKernel 0.64.0
JavaCad Version: 0.25.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget$2.lambda$run$0(ScriptingFileWidget.java:294)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.sun.glass.ui.Application.reportException(Application.java:453)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.threed.BowlerStudio3dEngine.lambda$static$0(BowlerStudio3dEngine.java:218)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

javatechs, 2021-01-01:
With this code:
/**
 * 
 */
@Grab(group='org.python', module='jython', version='2.7.2')
import org.python.util.PythonInterpreter;

x= "test"
println "Running LoadGameController"
PythonInterpreter pi = new PythonInterpreter();
pi.exec("from pymodule import square");

javatechs, 2021-01-02:
Auto Reported Issue 
BowlerStudio Build 1.16.0
BowlerKernel 0.64.0
JavaCad Version: 0.25.0
Java-Bowler Version: 3.32.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget$2.lambda$run$0(ScriptingFileWidget.java:294)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.sun.glass.ui.Application.reportException(Application.java:453)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.threed.BowlerStudio3dEngine.lambda$static$0(BowlerStudio3dEngine.java:218)
	at com.sun.javafx.application.PlatformImpl.lambda$null$5(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$6(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$10(GtkApplication.java:245)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2021-01-02:
the Jython runtime is installed already. Just run the .py script like the .groovy script and it will be executed by jython

madhephaestus, 2021-01-02:
also, a link to the source that caused the exception would be helpful

madhephaestus, 2021-01-02:
at least this issue is fixed in 1.17.1

closed 2021-01-02 by madhephaestus

javatechs, 2021-01-02:
**This code (No link. Not checked in):**
```
import org.python.util.PythonInterpreter;

PythonInterpreter pi = new PythonInterpreter();
pi.exec("import roslibpy");
```
**Got this exception:**
```
Writing file contents
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named roslibpy

	at org.python.core.Py.ImportError(Py.java:304)
	at org.python.core.imp.import_first(imp.java:755)
	at org.python.core.imp.import_module_level(imp.java:837)
	at org.python.core.imp.importName(imp.java:917)
	at org.python.core.ImportFunction.__call__(__builtin__.java:1220)
	at org.python.core.PyObject.__call__(PyObject.java:357)
	at org.python.core.__builtin__.__import__(__builtin__.java:1173)
	at org.python.core.imp.importOne(imp.java:936)
	at org.python.pycode._pyx19.f$0(<string>:1)
	at org.python.pycode._pyx19.call_function(<string>)
	at org.python.core.PyTableCode.call(PyTableCode.java:165)
	at org.python.core.PyCode.call(PyCode.java:18)
	at org.python.core.Py.runCode(Py.java:1275)
	at org.python.core.Py.exec(Py.java:1319)
	at org.python.util.PythonInterpreter.exec(PythonInterpreter.java:206)
	at org.python.util.PythonInterpreter$exec.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:116)
	at aa.run(aa.groovy:4)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:61)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:178)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineFileScriptRun(ScriptingEngine.java:634)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget$2.run(ScriptingFileWidget.java:277)
```
**AND this popup:**

[image]

madhephaestus, 2021-01-02:
yeah, as i suspected, the libs installed with pip are not availible to jython.

You could try to pull all the source into one folder and try to run the file using local file references for the libs... maybe?

javatechs, 2021-01-02:
Note: The pop-up seems inaccurate and inappropriate.