CommonWealthRobotics/BowlerStudio #233: Build 1.24.0 com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine at line 1256

samsamam, 2021-03-01:
Auto Reported Issue 
BowlerStudio Build 1.24.0
BowlerKernel 0.65.0
JavaCad Version: 0.25.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.locateGit(ScriptingEngine.java:1256)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkOwner(ScriptingEngine.java:1371)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:82)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:131)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:126)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$7.run(BowlerStudioMenu.java:798)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:129)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:126)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$7.run(BowlerStudioMenu.java:798)

```

samsamam, 2021-03-02:
Auto Reported Issue 
BowlerStudio Build 1.24.0
BowlerKernel 0.65.0
JavaCad Version: 0.25.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.locateGit(ScriptingEngine.java:1256)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkOwner(ScriptingEngine.java:1371)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:82)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:131)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:126)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$7.run(BowlerStudioMenu.java:798)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:129)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:126)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$7.run(BowlerStudioMenu.java:798)

```

samsamam, 2021-03-02:
Auto Reported Issue 
BowlerStudio Build 1.25.1
BowlerKernel 0.65.0
JavaCad Version: 0.25.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.locateGit(ScriptingEngine.java:1256)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkOwner(ScriptingEngine.java:1371)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:82)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:131)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:126)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$7.run(BowlerStudioMenu.java:798)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:129)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:126)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$7.run(BowlerStudioMenu.java:798)

```

closed 2021-03-02 by samsamam

madhephaestus, 2021-03-02:
@samsamam I just pushed an update to address this. Close BowlerStudio and open it to get the update. It was having a problem with your non-git file. BowlerStudio assumes files opened in it are part of a git repo and gets confused sometimes when they are not. I believe i fixed it though, so you should be good to go!

commit a29dfa2b (2021-03-02): close #233

You should be able to open a non git file