CommonWealthRobotics/BowlerStudio #235: Build 1.26.1 com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab at line 268

madhephaestus, 2021-03-10:
Auto Reported Issue 
BowlerStudio Build 1.26.1
BowlerKernel 0.65.0
JavaCad Version: 0.25.0
Java-Bowler Version: 3.33.0
Java Version: 11.0.8
JavaFX Version: 14.0.2

OS = Linux amd64 x64
```
java.lang.NoClassDefFoundError: javafx/embed/swing/MySwingNode
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:268)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:126)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:539)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$8$1$1.run(BowlerStudioMenu.java:871)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at java.base/java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1055)
	at java.base/java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1050)
	at java.base/java.lang.Thread.dispatchUncaughtException(Thread.java:2002)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:129)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:126)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:539)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$8$1$1.run(BowlerStudioMenu.java:871)

```

JansenSmith, 2023-01-12:
Auto Reported Issue 
BowlerStudio Build 1.45.0
BowlerKernel 1.44.1
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 11.0.9
JavaFX Version: 15.0.0

OS = Windows 10 amd64 x64
```
java.lang.NoClassDefFoundError: javafx/embed/swing/MySwingNode
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:268)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:528)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$8$1$1.run(BowlerStudioMenu.java:914)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at java.base/java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1055)
	at java.base/java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1050)
	at java.base/java.lang.Thread.dispatchUncaughtException(Thread.java:2002)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:129)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:528)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$8$1$1.run(BowlerStudioMenu.java:914)

```

JansenSmith, 2023-01-12:
Success Login JansenSmith

Using my version of configuration database: https://github.com/JansenSmith/BowlerStudioConfiguration.git
Using my version of configuration database: https://github.com/JansenSmith/BowlerStudioConfiguration.git
 50.0% Pull BowlerStudioConfiguration  1.0 of task Pulling 
DONE!  Pulling  BowlerStudioConfiguration  Pull
 50.0% Pull BowlerStudioConfiguration  1.0 of task Pulling 
DONE!  Pulling  BowlerStudioConfiguration  Pull
Setting workspace:https://github.com/madhephaestus/carl-the-hexapod.git to [https://github.com/madhephaestus/carl-the-hexapod.git, 1668732826186] with previous being [https://github.com/madhephaestus/carl-the-hexapod.git, 1668732826186]
Setting workspace:https://github.com/JansenSmith/Modular-Maze.git to [JansenSmith/Modular-Maze, 1660419063183] with previous being [JansenSmith/Modular-Maze, 1660419063183]
Setting workspace:https://github.com/madhephaestus/potbot.git to [madhephaestus/potbot, 1673547357151] with previous being [madhephaestus/potbot, 1673547357151]
Setting workspace:https://gist.github.com/fb4cf429372deeb36f52.git to [https://gist.github.com/fb4cf429372deeb36f52.git, 1668732828973] with previous being [https://gist.github.com/fb4cf429372deeb36f52.git, 1668732828973]
Setting workspace:https://github.com/Halloween2022RodPuppet/RodPuppetKinematics.git to [Halloween2022RodPuppet/RodPuppetKinematics, 1665946521128] with previous being [Halloween2022RodPuppet/RodPuppetKinematics, 1665946521128]
Setting workspace:https://gist.github.com/0c31993200bb48a12613ce9b48a29b22.git to [https://gist.github.com/0c31993200bb48a12613ce9b48a29b22.git, 1668732828100] with previous being [https://gist.github.com/0c31993200bb48a12613ce9b48a29b22.git, 1668732828100]
Setting workspace:https://github.com/JansenSmith/technocopiaefficiency.git to [JansenSmith/technocopiaefficiency, 1670698480763] with previous being [JansenSmith/technocopiaefficiency, 1670698480763]
Setting workspace:https://github.com/JansenSmith/WalkTest_JansenSmith.git to [https://github.com/JansenSmith/WalkTest_JansenSmith.git, 1668736837032] with previous being [https://github.com/JansenSmith/WalkTest_JansenSmith.git, 1668736837032]
Setting workspace:https://github.com/JansenSmith/MoveLink_JansenSmith.git to [JansenSmith/MoveLink_JansenSmith, 1668730733954] with previous being [JansenSmith/MoveLink_JansenSmith, 1668730733954]
Setting workspace:https://github.com/JansenSmith/castledicetower.git to [https://github.com/JansenSmith/castledicetower.git, 1663892198300] with previous being [https://github.com/JansenSmith/castledicetower.git, 1663892198300]
Setting workspace:https://github.com/JansenSmith/TheCalledScript_JansenSmith.git to [https://github.com/JansenSmith/TheCalledScript_JansenSmith.git, 1659213442912] with previous being [https://github.com/JansenSmith/TheCalledScript_JansenSmith.git, 1659213442912]
Setting workspace:https://github.com/JansenSmith/FrontRightCad_JansenSmith.git to [https://github.com/JansenSmith/FrontRightCad_JansenSmith.git, 1662761506002] with previous being [https://github.com/JansenSmith/FrontRightCad_JansenSmith.git, 1662761506002]
Setting workspace:https://gist.github.com/b1bc6c6cb6058ec81dbc8ede0e851299.git to [https://gist.github.com/b1bc6c6cb6058ec81dbc8ede0e851299.git, 1667523161395] with previous being [https://gist.github.com/b1bc6c6cb6058ec81dbc8ede0e851299.git, 1667523161395]
Setting workspace:https://github.com/madhephaestus/basicincomeviatransactiontax.git to [madhephaestus/basicincomeviatransactiontax, 1670698278731] with previous being [madhephaestus/basicincomeviatransactiontax, 1670698278731]
Setting workspace:https://github.com/JansenSmith/JansenWalker_SeriesParallelHybrid.git to [https://github.com/JansenSmith/JansenWalker_SeriesParallelHybrid.git, 1660421700345] with previous being [https://github.com/JansenSmith/JansenWalker_SeriesParallelHybrid.git, 1660421700345]
Setting workspace:https://github.com/JansenSmith/CastleDiceTower.git to [https://github.com/JansenSmith/CastleDiceTower.git, 1664494181935] with previous being [https://github.com/JansenSmith/CastleDiceTower.git, 1664494181935]
Setting workspace:https://github.com/JansenSmith/41-dover.git to [JansenSmith/41-dover, 1671300568047] with previous being [JansenSmith/41-dover, 1671300568047]
Setting workspace:https://github.com/OperationSmallKat/SmallKat_V2.git to [https://github.com/OperationSmallKat/SmallKat_V2.git, 1668730873574] with previous being [https://github.com/OperationSmallKat/SmallKat_V2.git, 1668730873574]
Setting workspace:https://github.com/JansenSmith/MoveLimb_JansenSmith.git to [https://github.com/JansenSmith/MoveLimb_JansenSmith.git, 1668126701945] with previous being [https://github.com/JansenSmith/MoveLimb_JansenSmith.git, 1668126701945]
Setting workspace:https://github.com/JansenSmith/gistGetTipLocations_JansenSmith.git to [JansenSmith/gistGetTipLocations_JansenSmith, 1668127961967] with previous being [JansenSmith/gistGetTipLocations_JansenSmith, 1668127961967]
Setting katapult:gameControllerNames to [Play, Game, X-Box, Dragon] with previous being [Play, Game, X-Box, Dragon]
Setting katapult:linkDeviceName to hephaestus-16c00486 with previous being hephaestus-16c00486
Setting katapult:robotName to HephaestusArmV2_Janz with previous being HephaestusArmV2_Janz
Setting katapult:robotGitFile to HephaestusArmV2_Janz.xml with previous being HephaestusArmV2_Janz.xml
[... cut ...]
java.lang.NoClassDefFoundError: javafx/embed/swing/MySwingNode
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:268)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:528)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$8$1$1.run(BowlerStudioMenu.java:914)

madhephaestus, 2023-01-13:
@JansenSmith that eclipse project is pointing at a very old jar file. try deleting the project in eclipse, and deleting the gitcache, then opening BowlerStudio and opening the project again. 

Alternatly you can right click on the project, Properties->Java Build Path -> Libraries right click on the BowlerStudio.jar and remove. Then click add External Jar and select the latest BowlerStudio.jar file.

closed 2023-01-13 by madhephaestus