CommonWealthRobotics/BowlerStudio #239: Build 1.28.1 org.eclipse.jgit.api.CreateBranchCommand at line 279

madhephaestus, 2021-05-30:
Auto Reported Issue 
BowlerStudio Build 1.28.1
BowlerKernel 0.65.0
JavaCad Version: 0.25.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.RefNotFoundException: Ref origin/puppet cannot be resolved
	at org.eclipse.jgit.api.CreateBranchCommand.getStartPointObjectId(CreateBranchCommand.java:279)
	at org.eclipse.jgit.api.CreateBranchCommand.call(CreateBranchCommand.java:132)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1109)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1075)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.switchToThisNewBranch(BowlerStudioMenu.java:764)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.run(BowlerStudioMenu.java:752)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1151)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1075)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.switchToThisNewBranch(BowlerStudioMenu.java:764)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.run(BowlerStudioMenu.java:752)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:114)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:293)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

madhephaestus, 2021-09-29:
Auto Reported Issue 
BowlerStudio Build 1.36.4
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.RefNotFoundException: Ref origin/puppet cannot be resolved
	at org.eclipse.jgit.api.CreateBranchCommand.getStartPointObjectId(CreateBranchCommand.java:279)
	at org.eclipse.jgit.api.CreateBranchCommand.call(CreateBranchCommand.java:132)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1257)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1220)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.switchToThisNewBranch(BowlerStudioMenu.java:777)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.run(BowlerStudioMenu.java:765)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1305)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1220)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.switchToThisNewBranch(BowlerStudioMenu.java:777)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$6.run(BowlerStudioMenu.java:765)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:118)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:303)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

keionbis, 2021-11-17:
Auto Reported Issue 
BowlerStudio Build 1.36.6
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.RefNotFoundException: Ref origin/1.25.1 cannot be resolved
	at org.eclipse.jgit.api.CreateBranchCommand.getStartPointObjectId(CreateBranchCommand.java:279)
	at org.eclipse.jgit.api.CreateBranchCommand.call(CreateBranchCommand.java:132)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1257)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1428)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:761)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadFile(AssetFactory.java:52)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAsset(AssetFactory.java:84)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadIcon(AssetFactory.java:127)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:85)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:131)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:146)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1265)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1305)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1428)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:761)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadFile(AssetFactory.java:52)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAsset(AssetFactory.java:84)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadIcon(AssetFactory.java:127)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:85)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:131)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:146)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1265)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:118)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:303)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

madhephaestus, 2021-11-18:
i forgot to make the new tagged assets

keionbis, 2021-11-29:
Auto Reported Issue 
BowlerStudio Build 1.36.6
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.RefNotFoundException: Ref origin/1.25.1 cannot be resolved
	at org.eclipse.jgit.api.CreateBranchCommand.getStartPointObjectId(CreateBranchCommand.java:279)
	at org.eclipse.jgit.api.CreateBranchCommand.call(CreateBranchCommand.java:132)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1257)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1428)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:761)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadFile(AssetFactory.java:52)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAsset(AssetFactory.java:84)
	at com.neuronrobotics.bowlerstudio.scripting.ExternalEditorController.<init>(ExternalEditorController.java:30)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:181)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:77)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:131)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:146)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1265)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1305)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1428)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:761)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadFile(AssetFactory.java:52)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAsset(AssetFactory.java:84)
	at com.neuronrobotics.bowlerstudio.scripting.ExternalEditorController.<init>(ExternalEditorController.java:30)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:181)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:77)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:131)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:133)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:146)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1265)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:118)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:303)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

JansenSmith, 2022-03-08:
Auto Reported Issue 
BowlerStudio Build 1.41.4
BowlerKernel 1.40.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.RefNotFoundException: Ref origin/deploy cannot be resolved
	at org.eclipse.jgit.api.CreateBranchCommand.getStartPointObjectId(CreateBranchCommand.java:279)
	at org.eclipse.jgit.api.CreateBranchCommand.call(CreateBranchCommand.java:132)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.shallowCheckout(ScriptingEngine.java:1516)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1458)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1625)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:952)
	at com.neuronrobotics.bowlerstudio.Tutorial.getHomeUrl(Tutorial.java:27)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:406)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1498)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1625)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:952)
	at com.neuronrobotics.bowlerstudio.Tutorial.getHomeUrl(Tutorial.java:27)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:406)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:124)
	at com.neuronrobotics.bowlerstudio.PsudoSplash.setMessage(PsudoSplash.java:193)
	at com.neuronrobotics.bowlerstudio.SplashManager.renderSplashFrame(SplashManager.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.renderSplashFrame(BowlerStudio.java:521)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:260)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

keionbis, 2022-03-14:
Auto Reported Issue 
BowlerStudio Build 1.42.0
BowlerKernel 1.40.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.RefNotFoundException: Ref origin/deploy cannot be resolved
	at org.eclipse.jgit.api.CreateBranchCommand.getStartPointObjectId(CreateBranchCommand.java:279)
	at org.eclipse.jgit.api.CreateBranchCommand.call(CreateBranchCommand.java:132)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.shallowCheckout(ScriptingEngine.java:1516)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1458)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1625)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:952)
	at com.neuronrobotics.bowlerstudio.Tutorial.getHomeUrl(Tutorial.java:27)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:406)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1498)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1625)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:952)
	at com.neuronrobotics.bowlerstudio.Tutorial.getHomeUrl(Tutorial.java:27)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:406)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:124)
	at com.neuronrobotics.bowlerstudio.PsudoSplash.setMessage(PsudoSplash.java:193)
	at com.neuronrobotics.bowlerstudio.SplashManager.renderSplashFrame(SplashManager.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.renderSplashFrame(BowlerStudio.java:521)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:260)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

madhephaestus, 2023-02-24:
Auto Reported Issue 
BowlerStudio Build source
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.RefNotFoundException: Ref origin/source cannot be resolved
	at org.eclipse.jgit.api.CreateBranchCommand.getStartPointObjectId(CreateBranchCommand.java:279)
	at org.eclipse.jgit.api.CreateBranchCommand.call(CreateBranchCommand.java:132)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.shallowCheckout(ScriptingEngine.java:1494)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1437)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1604)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.filesInGit(ScriptingEngine.java:705)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAllAssets(AssetFactory.java:155)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.setGitSource(AssetFactory.java:150)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:404)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1477)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1604)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.filesInGit(ScriptingEngine.java:705)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAllAssets(AssetFactory.java:155)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.setGitSource(AssetFactory.java:150)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:404)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:124)
	at com.neuronrobotics.bowlerstudio.PsudoSplash.setMessage(PsudoSplash.java:193)
	at com.neuronrobotics.bowlerstudio.SplashManager.renderSplashFrame(SplashManager.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.renderSplashFrame(BowlerStudio.java:506)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:263)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```