CommonWealthRobotics/BowlerStudio #241: Build 1.29.0 com.neuronrobotics.bowlerstudio.BowlerStudioController$2 at line 136

convalation, 2021-07-16:
Auto Reported Issue 
BowlerStudio Build 1.29.0
BowlerKernel 0.65.1
JavaCad Version: 0.25.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.BowlerStudioController$2.run(BowlerStudioController.java:136)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:130)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:145)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1252)
	at java.lang.Thread.run(Thread.java:748)

```

convalation, 2021-07-16:
Auto Reported Issue 
BowlerStudio Build 1.29.0
BowlerKernel 0.65.1
JavaCad Version: 0.25.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.BowlerStudioController$2.run(BowlerStudioController.java:136)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:130)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:145)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1252)
	at java.lang.Thread.run(Thread.java:748)

```

acamilo, 2021-08-05:
Auto Reported Issue 
BowlerStudio Build 1.30.0
BowlerKernel 0.65.1
JavaCad Version: 0.25.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.BowlerStudioController$2.run(BowlerStudioController.java:136)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:130)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:145)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1252)
	at java.lang.Thread.run(Thread.java:748)

```

JansenSmith, 2021-08-08:
Auto Reported Issue 
BowlerStudio Build 1.30.0
BowlerKernel 0.65.1
JavaCad Version: 0.25.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_302
JavaFX Version: 8.0.302

OS = Windows 10 amd64 x64
```
java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.BowlerStudioController$2.run(BowlerStudioController.java:136)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:130)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:348)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.createFileTab(BowlerStudioMenu.java:145)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$13$2$1.run(BowlerStudioMenu.java:1252)
	at java.lang.Thread.run(Thread.java:748)

```

commit f421b180 (2021-08-05): close #241
closed 2021-09-02 by madhephaestus