CommonWealthRobotics/BowlerStudio #249: Build 1.34.0 com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine at line 220

vermontolympian, 2021-08-30:
Auto Reported Issue 
BowlerStudio Build 1.34.0
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:220)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:194)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkOwner(ScriptingEngine.java:1541)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:355)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.lambda$makeTimeoutThread$0(ScriptingEngine.java:224)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:118)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:303)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

madhephaestus, 2021-09-01:
Auto Reported Issue 
BowlerStudio Build 1.34.0
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:220)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:194)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:926)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:903)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1129)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1180)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$pull$0.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:116)
	at LabCodeRepoSetupMain.createTeamRepo(LabCodeRepoSetupMain.groovy:311)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.codehaus.groovy.reflection.CachedMethod.invoke(CachedMethod.java:90)
	at org.codehaus.groovy.runtime.callsite.StaticMetaMethodSite$StaticMetaMethodSiteNoUnwrapNoCoerce.invoke(StaticMetaMethodSite.java:148)
	at org.codehaus.groovy.runtime.callsite.StaticMetaMethodSite.callStatic(StaticMetaMethodSite.java:99)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCallStatic(CallSiteArray.java:53)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callStatic(AbstractCallSite.java:157)
	at LabCodeRepoSetupMain.processAllRepositories(LabCodeRepoSetupMain.groovy:224)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.codehaus.groovy.reflection.CachedMethod.invoke(CachedMethod.java:90)
	at org.codehaus.groovy.runtime.callsite.StaticMetaMethodSite$StaticMetaMethodSiteNoUnwrapNoCoerce.invoke(StaticMetaMethodSite.java:148)
	at org.codehaus.groovy.runtime.callsite.StaticMetaMethodSite.callStatic(StaticMetaMethodSite.java:99)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCallStatic(CallSiteArray.java:53)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callStatic(AbstractCallSite.java:157)
	at LabCodeRepoSetupMain.start(LabCodeRepoSetupMain.groovy:106)
	at LabCodeRepoSetupMain.run(LabCodeRepoSetupMain.groovy:429)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:61)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:270)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineFileScriptRun(ScriptingEngine.java:731)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget$2.run(ScriptingFileWidget.java:291)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.lambda$0(ScriptingEngine.java:224)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:118)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:303)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

madhephaestus, 2021-09-01:
Auto Reported Issue 
BowlerStudio Build 1.34.0
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:220)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:194)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:926)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:903)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1129)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1180)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine$pull$0.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:45)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:108)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:116)
	at LabCodeRepoSetupMain.createTeamRepo(LabCodeRepoSetupMain.groovy:314)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.codehaus.groovy.reflection.CachedMethod.invoke(CachedMethod.java:90)
	at org.codehaus.groovy.runtime.callsite.StaticMetaMethodSite$StaticMetaMethodSiteNoUnwrapNoCoerce.invoke(StaticMetaMethodSite.java:148)
	at org.codehaus.groovy.runtime.callsite.StaticMetaMethodSite.callStatic(StaticMetaMethodSite.java:99)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCallStatic(CallSiteArray.java:53)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callStatic(AbstractCallSite.java:157)
	at LabCodeRepoSetupMain.processAllRepositories(LabCodeRepoSetupMain.groovy:224)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.codehaus.groovy.reflection.CachedMethod.invoke(CachedMethod.java:90)
	at org.codehaus.groovy.runtime.callsite.StaticMetaMethodSite$StaticMetaMethodSiteNoUnwrapNoCoerce.invoke(StaticMetaMethodSite.java:148)
	at org.codehaus.groovy.runtime.callsite.StaticMetaMethodSite.callStatic(StaticMetaMethodSite.java:99)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCallStatic(CallSiteArray.java:53)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callStatic(AbstractCallSite.java:157)
	at LabCodeRepoSetupMain.start(LabCodeRepoSetupMain.groovy:106)
	at LabCodeRepoSetupMain.run(LabCodeRepoSetupMain.groovy:432)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:61)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:74)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:270)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineFileScriptRun(ScriptingEngine.java:731)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget$2.run(ScriptingFileWidget.java:291)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.lambda$0(ScriptingEngine.java:224)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:118)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:303)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

closed 2021-09-02 by madhephaestus

Gregory-Marshall-2023, 2021-09-13:
Auto Reported Issue 
BowlerStudio Build 1.36.0
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:220)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:194)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.commit(ScriptingEngine.java:581)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pushCodeToGit(ScriptingEngine.java:633)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pushCodeToGit(ScriptingEngine.java:548)
	at com.neuronrobotics.bowlerstudio.scripting.EclipseExternalEditor.lambda$launch$0(EclipseExternalEditor.java:99)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.lambda$makeTimeoutThread$0(ScriptingEngine.java:224)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:118)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:303)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

Gregory-Marshall, 2021-09-14:
Auto Reported Issue 
BowlerStudio Build 1.36.0
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:220)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:194)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1080)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1183)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$1.run(BowlerStudioMenu.java:166)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.lambda$makeTimeoutThread$0(ScriptingEngine.java:224)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:118)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:303)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```