CommonWealthRobotics/BowlerStudio #251: Build 1.36.1 com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine at line 221

Gregory-Marshall, 2021-09-14:
Auto Reported Issue 
BowlerStudio Build 1.36.1
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);|nhttps://github.com/Hephaestus-Arm/HephaestusArm2.git
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:221)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:194)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1082)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1188)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$1.run(BowlerStudioMenu.java:166)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.lambda$makeTimeoutThread$0(ScriptingEngine.java:225)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:118)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:303)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

Gregory-Marshall, 2021-09-16:
Auto Reported Issue 
BowlerStudio Build 1.36.3
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/Hephaestus-Arm/HephaestusArm2.git

	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:221)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:194)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1082)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:790)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:752)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseCadManager.setGitCadEngine(MobileBaseCadManager.java:901)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseCadManager.setDefaultLinkLevelCadEngine(MobileBaseCadManager.java:848)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseCadManager.generateBody(MobileBaseCadManager.java:388)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseCadManager$6.run(MobileBaseCadManager.java:824)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.lambda$makeTimeoutThread$0(ScriptingEngine.java:225)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:118)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:303)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

esuts, 2021-11-14:
Auto Reported Issue 
BowlerStudio Build 1.36.4
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://gist.github.com/b8e78fdb8f0ec1e9b746ce88e4900b9f.git

	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:221)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:194)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.locateGit(ScriptingEngine.java:1448)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fork(ScriptingEngine.java:1640)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingWebWidget.doFork(ScriptingWebWidget.java:141)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingWebWidget.access$500(ScriptingWebWidget.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingWebWidget$2.run(ScriptingWebWidget.java:109)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:198)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:79)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.lambda$makeTimeoutThread$0(ScriptingEngine.java:225)
	at java.lang.Thread.run(Thread.java:748)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:118)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:303)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

madhephaestus, 2021-11-14:
Auto Reported Issue 
BowlerStudio Build 1.36.5
BowlerKernel 0.66.0
JavaCad Version: 0.27.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/esuts/Inkscape-Plugin-Demo.git

	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:221)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:194)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.locateGit(ScriptingEngine.java:1448)
	at com.neuronrobotics.bowlerstudio.scripting.SVGExternalEditor.launch(SVGExternalEditor.java:30)
	at com.neuronrobotics.bowlerstudio.scripting.ExternalEditorController.lambda$new$0(ExternalEditorController.java:52)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:49)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Node.fireEvent(Node.java:8411)
	at javafx.scene.control.Button.fire(Button.java:185)
	at com.sun.javafx.scene.control.behavior.ButtonBehavior.mouseReleased(ButtonBehavior.java:182)
	at com.sun.javafx.scene.control.skin.BehaviorSkinBase$1.handle(BehaviorSkinBase.java:96)
	at com.sun.javafx.scene.control.skin.BehaviorSkinBase$1.handle(BehaviorSkinBase.java:89)
	at com.sun.javafx.event.CompositeEventHandler$NormalEventHandlerRecord.handleBubblingEvent(CompositeEventHandler.java:218)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:80)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
[... 25 lines cut ...]
[... cut ...]
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

closed 2021-11-15 by madhephaestus