CommonWealthRobotics/BowlerStudio #255: Support committing changes to parameters

Gregory-Marshall, 2021-11-05:
It would be helpful to be able to update the values of parameters in a way that updates within the repo. i.e. the ability to change some parameter on one machine, commit and push changes to that parameter then pull the changes to a different machine.

madhephaestus, 2021-11-05:
you can! a file called "csgDatabase.json " in the root of a reop shoudl provide parameter data for a given file loading. Here is the example in the JavaCadCheatSheed example: https://gist.github.com/madhephaestus/4814b39ee72e9f590757#file-gitignore

madhephaestus, 2021-11-17:
@Gregory-Marshall were you able to make this work, would you like a vid-call to walk through this or any other issues you have been having?

Gregory-Marshall, 2021-11-19:
Got it working, thanks for the help.

closed 2021-12-19 by madhephaestus