CommonWealthRobotics/BowlerStudio #256: Release for 20.04?

keionbis, 2021-11-16:
Can you make an updated release for 20.04

madhephaestus, 2021-11-16:
the any debian instructions still work, and the standalone deb as well. One of these days ill update the default ppa, but its not broken on 20.04 so... the priority  is low ;)

keionbis, 2021-11-17:
Just a note. Installing the dpkg broke all my system inputs lol. Uninstalling through recovery mode brought them back to life. Probably more of some weird conflict or an old package but that was no fun

keionbis, 2021-11-17:
Update following the debian instructions also did the same. Purging bowler fixed it again

madhephaestus, 2021-11-18:
what happens if you just run the bash script? https://github.com/CommonWealthRobotics/BowlerStudio/releases/download/1.24.0/bowlerstudio

keionbis, 2021-11-23:
running the script works and doesnt break my laptop. Just to clarify when installing through the deb pfg it would work fine and i was able to use it but after a reboot i would have no system inputs so i assume its some conflict going on

madhephaestus, 2022-05-14:
I have changed the default "linux" instrutions to be just the script, depricating the deb entirely

closed 2022-05-14 by madhephaestus