CommonWealthRobotics/BowlerStudio #257: Ros 2 support

keionbis, 2021-11-17:
Discussed this with you, just wanted an issue to track it with so i could know the progress

keionbis, 2021-11-17:
Along with this we could consider adding in XML to URDF exporting

madhephaestus, 2021-11-18:
I wrote a URDF loader/exporter for work and the boss has expressed interest in open sourcing it. I think once thats freed up ill take a stab at this

madhephaestus, 2021-12-19:
At work we use this library to bridge our java real-time stack over into ros2 https://github.com/ihmcrobotics/ihmc-java-ros2-communication

madhephaestus, 2022-08-13:
In order for me to make this add-on, i need the following pre-steps completed:

1) make a new project and repo 

2) add me as collaboration 

3) add whatever starting point you have 

4) add a workspace repo with submodules for the ros projects to test the Java code 

5) add a stub workspace with code to mock the tester and provide example of the coms you want.