CommonWealthRobotics/BowlerStudio #265: BowlerStudio doesn't launch on macOS Big Sur 11.6.5

WillAdams, 2022-04-14:
### Installed Version https://github.com/CommonWealthRobotics/BowlerStudio/releases/download/1.24.0/MacOSX-BowlerStudio-1.24.0.zip

### OS type and version number macOS Big Sur 11.6.5

### Expected behavior app launches

### Actual Behavior app bounces once and does not launch

### Steps to reproduce the behavior download and unzip the archive, place the app in Applications, go to System Preferences | Security & Privacy and allow the app to launch, then try launching the file

madhephaestus, 2022-04-20:
The Macos implementation of JavaFX (this application is built on JavaFX) is still a partial implementation. It also just released, and was not availible until recently. I will be looking into this, but Mac has changed some serious things (Depricated OpenGL, the underlayer of JavaFX) so the progress is slow and mostly blocked.

madhephaestus, 2022-05-17:
@WillAdams I have updated the MacOS instructions, it is now a shell script, not an .app In the shell script should be the ability to run on M1 silicon and on 11+ 

I would be much appreciative if you would test the update for me so i can add MacOS 11+ to the supported platforms!

WillAdams, 2022-05-17:
Worked perfectly! I now have Bowler Studio v 1.48.2 running on my MacBook (which is currently still on 11.6.5). Thanks!

madhephaestus, 2022-06-06:
can you set something like Chrome or Firefox as the default browser?

madhephaestus, 2022-06-20:
did this change work for you?