CommonWealthRobotics/BowlerStudio #276: Tripping on null MobileBase

JansenSmith, 2022-05-10:
```
Generating link: 3
java.lang.Exception
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at org.codehaus.groovy.reflection.CachedConstructor.invoke(CachedConstructor.java:77)
	at org.codehaus.groovy.runtime.callsite.ConstructorSite$ConstructorSiteNoUnwrapNoCoerce.callConstructor(ConstructorSite.java:102)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callConstructor(AbstractCallSite.java:186)
	at ThreeDPrintCad$1.generateCad(ThreeDPrintCad.groovy:46)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseCadManager.generateCad(MobileBaseCadManager.java:726)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseCadManager.generateBody(MobileBaseCadManager.java:481)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseCadManager$6.run(MobileBaseCadManager.java:833)
Generating link: 4
java.lang.Exception
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at org.codehaus.groovy.reflection.CachedConstructor.invoke(CachedConstructor.java:77)
	at org.codehaus.groovy.runtime.callsite.ConstructorSite$ConstructorSiteNoUnwrapNoCoerce.callConstructor(ConstructorSite.java:102)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callConstructor(AbstractCallSite.java:186)
	at ThreeDPrintCad$1.generateCad(ThreeDPrintCad.groovy:46)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseCadManager.generateCad(MobileBaseCadManager.java:726)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseCadManager.generateBody(MobileBaseCadManager.java:481)
	at com.neuronrobotics.bowlerstudio.creature.MobileBaseCadManager$6.run(MobileBaseCadManager.java:833)

```

madhephaestus, 2022-05-14:
@JansenSmith can you please commit the cad generator script so i can reproduce this error?

madhephaestus, 2022-05-14:
@JansenSmith this should be fixed, please check on your end?

closed 2022-05-14 by madhephaestus

JansenSmith, 2022-07-09:
@madhephaestus ThreeDPrintCad.groovy is now able to instantiate a MobileBase object as class variable and save as the passed MobileBase.