CommonWealthRobotics/BowlerStudio #278: bug rendering underscores in csg parameters

NotOctogonapus, 2022-06-25:
### Installed Version 
v1.50.1

### OS type and version number
windows 10

I'm working on vitamins in [this PR](https://github.com/madhephaestus/Hardware-Dimensions/pull/25/files)
Some sizes have two underscores in their name (e.g. `1_2_in` for 1/2")
These size names are rendered wrong; you can see the first underscore disappears.
[image]

madhephaestus, 2022-06-25:
hmm i have noticed that too, it seems to be some sort of JavaFX text rendering issue... Ill look into making that render correctly

madhephaestus, 2022-06-25:
See: https://stackoverflow.com/questions/51117694/javafx-cant-use-underscores-in-menuitem

closed 2022-06-25 by madhephaestus

madhephaestus, 2022-06-25:
this is a very old bug and i never bothered to look into a solution, turns out it was an easy fix. Thanks for the prompt!

commit 5c69d4a6 (2022-06-25): Close #278 Set the menu items
								customMenuItem.setMnemonicParsing(false);