CommonWealthRobotics/BowlerStudio #281: Build 1.54.5 org.eclipse.jgit.api.CheckoutCommand at line 340

robbyrozentals, 2022-08-06:
Auto Reported Issue 
BowlerStudio Build 1.54.5
BowlerKernel 1.47.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.33.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.JGitInternalException: Cannot lock /home/robert/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/.git/index. Ensure that no other process has an open file handle on the lock file /home/robert/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/.git/index.lock, then you may delete the lock file and retry.
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:340)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranchLocal(ScriptingEngine.java:1173)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1141)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1107)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1480)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1641)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.filesInGit(ScriptingEngine.java:704)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAllAssets(AssetFactory.java:155)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.setGitSource(AssetFactory.java:150)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:361)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: org.eclipse.jgit.errors.LockFailedException: Cannot lock /home/robert/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/.git/index. Ensure that no other process has an open file handle on the lock file /home/robert/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/.git/index.lock, then you may delete the lock file and retry.
	at org.eclipse.jgit.dircache.DirCache.lock(DirCache.java:251)
	at org.eclipse.jgit.dircache.DirCache.lock(DirCache.java:322)
	at org.eclipse.jgit.dircache.DirCache.lock(DirCache.java:288)
	at org.eclipse.jgit.lib.Repository.lockDirCache(Repository.java:1292)
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:270)
	... 20 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1507)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1641)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.filesInGit(ScriptingEngine.java:704)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAllAssets(AssetFactory.java:155)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.setGitSource(AssetFactory.java:150)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:361)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:125)
	at com.neuronrobotics.bowlerstudio.PsudoSplash.setMessage(PsudoSplash.java:193)
	at com.neuronrobotics.bowlerstudio.SplashManager.renderSplashFrame(SplashManager.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.renderSplashFrame(BowlerStudio.java:472)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

EmmanuelAsumadu, 2023-02-09:
Auto Reported Issue 
BowlerStudio Build 1.74.0
BowlerKernel 1.58.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.JGitInternalException: Cannot lock /home/technocopia/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/.git/index. Ensure that no other process has an open file handle on the lock file /home/technocopia/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/.git/index.lock, then you may delete the lock file and retry.
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:340)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.shallowCheckout(ScriptingEngine.java:1531)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1477)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1646)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.filesInGit(ScriptingEngine.java:705)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAllAssets(AssetFactory.java:155)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.setGitSource(AssetFactory.java:150)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:405)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: org.eclipse.jgit.errors.LockFailedException: Cannot lock /home/technocopia/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/.git/index. Ensure that no other process has an open file handle on the lock file /home/technocopia/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/.git/index.lock, then you may delete the lock file and retry.
	at org.eclipse.jgit.dircache.DirCache.lock(DirCache.java:251)
	at org.eclipse.jgit.dircache.DirCache.lock(DirCache.java:322)
	at org.eclipse.jgit.dircache.DirCache.lock(DirCache.java:288)
	at org.eclipse.jgit.lib.Repository.lockDirCache(Repository.java:1292)
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:270)
	... 18 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1517)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1646)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.filesInGit(ScriptingEngine.java:705)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAllAssets(AssetFactory.java:155)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.setGitSource(AssetFactory.java:150)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:405)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:125)
	at com.neuronrobotics.bowlerstudio.PsudoSplash.setMessage(PsudoSplash.java:193)
	at com.neuronrobotics.bowlerstudio.SplashManager.renderSplashFrame(SplashManager.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.renderSplashFrame(BowlerStudio.java:507)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:263)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
[... cut ...] l.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

EmmanuelAsumadu, 2023-02-09:
Auto Reported Issue 
BowlerStudio Build 1.74.0
BowlerKernel 1.58.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
org.eclipse.jgit.api.errors.JGitInternalException: Cannot lock /home/technocopia/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/.git/index. Ensure that no other process has an open file handle on the lock file /home/technocopia/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/.git/index.lock, then you may delete the lock file and retry.
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:340)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranchLocal(ScriptingEngine.java:1174)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1142)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1108)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1485)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1646)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:957)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadFile(AssetFactory.java:53)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAsset(AssetFactory.java:85)
	at com.neuronrobotics.bowlerstudio.threed.BowlerStudio3dEngine$13.run(BowlerStudio3dEngine.java:946)
Caused by: org.eclipse.jgit.errors.LockFailedException: Cannot lock /home/technocopia/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/.git/index. Ensure that no other process has an open file handle on the lock file /home/technocopia/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/.git/index.lock, then you may delete the lock file and retry.
	at org.eclipse.jgit.dircache.DirCache.lock(DirCache.java:251)
	at org.eclipse.jgit.dircache.DirCache.lock(DirCache.java:322)
	at org.eclipse.jgit.dircache.DirCache.lock(DirCache.java:288)
	at org.eclipse.jgit.lib.Repository.lockDirCache(Repository.java:1292)
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:270)
	... 9 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1512)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1646)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:957)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadFile(AssetFactory.java:53)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAsset(AssetFactory.java:85)
	at com.neuronrobotics.bowlerstudio.threed.BowlerStudio3dEngine$13.run(BowlerStudio3dEngine.java:946)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:125)
	at com.neuronrobotics.bowlerstudio.PsudoSplash.setMessage(PsudoSplash.java:193)
	at com.neuronrobotics.bowlerstudio.SplashManager.renderSplashFrame(SplashManager.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.renderSplashFrame(BowlerStudio.java:507)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:263)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

madhephaestus, 2023-02-09:
try closing bowler studio, erasing `/home/technocopia/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/` and opening it back up again?

JansenSmith, 2023-02-16:
Auto Reported Issue 
BowlerStudio Build 1.74.2
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
org.eclipse.jgit.api.errors.JGitInternalException: Not a DIRC file.
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:340)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.shallowCheckout(ScriptingEngine.java:1490)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1437)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1604)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.filesInGit(ScriptingEngine.java:705)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAllAssets(AssetFactory.java:155)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.setGitSource(AssetFactory.java:150)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:404)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: org.eclipse.jgit.errors.CorruptObjectException: Not a DIRC file.
	at org.eclipse.jgit.dircache.DirCache.readFrom(DirCache.java:480)
	at org.eclipse.jgit.dircache.DirCache.read(DirCache.java:429)
	at org.eclipse.jgit.dircache.DirCache.lock(DirCache.java:254)
	at org.eclipse.jgit.dircache.DirCache.lock(DirCache.java:322)
	at org.eclipse.jgit.dircache.DirCache.lock(DirCache.java:288)
	at org.eclipse.jgit.lib.Repository.lockDirCache(Repository.java:1292)
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:270)
	... 18 more

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:209)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:50)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1477)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1604)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.filesInGit(ScriptingEngine.java:705)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.loadAllAssets(AssetFactory.java:155)
	at com.neuronrobotics.bowlerstudio.assets.AssetFactory.setGitSource(AssetFactory.java:150)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:404)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:40)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.<clinit>(ScriptingEngine.java:124)
	at com.neuronrobotics.bowlerstudio.PsudoSplash.setMessage(PsudoSplash.java:193)
	at com.neuronrobotics.bowlerstudio.SplashManager.renderSplashFrame(SplashManager.java:49)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.renderSplashFrame(BowlerStudio.java:506)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:263)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```