CommonWealthRobotics/BowlerStudio #293: Clone Repo Error

JansenSmith, 2022-11-11:
50.0% Pull MoveLimb_JansenSmith  1.0 of task Pulling 
100%  Pull MoveLimb_JansenSmith  Pulling 
ERROR Creating master in https://github.com/JansenSmith/gistGetTipLocations_JansenSmith.git
org.eclipse.jgit.api.errors.RefNotFoundException: Ref HEAD cannot be resolved
	at org.eclipse.jgit.api.CreateBranchCommand.getStartPointObjectId(CreateBranchCommand.java:279)
	at org.eclipse.jgit.api.CreateBranchCommand.call(CreateBranchCommand.java:132)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranchLocal(ScriptingEngine.java:1165)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1143)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1107)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1348)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1410)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$0(BowlerStudioMenuWorkspace.java:43)
	at java.lang.Thread.run(Thread.java:748)
27310 [Thread-71] WARN org.eclipse.jgit.lib.Repository - close() called when useCnt is already zero for Repository[C:\Users\lenovo\bowler-workspace\gitcache\github.com\JansenSmith\gistGetTipLocations_JansenSmith\.git]
org.eclipse.jgit.api.errors.RefNotFoundException: Ref master cannot be resolved
	at org.eclipse.jgit.api.CheckoutCommand.call(CheckoutCommand.java:256)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranchLocal(ScriptingEngine.java:1173)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1143)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1107)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1348)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1410)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$0(BowlerStudioMenuWorkspace.java:43)
	at java.lang.Thread.run(Thread.java:748)
java.lang.RuntimeException: remoteURI https://github.com/JansenSmith/gistGetTipLocations_JansenSmith.git branch master Ref master cannot be resolved
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1359)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1410)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$0(BowlerStudioMenuWorkspace.java:43)
	at java.lang.Thread.run(Thread.java:748)
java.lang.RuntimeException: remoteURI https://github.com/JansenSmith/gistGetTipLocations_JansenSmith.git branch master remoteURI https://github.com/JansenSmith/gistGetTipLocations_JansenSmith.git branch master Ref master cannot be resolved
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1402)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1410)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$0(BowlerStudioMenuWorkspace.java:43)
	at java.lang.Thread.run(Thread.java:748)
 50.0% Pull gistGetTipLocations_JansenSmith  1.0 of task Pulling 
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.waitForRepo(ScriptingEngine.java:611)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1281)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1410)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$0(BowlerStudioMenuWorkspace.java:48)
	at java.lang.Thread.run(Thread.java:748)
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/JansenSmith/gistGetTipLocations_JansenSmith.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:14)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:381)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:348)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1137)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1107)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1348)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1410)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$0(BowlerStudioMenuWorkspace.java:43)
	at java.lang.Thread.run(Thread.java:748)

Paused pull by another thread, waiting for repo https://github.com/JansenSmith/gistGetTipLocations_JansenSmith.git
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.waitForRepo(ScriptingEngine.java:611)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1281)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1410)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$0(BowlerStudioMenuWorkspace.java:48)
	at java.lang.Thread.run(Thread.java:748)
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/JansenSmith/gistGetTipLocations_JansenSmith.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:14)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:381)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:348)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1137)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1107)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1348)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1410)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$0(BowlerStudioMenuWorkspace.java:43)
	at java.lang.Thread.run(Thread.java:748)

madhephaestus, 2022-11-20:
Erase any repos that were incorrectly crated. The bug that caused the incorrect creation is fixed as of 1.64.1

closed 2022-11-20 by madhephaestus

madhephaestus, 2022-11-20:
for clarity: A pervious bug had been filed because during forks the git object was left open during the creation of the new repo causing a hang. I had added code to close that git object, then tried to reopen it after the repo was made. The solution was to make the new repo first, closing out the git objects, then moving on to cloning the source and updating the origin url for pushing the old code to the newly created empty repo. the exception seen here is the result of cloning the broken repo, the empty one with nothing in it.

madhephaestus, 2022-11-20:
i would like to be able to repair the broken repos, but by that point in the code the context of the source repo is missing and there is no information to push. This exception, if it comes up again, is the result of incorrectly created or initialized repositories.