CommonWealthRobotics/BowlerStudio #339: UnknownHostKey: github.com. RSA key

sammcj, 2023-04-13:
Error when launching the app: `UnknownHostKey: github.com. RSA key`

I'm assuming this relates to https://github.blog/2023-03-23-we-updated-our-rsa-ssh-host-key/

### Installed Version 

Latest (1.77.1)

### OS type and version number

macOS 13.3.1

### Expected behaviour

The app to launch

### Actual Behaviour

The app crashes

### Steps to reproduce the behaviour

Download the app, run it

---

```shell
/Users/samm/bin/java8//bin/java -Dprism.forceGPU=true -jar /Users/samm/bin/BowlerStudioInstall/1.77.1/BowlerStudio.jar

No splash screen availible!
 Splash Rendering 2 Testing Internet
 Splash Rendering 5 Loging In...
Attempt to log in with disk credentials
No login ID found!
 Splash Rendering 20 DL'ing Image Assets
Workspace: /Users/samm/bowler-workspace
org.eclipse.jgit.api.errors.TransportException: git@github.com:madhephaestus/BowlerStudioImageAssets.git: UnknownHostKey: github.com. RSA key fingerprint is d5:2c:63:d9:bc:75:9d:de:b1:4e:36:28:9f:7a:9c:39
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:255)
	at org.eclipse.jgit.api.PullCommand.call(PullCommand.java:296)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1259)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:397)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: org.eclipse.jgit.errors.TransportException: git@github.com:madhephaestus/BowlerStudioImageAssets.git: UnknownHostKey: github.com. RSA key fingerprint is d5:2c:63:d9:bc:75:9d:de:b1:4e:36:28:9f:7a:9c:39
	at org.eclipse.jgit.transport.JschConfigSessionFactory.getSession(JschConfigSessionFactory.java:191)
	at org.eclipse.jgit.transport.SshTransport.getSession(SshTransport.java:140)
	at org.eclipse.jgit.transport.TransportGitSsh$SshFetchConnection.<init>(TransportGitSsh.java:280)
	at org.eclipse.jgit.transport.TransportGitSsh.openFetch(TransportGitSsh.java:170)
	at org.eclipse.jgit.transport.FetchProcess.executeImp(FetchProcess.java:138)
	at org.eclipse.jgit.transport.FetchProcess.execute(FetchProcess.java:124)
	at org.eclipse.jgit.transport.Transport.fetch(Transport.java:1293)
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:244)
	... 14 more
Caused by: com.jcraft.jsch.JSchException: UnknownHostKey: github.com. RSA key fingerprint is d5:2c:63:d9:bc:75:9d:de:b1:4e:36:28:9f:7a:9c:39
	at com.jcraft.jsch.Session.checkHost(Session.java:805)
	at com.jcraft.jsch.Session.connect(Session.java:345)
	at org.eclipse.jgit.transport.JschConfigSessionFactory.getSession(JschConfigSessionFactory.java:145)
	... 21 more
java.lang.RuntimeException: remoteURI https://github.com/madhephaestus/BowlerStudioImageAssets.git branch main git@github.com:madhephaestus/BowlerStudioImageAssets.git: UnknownHostKey: github.com. RSA key fingerprint is d5:2c:63:d9:bc:75:9d:de:b1:4e:36:28:9f:7a:9c:39
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1343)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:397)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
LatestFromGithubLaunch clean exit
```

sammcj, 2023-04-13:
When forcing my ssh known hosts to repopulate the RSA key this is using the following errors remain:

```
~/git/3dprinting/BowlerStudio on development ssh-keyscan -H -t rsa github.com >> ~/.ssh/known_hosts
# github.com:22 SSH-2.0-babeld-f06bbde2

~/git/3dprinting/BowlerStudio on development $JAVA_HOME/bin/java -jar build/libs/BowlerStudio*.jar
No splash screen availible!
 Splash Rendering 2 Testing Internet
 Splash Rendering 5 Loging In...
Attempt to log in with disk credentials
No login ID found!
 Splash Rendering 20 DL'ing Image Assets
Workspace: /Users/samm/bowler-workspace
org.eclipse.jgit.api.errors.TransportException: git@github.com:madhephaestus/BowlerStudioImageAssets.git: java.lang.NullPointerException
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:255)
	at org.eclipse.jgit.api.PullCommand.call(PullCommand.java:296)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1259)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:397)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: org.eclipse.jgit.errors.TransportException: git@github.com:madhephaestus/BowlerStudioImageAssets.git: java.lang.NullPointerException
	at org.eclipse.jgit.transport.JschConfigSessionFactory.getSession(JschConfigSessionFactory.java:191)
	at org.eclipse.jgit.transport.SshTransport.getSession(SshTransport.java:140)
	at org.eclipse.jgit.transport.TransportGitSsh$SshFetchConnection.<init>(TransportGitSsh.java:280)
	at org.eclipse.jgit.transport.TransportGitSsh.openFetch(TransportGitSsh.java:170)
	at org.eclipse.jgit.transport.FetchProcess.executeImp(FetchProcess.java:138)
	at org.eclipse.jgit.transport.FetchProcess.execute(FetchProcess.java:124)
	at org.eclipse.jgit.transport.Transport.fetch(Transport.java:1293)
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:244)
	... 14 more
Caused by: com.jcraft.jsch.JSchException: java.lang.NullPointerException
	at com.jcraft.jsch.Session.updateKeys(Session.java:1214)
	at com.jcraft.jsch.Session.receive_newkeys(Session.java:1098)
	at com.jcraft.jsch.Session.connect(Session.java:367)
	at org.eclipse.jgit.transport.JschConfigSessionFactory.getSession(JschConfigSessionFactory.java:145)
	... 21 more
Caused by: java.lang.NullPointerException
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:264)
	at com.jcraft.jsch.Session.updateKeys(Session.java:1173)
	... 24 more
java.lang.RuntimeException: remoteURI https://github.com/madhephaestus/BowlerStudioImageAssets.git branch main git@github.com:madhephaestus/BowlerStudioImageAssets.git: java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1343)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:397)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
1552 [main] WARN org.eclipse.jgit.lib.Repository - close() called when useCnt is already zero for Repository[/Users/samm/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/.git]
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
[... cut ...] /github.com/madhephaestus/BowlerStudioImageAssets.git branch main remoteURI https://github.com/madhephaestus/BowlerStudioImageAssets.git branch main git@github.com:madhephaestus/BowlerStudioImageAssets.git: java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1360)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:397)
	... 11 more
Exception running application com.neuronrobotics.bowlerstudio.BowlerStudio
```

madhephaestus, 2023-04-13:
I think you need to generate a new key after updating to the latest ssh-keygen, and add that new ssh key to your github profile. Let me know if that fixes this issue?

madhephaestus, 2023-04-13:
Hmm, where did "git@github.com:madhephaestus/BowlerStudioImageAssets.git" come from? In my sources that is always in the https:// form not the git@ form. Did you make that change?

madhephaestus, 2023-04-13:
You could also try erasing `/Users/samm/bowler-workspace/gitcache` and running the release jar again. I dont understand how that repo ended up cloned using SSH, but in any case a quick test would be to erase it and try running again.

sammcj, 2023-04-14:
Thanks for taking the time to reply - I really appreciate that!

I suspect the git problem was related to github changing their RSA key - I fixed that by adding their latest RSA keys to my known hosts as per https://docs.github.com/en/authentication/keeping-your-account-and-data-secure/githubs-ssh-key-fingerprints

I did a completely fresh clone and install however Java still explodes so there's something up.

No expectations on fixing this - it's not a big problem for me - the prioject just looked interesting so I wanted to try it out - completely understand if you want to just close this if I'm a special case 😅 

```shell
$JAVA_HOME/bin/java -jar build/libs/BowlerStudio*.jar
No splash screen availible!
 Splash Rendering 2 Testing Internet
 Splash Rendering 5 Loging In...
Attempt to log in with disk credentials
No login ID found!
 Splash Rendering 20 DL'ing Image Assets
Workspace: /Users/samm/bowler-workspace
Cloning files from: https://github.com/madhephaestus/BowlerStudioImageAssets.git
                to: /Users/samm/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: java.lang.RuntimeException: org.eclipse.jgit.api.errors.TransportException: git@github.com:madhephaestus/BowlerStudioImageAssets.git: java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1598)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:396)
	... 11 more
Caused by: org.eclipse.jgit.api.errors.TransportException: git@github.com:madhephaestus/BowlerStudioImageAssets.git: java.lang.NullPointerException
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:255)
	at org.eclipse.jgit.api.CloneCommand.fetch(CloneCommand.java:305)
	at org.eclipse.jgit.api.CloneCommand.call(CloneCommand.java:204)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepoLocal(ScriptingEngine.java:203)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1577)
	... 12 more
Caused by: org.eclipse.jgit.errors.TransportException: git@github.com:madhephaestus/BowlerStudioImageAssets.git: java.lang.NullPointerException
	at org.eclipse.jgit.transport.JschConfigSessionFactory.getSession(JschConfigSessionFactory.java:191)
	at org.eclipse.jgit.transport.SshTransport.getSession(SshTransport.java:140)
	at org.eclipse.jgit.transport.TransportGitSsh$SshFetchConnection.<init>(TransportGitSsh.java:280)
	at org.eclipse.jgit.transport.TransportGitSsh.openFetch(TransportGitSsh.java:170)
	at org.eclipse.jgit.transport.FetchProcess.executeImp(FetchProcess.java:138)
	at org.eclipse.jgit.transport.FetchProcess.execute(FetchProcess.java:124)
	at org.eclipse.jgit.transport.Transport.fetch(Transport.java:1293)
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:244)
	... 16 more
Caused by: com.jcraft.jsch.JSchException: java.lang.NullPointerException
	at com.jcraft.jsch.Session.updateKeys(Session.java:1214)
	at com.jcraft.jsch.Session.receive_newkeys(Session.java:1098)
	at com.jcraft.jsch.Session.connect(Session.java:367)
	at org.eclipse.jgit.transport.JschConfigSessionFactory.getSession(JschConfigSessionFactory.java:145)
	... 23 more
Caused by: java.lang.NullPointerException
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:264)
	at com.jcraft.jsch.Session.updateKeys(Session.java:1173)
	... 26 more
Exception running application com.neuronrobotics.bowlerstudio.BowlerStudio
```

---

Just FYI as to why git is used rather than HTTPS - my git client is setup to prefer git protocol over https for github.com (this is pretty common when working across a few github enterprise cloud accounts as it helps with authentication to private repos) - so git:// will be coming from my ~/.gitconfig:

```
[url "git@github.com:"]
  insteadOf = https://github.com/
```

madhephaestus, 2023-04-14:
Oh boy, yeah, thats going to mess up this stack :/ I need to process ssh connectetion differently than https and detecte it by the URP passed. I dont know how to get that alteration into my code, but i will look into it. It is good to have a specific config file to change to reproduce this issue. 

I have managed to reproduce that error now, and ill look into what it would take to make this use case work!

madhephaestus, 2023-04-14:
I was able to get the clone to fail-over to the SSH credentials, but since this is not usually a feature thats used (clones using SSH generally happen outside of the BowlerStudio pipeline)

Since the URL is changed at the lowest layer, i cant detect it, just try the HTTPS way, fail, and try the SSH way. 

Turns out there is some bug in ssh cloning too, and thats just not working at all :/

This looks solvable, but not going to be a quick fix like i was hoping...

madhephaestus, 2023-04-14:
WELL! I was able to update the jgit library, change how the SSH client uses the underlaying SSH layer, add fail-retry as ssh all the places where the credential providers are used and was able to get my personal system to work with the given SSH config. I will pass the changes through review and release a fixed version tonight.

sammcj, 2023-04-14:
Interesting! So it was a bug! I'm actually quite glad I wasn't just being an egg with a misconfigured ssh client - that would have been a bit embarrassing 😅

Well done figuring all that out - sounds like a fun one.

When you've got a release (or branch) out I'll give it a go, again no rush at all!

madhephaestus, 2023-04-14:
Nah, embarrassment is all my side im afraid. I was using a 2016 abandon-ware layer to manage the SSH connections for the Git subsystem. 

If you would like build it from source the branch with the fix thats in review is `kh/ssh-override-gitconfig` be sure to grab all the submodules recursively, this one was deep :)

Otherwise it should be through review and published tonight or tomorrow morning.

madhephaestus, 2023-04-14:
For those in the future that are having similar problems, this is the line that was changed: https://github.com/CommonWealthRobotics/bowler-script-kernel/pull/26/files#r1167308324

and the changes in the buildfile for the updated version of jgit: https://github.com/CommonWealthRobotics/bowler-script-kernel/pull/26/files#r1167309104

sammcj, 2023-04-14:
> I was using a 2016 abandon-ware

You and half the projects on Github 😛.

Awesome to see such a response to logging a bug report by the way - it's always a fine line to tread between wanting to share feedback / issues and not coming off like you're expecting someone to fix something they made and provided for free.

---

It looks like now it's trying to log into github.com - which I wouldn't have expected unless it was a private repo:

```
samm at samm-digio-mbp in ~/git/3dprinting/BowlerStudio on kh/ssh-override-gitconfig $JAVA_HOME/bin/java -jar build/libs/BowlerStudio*.jar
No splash screen availible!
 Splash Rendering 2 Testing Internet
 Splash Rendering 5 Loging In...
Attempt to log in with disk credentials
No login ID found!
 Splash Rendering 20 DL'ing Image Assets
Workspace: /Users/samm/bowler-workspace
Cloning files from: https://github.com/madhephaestus/BowlerStudioImageAssets.git
                to: /Users/samm/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets
113 [main] INFO org.apache.sshd.common.util.security.eddsa.EdDSASecurityProviderRegistrar - getOrCreateProvider(EdDSA) created instance of net.i2p.crypto.eddsa.EdDSASecurityProvider
167 [main] INFO org.apache.sshd.common.io.DefaultIoServiceFactoryFactory - No detected/configured IoServiceFactoryFactory using Nio2ServiceFactoryFactory
232 [sshd-JGitSshClient[475835b1]-nio2-thread-1] WARN org.eclipse.jgit.internal.transport.sshd.JGitClientSession - Ssh config JGitClientSession[null@github.com/20.248.137.48:22]: ignoring unknown algorithm 'curve25519-sha256@libssh.org' in KexAlgorithms curve25519-sha256@libssh.org,ecdh-sha2-nistp521,ecdh-sha2-nistp384,ecdh-sha2-nistp256,diffie-hellman-group-exchange-sha256
org.eclipse.jgit.api.errors.TransportException: git@github.com:madhephaestus/BowlerStudioImageAssets.git: Cannot log in at github.com:22
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:224)
	at org.eclipse.jgit.api.CloneCommand.fetch(CloneCommand.java:311)
	at org.eclipse.jgit.api.CloneCommand.call(CloneCommand.java:182)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepoLocalSelectAuth(ScriptingEngine.java:192)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepoLocal(ScriptingEngine.java:212)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1606)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:387)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: org.eclipse.jgit.errors.TransportException: git@github.com:madhephaestus/BowlerStudioImageAssets.git: Cannot log in at github.com:22
	at org.eclipse.jgit.transport.sshd.SshdSession.connect(SshdSession.java:174)
	at org.eclipse.jgit.transport.sshd.SshdSession.connect(SshdSession.java:99)
	at org.eclipse.jgit.transport.sshd.SshdSessionFactory.getSession(SshdSessionFactory.java:235)
	at org.eclipse.jgit.transport.sshd.SshdSessionFactory.getSession(SshdSessionFactory.java:1)
	at org.eclipse.jgit.transport.SshTransport.getSession(SshTransport.java:107)
	at org.eclipse.jgit.transport.TransportGitSsh$SshFetchConnection.<init>(TransportGitSsh.java:281)
	at org.eclipse.jgit.transport.TransportGitSsh.openFetch(TransportGitSsh.java:153)
	at org.eclipse.jgit.transport.FetchProcess.executeImp(FetchProcess.java:142)
	at org.eclipse.jgit.transport.FetchProcess.execute(FetchProcess.java:94)
	at org.eclipse.jgit.transport.Transport.fetch(Transport.java:1309)
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:213)
	... 17 more
Caused by: org.apache.sshd.common.SshException: No more authentication methods available
	at org.apache.sshd.common.future.AbstractSshFuture.verifyResult(AbstractSshFuture.java:126)
	at org.apache.sshd.client.future.DefaultAuthFuture.verify(DefaultAuthFuture.java:39)
	at org.apache.sshd.client.future.DefaultAuthFuture.verify(DefaultAuthFuture.java:32)
	at org.apache.sshd.common.future.VerifiableFuture.verify(VerifiableFuture.java:68)
	at org.eclipse.jgit.transport.sshd.SshdSession.connect(SshdSession.java:164)
	... 27 more
Caused by: org.apache.sshd.common.SshException: No more authentication methods available
	at org.apache.sshd.client.session.ClientUserAuthService.tryNext(ClientUserAuthService.java:353)
	at org.apache.sshd.client.session.ClientUserAuthService.processUserAuth(ClientUserAuthService.java:288)
	at org.apache.sshd.client.session.ClientUserAuthService.process(ClientUserAuthService.java:225)
	at org.apache.sshd.common.session.helpers.AbstractSession.doHandleMessage(AbstractSession.java:503)
	at org.apache.sshd.common.session.helpers.AbstractSession.handleMessage(AbstractSession.java:429)
[... cut ...]
	at org.apache.sshd.common.io.nio2.Nio2CompletionHandler.completed(Nio2CompletionHandler.java:37)
	at sun.nio.ch.Invoker.invokeUnchecked(Invoker.java:126)
	at sun.nio.ch.Invoker$2.run(Invoker.java:218)
	at sun.nio.ch.AsynchronousChannelGroupImpl$1.run(AsynchronousChannelGroupImpl.java:112)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:750)
Exception running application com.neuronrobotics.bowlerstudio.BowlerStudio
```

sammcj, 2023-04-14:
And if I completely remove my known hosts file:

```
No splash screen availible!
 Splash Rendering 2 Testing Internet
 Splash Rendering 5 Loging In...
Attempt to log in with disk credentials
No login ID found!
 Splash Rendering 20 DL'ing Image Assets
Workspace: /Users/samm/bowler-workspace
Cloning files from: https://github.com/madhephaestus/BowlerStudioImageAssets.git
                to: /Users/samm/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets
136 [main] INFO org.apache.sshd.common.util.security.eddsa.EdDSASecurityProviderRegistrar - getOrCreateProvider(EdDSA) created instance of net.i2p.crypto.eddsa.EdDSASecurityProvider
195 [main] INFO org.apache.sshd.common.io.DefaultIoServiceFactoryFactory - No detected/configured IoServiceFactoryFactory using Nio2ServiceFactoryFactory
264 [sshd-JGitSshClient[475835b1]-nio2-thread-1] WARN org.eclipse.jgit.internal.transport.sshd.JGitClientSession - Ssh config JGitClientSession[null@github.com/20.248.137.48:22]: ignoring unknown algorithm 'curve25519-sha256@libssh.org' in KexAlgorithms curve25519-sha256@libssh.org,ecdh-sha2-nistp521,ecdh-sha2-nistp384,ecdh-sha2-nistp256,diffie-hellman-group-exchange-sha256
981 [sshd-JGitSshClient[475835b1]-nio2-thread-3] WARN org.eclipse.jgit.internal.transport.sshd.JGitClientSession - exceptionCaught(JGitClientSession[git@github.com/20.248.137.48:22])[state=Opened] SshException: Server key did not validate
982 [sshd-JGitSshClient[475835b1]-nio2-thread-3] INFO org.eclipse.jgit.internal.transport.sshd.JGitClientSession - Disconnecting(JGitClientSession[git@github.com/20.248.137.48:22]): SSH2_DISCONNECT_HOST_KEY_NOT_VERIFIABLE - Server key did not validate
org.eclipse.jgit.api.errors.TransportException: git@github.com:madhephaestus/BowlerStudioImageAssets.git: Server key did not validate
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:224)
	at org.eclipse.jgit.api.CloneCommand.fetch(CloneCommand.java:311)
	at org.eclipse.jgit.api.CloneCommand.call(CloneCommand.java:182)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepoLocalSelectAuth(ScriptingEngine.java:192)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepoLocal(ScriptingEngine.java:212)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1606)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:387)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: org.eclipse.jgit.errors.TransportException: git@github.com:madhephaestus/BowlerStudioImageAssets.git: Server key did not validate
	at org.eclipse.jgit.transport.sshd.SshdSessionFactory.getSession(SshdSessionFactory.java:248)
	at org.eclipse.jgit.transport.sshd.SshdSessionFactory.getSession(SshdSessionFactory.java:1)
	at org.eclipse.jgit.transport.SshTransport.getSession(SshTransport.java:107)
	at org.eclipse.jgit.transport.TransportGitSsh$SshFetchConnection.<init>(TransportGitSsh.java:281)
	at org.eclipse.jgit.transport.TransportGitSsh.openFetch(TransportGitSsh.java:153)
	at org.eclipse.jgit.transport.FetchProcess.executeImp(FetchProcess.java:142)
	at org.eclipse.jgit.transport.FetchProcess.execute(FetchProcess.java:94)
	at org.eclipse.jgit.transport.Transport.fetch(Transport.java:1309)
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:213)
	... 17 more
Caused by: org.apache.sshd.common.SshException: Server key did not validate
	at org.apache.sshd.common.future.AbstractSshFuture.verifyResult(AbstractSshFuture.java:126)
	at org.apache.sshd.client.future.DefaultAuthFuture.verify(DefaultAuthFuture.java:39)
	at org.apache.sshd.client.future.DefaultAuthFuture.verify(DefaultAuthFuture.java:32)
	at org.apache.sshd.common.future.VerifiableFuture.verify(VerifiableFuture.java:68)
	at org.eclipse.jgit.transport.sshd.SshdSession.connect(SshdSession.java:164)
	at org.eclipse.jgit.transport.sshd.SshdSession.connect(SshdSession.java:99)
	at org.eclipse.jgit.transport.sshd.SshdSessionFactory.getSession(SshdSessionFactory.java:235)
	... 25 more
Caused by: org.apache.sshd.common.SshException: Server key did not validate
	at org.apache.sshd.client.session.AbstractClientSession.checkKeys(AbstractClientSession.java:629)
	at org.apache.sshd.common.session.helpers.AbstractSession.handleKexMessage(AbstractSession.java:612)
	at org.apache.sshd.common.session.helpers.AbstractSession.doHandleMessage(AbstractSession.java:501)
	at org.apache.sshd.common.session.helpers.AbstractSession.handleMessage(AbstractSession.java:429)
	at org.apache.sshd.common.session.helpers.AbstractSession.decode(AbstractSession.java:1466)
	at org.apache.sshd.common.session.helpers.AbstractSession.messageReceived(AbstractSession.java:389)
[... cut ...]
	at sun.nio.ch.Invoker.invokeUnchecked(Invoker.java:126)
	at sun.nio.ch.Invoker$2.run(Invoker.java:218)
	at sun.nio.ch.AsynchronousChannelGroupImpl$1.run(AsynchronousChannelGroupImpl.java:112)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:750)
Exception running application com.neuronrobotics.bowlerstudio.BowlerStudio
```

Never seen anything like this!

madhephaestus, 2023-04-14:
in your ~/.gitconfig, do you have a full user section?

```
[user]
	email = mad.hephaestus@gmail.com
	name = Kevin Harrington
```

BTW `https://github.com/madhephaestus/BowlerStudioImageAssets.git` is a personal repo, but is not private. The pattern is to search for the logged in users repo, with that being the 'default' in case the user does not have their own. 

Can you try to run in bash:

```
git clone git@github.com:madhephaestus/BowlerStudioImageAssets.git
```

just to sanity check the environment?

madhephaestus, 2023-04-14:
for reference, when this works, these INFO logs print:

```
322 [main] INFO org.apache.sshd.common.util.security.eddsa.EdDSASecurityProviderRegistrar - getOrCreateProvider(EdDSA) created instance of net.i2p.crypto.eddsa.EdDSASecurityProvider
434 [main] INFO org.apache.sshd.common.io.DefaultIoServiceFactoryFactory - No detected/configured IoServiceFactoryFactory using Nio2ServiceFactoryFactory
```

sammcj, 2023-04-15:
Yep the gitconfig looks pretty normal - and it works with every other Github git repository I've tried it with (I have 138 repos cloned right now).

```
~/git ls -l . client1/ client2/ sammcj/ 3dprinting/|wc -l
     138
```

I can clone `git clone git@github.com:madhephaestus/BowlerStudioImageAssets.git` without any issue at all - it only fails when the app / Java is trying to do it.

```
git clone git@github.com:madhephaestus/BowlerStudioImageAssets.git
Cloning into 'BowlerStudioImageAssets'...
remote: Enumerating objects: 1240, done.
remote: Counting objects: 100% (153/153), done.
remote: Compressing objects: 100% (62/62), done.
remote: Total 1240 (delta 98), reused 143 (delta 89), pack-reused 1087
Receiving objects: 100% (1240/1240), 2.43 MiB | 2.88 MiB/s, done.
Resolving deltas: 100% (651/651), done.
```

---

I just temporarily removed the setting that makes sure Github uses the git protocol rather than an HTTPS URL and it worked!

So I suspect the problem must be with how the app / Java is doing the git clone - I wonder if it's assuming people are using git over HTTPS?

Also - is there any reason the git clone is run from within Java rather than during an install / run script?

sammcj, 2023-04-15:
BTW when it launched from your branch it wanted to grant Common Wealth Robotics full admin access to all my Github + Organisations and teams I'm a member of!

[image]

madhephaestus, 2023-04-15:
thats cool that its working! 

It accesses repos because it can be used to manage projects acrodd your github projects. Its fine to deny access to any current projects. It just needs to be able to make new repos, thats how it understands "projects" internally.  Thus the admin permissions. I have some users that have sensitive stuff in their regular github account, so they just made burners for use in bowlerstudio. 

It is assuming https access, because all the internal access is via https and the token requested from github in the webflow step.  You are teh first user to have had this substitution based change from https to ssh in a gitconfig. You are not wrong, so its just a matter of making the java git client, and in this case the Apache SSHD layer to play nice.

madhephaestus, 2023-04-15:
as for the reason java layer is doing the git clones, for the most part it treats git as the filesystem to read and write from for the scripts. The cad layer uses this git filesystem to ensure saving your code, is actually making a commit and a push. 

This means that any code that calls another piece of code, does so by its git url and file path. any code that runs on my system, therefore, will run on anyone elses. 

This is meant to enable a sharing first mentality to CAD and put collaboration as a first class activity. It was meant as a teaching tool for students learning programming for the first time, or for developers familliar with programming that want a powerful cad package without learning complicated new UI's. Getting this working for someone with a sophisticated git setup is definantly a use-case i want to support.

madhephaestus, 2023-04-15:
Oh, and without the API key the local source copy will not be able to log in, so i think if you just run the binary copy with the special case in the gitconfig turned off it should be able to login.

closed 2023-04-15 by madhephaestus

sammcj, 2023-04-15:
That's cool, makes sense!

Awesome thanks for getting this working and explaining the internals a bit along the way.

madhephaestus, 2023-04-17:
I was able to get the git@ override working for me, and tbh it may be something i use long term, its a great shortcut. I do not however get the failed authentication you get. can you open an issue with the place where it failed using the git@ overrides for you?