CommonWealthRobotics/BowlerStudio #371: Undeclared Mem error in Windows

JansenSmith, 2023-07-23:
debug info:

java.lang.OutOfMemoryError: Java heap space
	at java.awt.image.DataBufferInt.<init>(DataBufferInt.java:75)
	at java.awt.image.Raster.createPackedRaster(Raster.java:467)
	at java.awt.image.DirectColorModel.createCompatibleWritableRaster(DirectColorModel.java:1032)
	at java.awt.image.BufferedImage.<init>(BufferedImage.java:351)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:420)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$686/1388524169.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

madhephaestus, 2023-07-24:
Hmm, can you take a look at your BowlerStudio.bat file and see what heap size is specified?

JansenSmith, 2023-07-24:
> Hmm, can you take a look at your BowlerStudio.bat file and see what heap size is specified?

C:\Program Files (x86)\Commonwealth Robotics BowlerStudio\BowlerStudioApp\BowlerStudio.bat

@madhephaestus i don't see anything referencing heap in the contents
"%~dp0\jre\bin\java.exe" -jar "%~dp0\LatestFromGithubLaunch.jar"  CommonWealthRobotics BowlerStudio BowlerStudio.jar  "%~dp0\jre\bin\java.exe" -jar

madhephaestus, 2023-07-24:
what files are are you opening?

How many chars?

madhephaestus, 2023-07-24:
Did that fix it?

JansenSmith, 2023-07-24:
checking

JansenSmith, 2023-07-24:
@madhephaestus 
crashed again. only one file open - https://github.com/Halloween2020TheChild/Zoltar/blob/main/CrystalStand.groovy
901 Bytes

Code in C:\Users\Jansen\bowler-workspace\gitcache\github.com\Halloween2020TheChild\Zoltar\CrystalStand.groovy changed
java.lang.OutOfMemoryError: Java heap space
	at java.awt.image.DataBufferInt.<init>(DataBufferInt.java:75)
	at java.awt.image.Raster.createPackedRaster(Raster.java:467)
	at java.awt.image.DirectColorModel.createCompatibleWritableRaster(DirectColorModel.java:1032)
	at java.awt.image.BufferedImage.<init>(BufferedImage.java:351)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:420)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$689/158004957.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:228)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

Saving history

JansenSmith, 2023-07-24:
@madhephaestus 
this time, didn't crash..... hanged on running the script, remained hanged for ~6 minutes before i tried to press stop in Bowler. eventually seemed to successfully Stop but then crashed with this different error:
Skip Writing file contents, file is same
Script exception of type= java.lang.RuntimeException

C:\Users\Jansen\bowler-workspace\gitcache\github.com\Halloween2020TheChild\Zoltar\CrystalStand.groovy Interupted

java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:228)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$689/1430910706.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
script error
Saving history

JansenSmith, 2023-07-25:
crashed again
java.lang.OutOfMemoryError: Java heap space
	at sun.swing.JLightweightFrame.syncCopyBuffer(JLightweightFrame.java:276)
	at sun.swing.JLightweightFrame.resizeBuffer(JLightweightFrame.java:424)
	at sun.swing.JLightweightFrame.reshape(JLightweightFrame.java:407)
	at java.awt.Component.setBounds(Component.java:2261)
	at java.awt.Window.setBounds(Window.java:3503)
	at java.awt.Component.resize(Component.java:2184)
	at java.awt.Component.setSize(Component.java:2173)
	at java.awt.Window.setSize(Window.java:908)
	at javafx.embed.swing.SwingNode.lambda$locateLwFrame$20(SwingNode.java:683)
	at javafx.embed.swing.SwingNode$$Lambda$690/2124599769.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:228)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:58)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

Saving history

madhephaestus, 2023-07-25:
Can you run JVisualVM at the same time and look for runnaway threads or nail down which thread is using all the memory? Even better, pause the thread using the memory in the debugger to see what its doing?

madhephaestus, 2023-07-25:
I have labeled this a hardware problem until the system passes memtest.

madhephaestus, 2025-01-08:
this may also be solved by https://github.com/CommonWealthRobotics/BowlerStudio/commit/2ad35f38bd16a212ff49594c70d086bc670fa30c

closed 2025-01-08 by madhephaestus