CommonWealthRobotics/BowlerStudio #372: Build 2.5.2 com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread at line 15

madhephaestus, 2023-07-24:
Auto Reported Issue 
BowlerStudio Build 2.5.2
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/madhephaestus/SVGBowlerExtrude.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:15)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:400)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:367)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.locateGit(ScriptingEngine.java:1669)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkOwner(ScriptingEngine.java:1792)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:119)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:89)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:131)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:142)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:352)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:586)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:282)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1400(BowlerStudioMenu.java:69)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$run$0(BowlerStudioMenu.java:250)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:228)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

dromero2025, 2024-01-18:
Auto Reported Issue 
BowlerStudio Build 2.11.0
BowlerKernel 1.59.0
JavaCad Version: 0.30.1
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/dromero2025/BowlerStudioConfiguration.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:15)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:407)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:374)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1315)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1442)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getDatabase(ConfigurationDatabase.java:106)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.setGitSource(ConfigurationDatabase.java:224)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.setRepo(ConfigurationDatabase.java:213)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getGitSource(ConfigurationDatabase.java:185)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.loadFile(ConfigurationDatabase.java:131)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getDatabase(ConfigurationDatabase.java:104)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.getParamMap(ConfigurationDatabase.java:50)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.setObject(ConfigurationDatabase.java:57)
	at com.neuronrobotics.bowlerstudio.Tutorial.getHomeUrl(Tutorial.java:24)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:439)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

JunmingYe, 2024-02-02:
Auto Reported Issue 
BowlerStudio Build 2.11.1
BowlerKernel 1.59.0
JavaCad Version: 0.30.1
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_332
JavaFX Version: 8.0.332

OS = Mac OS X x86_64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/BancroftKineticSystemsClass/KineticSystems_2024_04.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:15)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:407)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:374)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1159)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1130)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1379)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1444)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$loginEvent$0(BowlerStudioMenuWorkspace.java:51)
	at java.lang.Thread.run(Thread.java:750)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

GMKtillDecay, 2024-02-08:
Auto Reported Issue 
BowlerStudio Build 2.11.1
BowlerKernel 1.59.0
JavaCad Version: 0.30.1
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/GMKtillDecay/BowlerStudioConfiguration.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:15)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:407)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:374)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pushCodeToGit(ScriptingEngine.java:931)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pushCodeToGit(ScriptingEngine.java:800)
	at com.neuronrobotics.bowlerstudio.assets.ConfigurationDatabase.save(ConfigurationDatabase.java:80)
	at com.neuronrobotics.bowlerstudio.BowlerStudio$6.run(BowlerStudio.java:799)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

shstueber, 2024-02-08:
Auto Reported Issue 
BowlerStudio Build 2.11.1
BowlerKernel 1.59.0
JavaCad Version: 0.30.1
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/BancroftKineticSystemsClass/KineticSystems_2024_07.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:15)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:407)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:374)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1159)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.newBranch(ScriptingEngine.java:1130)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1379)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1444)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenuWorkspace.lambda$loginEvent$0(BowlerStudioMenuWorkspace.java:51)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

ZocP, 2024-04-30:
Auto Reported Issue 
BowlerStudio Build 2.18.7
BowlerKernel 1.59.0
JavaCad Version: 0.31.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_332
JavaFX Version: 8.0.332

OS = Mac OS X aarch64 x86
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/madhephaestus/conePointSetScrew.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:15)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:408)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:375)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1318)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1447)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.ensureUpdated(BowlerStudio.java:564)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:541)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

JansenSmith, 2024-05-09:
Auto Reported Issue 
BowlerStudio Build 2.21.1
BowlerKernel 1.59.0
JavaCad Version: 0.31.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/JansenSmith/OpensourcererSpaceMouse.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:15)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:408)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:375)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.commit(ScriptingEngine.java:2221)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.commit(ScriptingEngine.java:863)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.commit(ScriptingEngine.java:821)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pushCodeToGit(ScriptingEngine.java:900)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pushCodeToGit(ScriptingEngine.java:801)
	at com.neuronrobotics.nrconsole.util.CommitWidget$1.run(CommitWidget.java:106)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

JansenSmith, 2024-07-29:
Auto Reported Issue 
BowlerStudio Build 2.37.1
BowlerKernel 1.59.0
JavaCad Version: 1.4.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/JansenSmith/Technocopia_logo_chalk_painting.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:15)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:458)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:425)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.locateGit(ScriptingEngine.java:1783)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkOwner(ScriptingEngine.java:1915)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.load(ScriptingFileWidget.java:124)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:102)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:136)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:143)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:367)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:737)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$8$1$1.run(BowlerStudioMenu.java:1054)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

onxr92, 2024-11-19:
Auto Reported Issue 
BowlerStudio Build 3.1.3
BowlerKernel 1.59.0
JavaCad Version: 2.0.2
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_412
JavaFX Version: 8.0.412

OS = Windows 11 amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/madhephaestus/encoderCadGenerator.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:15)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:479)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:441)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1432)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1561)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.ensureUpdated(BowlerStudio.java:687)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.lambda$main$4(BowlerStudio.java:358)
	at java.lang.Thread.run(Thread.java:750)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

JansenSmith, 2024-12-20:
Auto Reported Issue 
BowlerStudio Build 3.2.1
BowlerKernel 1.59.0
JavaCad Version: 2.0.5
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/JansenSmith/ArtPiece.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:15)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:480)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:442)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.locateGit(ScriptingEngine.java:1846)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkOwner(ScriptingEngine.java:1977)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.load(ScriptingFileWidget.java:124)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:102)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:136)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:144)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:367)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:766)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:286)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1400(BowlerStudioMenu.java:74)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$run$0(BowlerStudioMenu.java:252)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

JansenSmith, 2024-12-20:
Auto Reported Issue 
BowlerStudio Build 3.2.1
BowlerKernel 1.59.0
JavaCad Version: 2.0.5
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/JansenSmith/TechOrnament.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:15)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:480)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:442)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.locateGit(ScriptingEngine.java:1846)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkOwner(ScriptingEngine.java:1977)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.load(ScriptingFileWidget.java:124)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:102)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:136)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:144)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:367)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:766)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:286)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1400(BowlerStudioMenu.java:74)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$run$0(BowlerStudioMenu.java:252)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

JansenSmith, 2024-12-20:
Auto Reported Issue 
BowlerStudio Build 3.2.1
BowlerKernel 1.59.0
JavaCad Version: 2.0.5
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/JansenSmith/ArtPiece.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:15)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:480)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:442)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.locateGit(ScriptingEngine.java:1846)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkOwner(ScriptingEngine.java:1977)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.load(ScriptingFileWidget.java:124)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:102)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:136)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:144)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:367)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:766)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$8$1$1.run(BowlerStudioMenu.java:1061)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

JansenSmith, 2024-12-20:
Auto Reported Issue 
BowlerStudio Build 3.2.1
BowlerKernel 1.59.0
JavaCad Version: 2.0.5
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/JansenSmith/ArtPiece.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:15)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:480)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:442)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.commit(ScriptingEngine.java:2336)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.commit(ScriptingEngine.java:974)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.commit(ScriptingEngine.java:929)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pushCodeToGit(ScriptingEngine.java:1015)
	at com.neuronrobotics.nrconsole.util.CommitWidget$1.run(CommitWidget.java:122)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

JansenSmith, 2024-12-20:
```
9280114 [Thread-1117] WARN org.eclipse.jgit.lib.Repository - close() called when useCnt is already zero for Repository[/home/jansen/Documents/bowler-workspace/gitcache/github.com/JansenSmith/ArtPiece/.git]
```
@madhephaestus this is what the terminal said

JansenSmith, 2025-01-08:
Auto Reported Issue 
BowlerStudio Build 3.3.0
BowlerKernel 1.59.0
JavaCad Version: 2.1.1
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.RuntimeException: Git opened here, timeout on close!!
When Done with the git object, Call:
 	ScriptingEngine.closeGit(git);
https://github.com/JansenSmith/ArtPiece.git

	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.<init>(GitTimeoutThread.java:15)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.makeTimeoutThread(ScriptingEngine.java:480)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.openGit(ScriptingEngine.java:442)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.locateGit(ScriptingEngine.java:1846)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkOwner(ScriptingEngine.java:1977)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.load(ScriptingFileWidget.java:124)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget.<init>(ScriptingFileWidget.java:102)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<init>(LocalFileScriptTab.java:136)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:144)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:367)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:766)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:286)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1400(BowlerStudioMenu.java:74)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$run$0(BowlerStudioMenu.java:252)
	at java.lang.Thread.run(Thread.java:748)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.scripting.GitTimeoutThread.run(GitTimeoutThread.java:26)

```

madhephaestus, 2025-01-09:
I re-archetectured the Git stack based on some Rust-like tricks of the Borrow Checker

https://github.com/CommonWealthRobotics/bowler-script-kernel/commit/b92e54396330383a23bb717fa4c46fad2ef3f4e0

closed 2025-01-09 by madhephaestus