CommonWealthRobotics/BowlerStudio #376: Build source javax.swing.text.AbstractDocument$BranchElement at line 2350

madhephaestus, 2023-08-17:
Auto Reported Issue 
BowlerStudio Build source
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.ArrayIndexOutOfBoundsException: -1
	at javax.swing.text.AbstractDocument$BranchElement.getElement(AbstractDocument.java:2350)
	at org.fife.ui.rsyntaxtextarea.RSyntaxDocument.getTokenListForLine(RSyntaxDocument.java:416)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.possiblyCloseCurlyBrace(RSyntaxTextAreaEditorKit.java:1567)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.insertNewlineWithAutoIndent(RSyntaxTextAreaEditorKit.java:1550)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.handleInsertBreak(RSyntaxTextAreaEditorKit.java:1488)
	at org.fife.ui.rsyntaxtextarea.AbstractJFlexCTokenMaker$CStyleInsertBreakAction.actionPerformedImpl(AbstractJFlexCTokenMaker.java:160)
	at org.fife.ui.rtextarea.RecordableTextAction.actionPerformed(RecordableTextAction.java:106)
	at org.fife.ui.rsyntaxtextarea.RSyntaxDocument.insertBreakSpecialHandling(RSyntaxDocument.java:440)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.actionPerformedImpl(RSyntaxTextAreaEditorKit.java:1433)
	at org.fife.ui.rtextarea.RecordableTextAction.actionPerformed(RecordableTextAction.java:106)
	at javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1668)
	at javax.swing.JComponent.processKeyBinding(JComponent.java:2882)
	at javax.swing.JComponent.processKeyBindings(JComponent.java:2929)
	at javax.swing.JComponent.processKeyEvent(JComponent.java:2845)
	at java.awt.Component.processEvent(Component.java:6316)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
	at java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:834)
	at java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1102)
	at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:973)
	at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:799)
	at java.awt.Component.dispatchEventImpl(Component.java:4760)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:228)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:60)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:42)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:50)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:142)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:352)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:586)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:282)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1(BowlerStudioMenu.java:267)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$0(BowlerStudioMenu.java:250)
	at java.lang.Thread.run(Thread.java:748)

```

EmmanuelAsumadu, 2023-08-20:
Auto Reported Issue 
BowlerStudio Build 2.7.1
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.ArrayIndexOutOfBoundsException: -1
	at javax.swing.text.AbstractDocument$BranchElement.getElement(AbstractDocument.java:2350)
	at org.fife.ui.rsyntaxtextarea.RSyntaxDocument.getTokenListForLine(RSyntaxDocument.java:416)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.possiblyCloseCurlyBrace(RSyntaxTextAreaEditorKit.java:1567)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.insertNewlineWithAutoIndent(RSyntaxTextAreaEditorKit.java:1550)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.handleInsertBreak(RSyntaxTextAreaEditorKit.java:1488)
	at org.fife.ui.rsyntaxtextarea.AbstractJFlexCTokenMaker$CStyleInsertBreakAction.actionPerformedImpl(AbstractJFlexCTokenMaker.java:160)
	at org.fife.ui.rtextarea.RecordableTextAction.actionPerformed(RecordableTextAction.java:106)
	at org.fife.ui.rsyntaxtextarea.RSyntaxDocument.insertBreakSpecialHandling(RSyntaxDocument.java:440)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.actionPerformedImpl(RSyntaxTextAreaEditorKit.java:1433)
	at org.fife.ui.rtextarea.RecordableTextAction.actionPerformed(RecordableTextAction.java:106)
	at javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1668)
	at javax.swing.JComponent.processKeyBinding(JComponent.java:2882)
	at javax.swing.JComponent.processKeyBindings(JComponent.java:2929)
	at javax.swing.JComponent.processKeyEvent(JComponent.java:2845)
	at java.awt.Component.processEvent(Component.java:6316)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
	at java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:834)
	at java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1102)
	at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:973)
	at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:799)
	at java.awt.Component.dispatchEventImpl(Component.java:4760)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:62)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:53)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:52)
[... cut ...]
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:352)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:586)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:282)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1400(BowlerStudioMenu.java:69)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$run$0(BowlerStudioMenu.java:250)
	at java.lang.Thread.run(Thread.java:748)

```

dnlr-p, 2023-08-26:
Auto Reported Issue 
BowlerStudio Build 2.7.4
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.ArrayIndexOutOfBoundsException: -1
	at javax.swing.text.AbstractDocument$BranchElement.getElement(AbstractDocument.java:2350)
	at org.fife.ui.rsyntaxtextarea.RSyntaxDocument.getTokenListForLine(RSyntaxDocument.java:416)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.possiblyCloseCurlyBrace(RSyntaxTextAreaEditorKit.java:1567)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.insertNewlineWithAutoIndent(RSyntaxTextAreaEditorKit.java:1550)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.handleInsertBreak(RSyntaxTextAreaEditorKit.java:1488)
	at org.fife.ui.rsyntaxtextarea.AbstractJFlexCTokenMaker$CStyleInsertBreakAction.actionPerformedImpl(AbstractJFlexCTokenMaker.java:160)
	at org.fife.ui.rtextarea.RecordableTextAction.actionPerformed(RecordableTextAction.java:106)
	at org.fife.ui.rsyntaxtextarea.RSyntaxDocument.insertBreakSpecialHandling(RSyntaxDocument.java:440)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.actionPerformedImpl(RSyntaxTextAreaEditorKit.java:1433)
	at org.fife.ui.rtextarea.RecordableTextAction.actionPerformed(RecordableTextAction.java:106)
	at javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1668)
	at javax.swing.JComponent.processKeyBinding(JComponent.java:2882)
	at javax.swing.JComponent.processKeyBindings(JComponent.java:2929)
	at javax.swing.JComponent.processKeyEvent(JComponent.java:2845)
	at java.awt.Component.processEvent(Component.java:6316)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
	at java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:834)
	at java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1102)
	at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:973)
	at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:799)
	at java.awt.Component.dispatchEventImpl(Component.java:4760)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:62)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:53)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:52)
[... cut ...]
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:352)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:586)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:282)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1400(BowlerStudioMenu.java:69)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$run$0(BowlerStudioMenu.java:250)
	at java.lang.Thread.run(Thread.java:748)

```

madhephaestus, 2023-08-31:
Auto Reported Issue 
BowlerStudio Build 2.7.5
BowlerKernel 1.59.0
JavaCad Version: 0.28.0
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.ArrayIndexOutOfBoundsException: -1
	at javax.swing.text.AbstractDocument$BranchElement.getElement(AbstractDocument.java:2350)
	at org.fife.ui.rsyntaxtextarea.RSyntaxDocument.getTokenListForLine(RSyntaxDocument.java:416)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.possiblyCloseCurlyBrace(RSyntaxTextAreaEditorKit.java:1567)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.insertNewlineWithAutoIndent(RSyntaxTextAreaEditorKit.java:1550)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.handleInsertBreak(RSyntaxTextAreaEditorKit.java:1488)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextAreaEditorKit$InsertBreakAction.actionPerformedImpl(RSyntaxTextAreaEditorKit.java:1438)
	at org.fife.ui.rtextarea.RecordableTextAction.actionPerformed(RecordableTextAction.java:106)
	at javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1668)
	at javax.swing.JComponent.processKeyBinding(JComponent.java:2882)
	at javax.swing.JComponent.processKeyBindings(JComponent.java:2929)
	at javax.swing.JComponent.processKeyEvent(JComponent.java:2845)
	at java.awt.Component.processEvent(Component.java:6316)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
	at java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:834)
	at java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1102)
	at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:973)
	at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:799)
	at java.awt.Component.dispatchEventImpl(Component.java:4760)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:62)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:53)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:52)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:142)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:352)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:586)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:282)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1400(BowlerStudioMenu.java:69)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$run$0(BowlerStudioMenu.java:250)
	at java.lang.Thread.run(Thread.java:748)

```