CommonWealthRobotics/BowlerStudio #385: GitHub Pull Remote Error

dnlr-p, 2023-08-28:
Linux Mint
Error preventing launch

~/bin$ bash $HOME/bin/bowlerstudio
Skipping adding ~/bin/ to /home/~~/.bashrc, command already exists
https://github.com/CommonWealthRobotics/BowlerStudio/releases/download/2.7.5/BowlerStudio.jar Size 351697429 bytes
Running:

/home/~~/bin/java8//bin/java -Dprism.forceGPU=true -jar /home/~~/bin/BowlerStudioInstall/2.7.5/BowlerStudio.jar

No splash screen availible!
 Splash Rendering 2 Testing Internet
 Splash Rendering 5 Loging In...
Attempt to log in with disk credentials
No login ID found!
 Splash Rendering 20 DL'ing Image Assets
Workspace: /home/~~/Documents/bowler-workspace

Success Login dnlr-p

1143 [main] WARN org.eclipse.jgit.lib.Repository - close() called when useCnt is already zero for Repository[/home/~~/Documents/bowler-workspace/gitcache/github.com/madhephaestus/BowlerStudioImageAssets/.git]
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:873)
Caused by: org.eclipse.jgit.api.errors.InvalidRemoteException: remoteURI https://github.com/madhephaestus/BowlerStudioImageAssets.git branch main remoteURI https://github.com/madhephaestus/BowlerStudioImageAssets.git branch main Invalid remote: origin
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.pull(ScriptingEngine.java:1426)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:425)
	... 11 more
Exception running application com.neuronrobotics.bowlerstudio.BowlerStudio
LatestFromGithubLaunch clean exit

madhephaestus, 2023-08-29:
it looks like the user was not set up like a standard user. The user directory being set with the wrong permissions and this suggest the user creation process failed in the first place. How was the user created?

madhephaestus, 2023-09-02:
Resolved by: 

Remove bowler-workspace/gitcache/ 

removing ~/bin/<everything that isnt a shell script>

The issue arose from https://github.com/CommonWealthRobotics/ESP32ArduinoEclipseInstaller and an issue should be raised there to change from cmd line sudo to the graphical sudo.

closed 2023-09-02 by madhephaestus