CommonWealthRobotics/BowlerStudio #399: Build source com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory at line 573

madhephaestus, 2024-05-17:
Auto Reported Issue 
BowlerStudio Build source
BowlerKernel 1.59.0
JavaCad Version: 1.0.5
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_242
JavaFX Version: 8.0.242

OS = Linux amd64 x64
```
java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.saveToXML(MobleBaseMenueFactory.java:573)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.reload(MobleBaseMenueFactory.java:547)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.access$2(MobleBaseMenueFactory.java:546)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory$7.run(MobleBaseMenueFactory.java:746)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:62)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:42)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:53)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:52)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:141)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:352)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:652)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:285)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1(BowlerStudioMenu.java:270)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$0(BowlerStudioMenu.java:253)
	at java.lang.Thread.run(Thread.java:748)

```

2442Steven, 2024-05-17:
Auto Reported Issue 
BowlerStudio Build 2.23.0
BowlerKernel 1.59.0
JavaCad Version: 1.0.2
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_412
JavaFX Version: 8.0.412

OS = Mac OS X aarch64 x86
```
java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.saveToXML(MobleBaseMenueFactory.java:573)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.reload(MobleBaseMenueFactory.java:547)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.access$500(MobleBaseMenueFactory.java:56)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory$7.run(MobleBaseMenueFactory.java:746)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:62)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1964)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:53)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:52)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:141)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:352)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:652)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:285)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1400(BowlerStudioMenu.java:72)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$run$0(BowlerStudioMenu.java:253)
	at java.lang.Thread.run(Thread.java:750)

```

jmartin-bancroft, 2024-05-17:
Auto Reported Issue 
BowlerStudio Build 2.23.0
BowlerKernel 1.59.0
JavaCad Version: 1.0.2
Java-Bowler Version: 3.34.0
Java Version: 1.8.0_212
JavaFX Version: 8.0.212

OS = Windows 10 amd64 x64
```
java.lang.NullPointerException
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.saveToXML(MobleBaseMenueFactory.java:573)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.reload(MobleBaseMenueFactory.java:547)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory.access$500(MobleBaseMenueFactory.java:56)
	at com.neuronrobotics.bowlerstudio.creature.MobleBaseMenueFactory$7.run(MobleBaseMenueFactory.java:746)

```

Caught and reported at: 
```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.except(IssueReportingExceptionHandler.java:234)
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.uncaughtException(IssueReportingExceptionHandler.java:51)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.uncaughtException(LocalFileScriptTab.java:62)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.lang.Thread.dispatchUncaughtException(Thread.java:1959)

```

IssueReportingExceptionHandler Created at:

```
java.lang.Exception
	at com.neuronrobotics.bowlerstudio.IssueReportingExceptionHandler.<init>(IssueReportingExceptionHandler.java:41)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab$1.<init>(LocalFileScriptTab.java:53)
	at com.neuronrobotics.bowlerstudio.tabs.LocalFileScriptTab.<clinit>(LocalFileScriptTab.java:52)
	at com.neuronrobotics.bowlerstudio.BowlerStudioController.createFileTab(BowlerStudioController.java:141)
	at com.neuronrobotics.bowlerstudio.BowlerStudioModularFrame.createFileTab(BowlerStudioModularFrame.java:352)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.createFileTab(BowlerStudio.java:652)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.openFilesInUI(BowlerStudioMenu.java:285)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu.access$1400(BowlerStudioMenu.java:72)
	at com.neuronrobotics.bowlerstudio.BowlerStudioMenu$2.lambda$run$0(BowlerStudioMenu.java:253)
	at java.lang.Thread.run(Thread.java:748)

```

commit d9f3e944 (2024-05-17): close #399
closed 2024-06-11 by madhephaestus