CommonWealthRobotics/BowlerStudio #401: Excessive access scope for GitHub OAuth app

mtilda, 2024-06-04:
It looks like the GitHub OAuth app is configured to request access to all scopes.

### Installed Version 

v2.25.1

### Expected behavior

A user should be able to specify which repositories and organizations are accessible by Bowler Studio.

I don't see why Bowler Studio would need access to personal user data.

### Actual Behavior

[image]

### Steps to reproduce the behavior

_You may need to revoke your existing OAuth credentials._

1. Open Bowler Studio.
2. In the toolbar, click GitHub > Log in to GItHub.
3. Type in your username, and click OK.

This will redirect you to GitHub's OAuth page, where you can see the access request.

mtilda, 2024-06-04:
It is an unintentional coincidence that this issue number (401) is the HTTP status code that means "Unauthorized."

madhephaestus, 2024-06-06:
BowlerStudio uses Git (and specifically github, although more are coming this summer) as its "filesystem". BowlerStudio creates, modifies then saves files to the git repository they come from. In the cad kernel, files can link to each other, but only by their public Git URL's. The point of the tool is to not simply encourage open source development, but to make closed source development out of pipeline. Using closed source repos in the BowlerStudio file system makes for potentially broken sources when the dependent script is published. The permissions requested allows users to use BowlerStudio on private repos, so long as they create them out of pipeline ant then open it in BowlerStudio later. 

That said, I totally understand the hesitance. The simplest solution would be to make a burner account in Github, an account you just point bowlerstudio at, and keep private information out of. 

I plan on making a gitlab and google docs plugin so the user can choose which one they wish to use. At the moment the github layer is perfused throughout the tool, so there is not a very good way to isolate certain content in github separate from from other content in a given account. 

I'll leave this issue open until there are other options, since the explanation might help other users in the future.

madhephaestus, 2024-06-07:
for reference, these are the scopes requested; https://github.com/CommonWealthRobotics/BowlerStudio/blob/ec69a0c5d13ee6f649aa1c961863b8c27f9b8bc6/src/main/java/com/neuronrobotics/bowlerstudio/BowlerStudio.java#L351 if you think there is a refinement to them i would be happy to test it. Running from source you will not be able to run a login without the API key. That gets added in CI during a release and encrypted into the binaries.

madhephaestus, 2024-06-07:
"repo" is added so the user can push/pull/tag a repo, this is how files are saved
"gist" is the same as repo, but got github gists
"user" is how BowlerStudio can look to see what repositories you have to populate the github menu. this list is how you add repos and is used to check if you have existing repos before forking 
"admin:org" is how forking and creation of new repos works. it also is used by project administration features where you create and delete repos for classroom management
"admin:org_hook" and "workflow" are needed to setup CI. When a Bowlerstudio script is released from in the tool it creates a CI script that compiles the CAD into release and production files (STL, SVG, OBJ, BoM) and puts those binaries into a release.

mtilda, 2024-07-16:
Makes sense to me! I see why this is this is low priority, given that there is an operational workaround:

> The simplest solution would be to make a burner account in Github, an account you just point bowlerstudio at, and keep private information out of.