CommonWealthRobotics/BowlerStudio #420: Polygon extrusion failure

JansenSmith, 2024-08-01:
run [this Groovy script](https://github.com/JansenSmith/TechnocopiaChalkPaintingFrame/blob/c1247b157b3cbd5bc00120522abe9cef5b17da49/TechnocopiaChalkPaintingFrame.groovy)

i got this error that implies something might be wrong w the Slice output:

Writing file contents
Slice took: 6.099 seconds
script error
java.lang.Exception: javax.vecmath.SingularMatrixException: cannot invert matrix
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget$3.run(ScriptingFileWidget.java:624)
Caused by: javax.vecmath.SingularMatrixException: cannot invert matrix
	at javax.vecmath.Matrix4d.invertGeneral(Matrix4d.java:1996)
	at javax.vecmath.Matrix4d.invert(Matrix4d.java:1951)
	at eu.mihosoft.vrl.v3d.Transform.inverse(Transform.java:594)
	at eu.mihosoft.vrl.v3d.ext.org.poly2tri.PolygonUtil.concaveToConvex(PolygonUtil.java:137)
	at eu.mihosoft.vrl.v3d.ext.org.poly2tri.PolygonUtil.concaveToConvex(PolygonUtil.java:105)
	at eu.mihosoft.vrl.v3d.Extrude.polygons(Extrude.java:166)
	at eu.mihosoft.vrl.v3d.Extrude.polygons(Extrude.java:131)
	at org.codehaus.groovy.vmplugin.v8.IndyInterface.fromCache(IndyInterface.java:321)
	at TechnocopiaChalkPaintingFrame.run(TechnocopiaChalkPaintingFrame.groovy:14)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:58)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:78)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:511)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineFileScriptRun(ScriptingEngine.java:1068)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingFileWidget$3.run(ScriptingFileWidget.java:582)

madhephaestus, 2024-08-02:
Where did the STL come from? Have you loaded it into blender and re-meshed it? 

Try

```
import com.neuronrobotics.bowlerstudio.util.GeometrySimplification

GeometrySimplification.remesh(logoSTL,1.0)// remesh the model file and save the remesh back to the same file
CSG logo  = Vitamins.get(logoSTL,true);// force a re-load and refresh the cache
```

madhephaestus, 2024-08-02:
If you need to simplify an SVG before loading into an STL, then use:

```
/**
	 * Simplify an SVG file
	 * @param incoming the incoming SVG file
	 * @param threshhold the threshhold value  (default is 0.002
	 * @return A new SVG file that is changed
	 */
	public static File simplifySVG(File incoming, double threshhold) {
		try {
			File inkscape = DownloadManager.getConfigExecutable("inkscape", null);
			File svg = File.createTempFile(incoming.getName(), ".svg");
			List <String >args = Arrays.asList(
					inkscape.getAbsolutePath(),
		            "--actions",
		            "\"select-all:all;path-simplify:threshold=" + threshhold+ ";;export-overwrite;export-do;quit-inkscape\"",
		            incoming.getAbsolutePath()
					);
			legacySystemRun(null, inkscape.getAbsoluteFile().getParentFile(), System.out, args);
			args = Arrays.asList(
					inkscape.getAbsolutePath(),
					"--export-plain-svg",
		            "--export-type=svg",
		            "--vacuum-defs",
		            "--export-filename="+svg.getAbsolutePath(),
		            incoming.getAbsolutePath()
					);
			legacySystemRun(null, inkscape.getAbsoluteFile().getParentFile(), System.out, args);
			return svg;
		} catch (Exception e) {
			// TODO Auto-generated catch block
			e.printStackTrace();
		}
		return incoming;
	}

```

JansenSmith, 2024-08-02:
i made the STL from a raster file, as an elevation map from brightness. so no SVG step to get to the STL. i then heavily edited the STL in Blender.... it still might be non-manifold tho. checking.

JansenSmith, 2024-08-02:
remesh(logoSTL,1.0)

is the voxel resolution then 1mm?

madhephaestus, 2024-08-02:
https://github.com/CommonWealthRobotics/bowler-script-kernel/blob/kh/add-openscad/src/main/java/com/neuronrobotics/bowlerstudio/util/GeometrySimplification.java